AdnanHodzic/displaylink-debian #172: This plugin is not recommended

kopax, 2018-11-19:
I've been watching and testing this module in debian in many kernel, clean version.

I keep trying again and again hoping I can find an usage for that screen.

I've just tried again tonight, on a new computer. Still does not work.

I can list two provider, but I can't see the screen in display, the `xrandr --setprovideroutputsource 1 0` does not turn anything on

AdnanHodzic, 2018-11-20:
@kopax what happens if you `mv /etc/X11/xorg.conf.d/20-displaylink.conf /tmp/20-displaylink.conf` and restart X (log out and log in). 

Also which version of Debian are using?

kopax, 2018-11-20:
Thanks for the quick reply @AdnanHodzic I appreciate.

I am using this version of Debian:

    Linux dev-12 4.9.0-8-amd64 #1 SMP Debian 4.9.110-3+deb9u6 (2018-10-08) x86_64 GNU/Linux

I will redo the test once I am done working on this computer. 

@AdnanHodzic, thanks for doing this module. I only use Debian and for about 3 years I was not able to use that MB168+ screen. An old version of this project used to work with Debian but that wasn't stable. After doing a Linux kernel update on my laptop (required by latest docker daemon for storage v2 driver), it stopped completely to work.

I expected to use a default Debian install would be working fine but it is not.
I would gladly contribute to testing if you can help me get this screen working in a reliable way, I don't have any knowledge with `xrandr`, I am web developer.

AdnanHodzic, 2018-11-21:
@kopax ok forget about removing `mv /etc/X11/xorg.conf.d/20-displaylink.conf /tmp/20-displaylink.conf`.

Every time you upgrade a new kernel you need to re-install, so could you please run "uninstall" and then reboot the computer and then do "reinstall". It seems like your displays are recognized just fine from what I see here.

kopax, 2018-11-22:
> Every time you upgrade a new kernel you need to re-install

As I said, I have one Lenovo X1Carbon Gen 1, I have updated the kernel once and since I am not able to use the screen, even with reunstalling, formating etc... I have no clue why, I am juste reporting.

> It seems like your displays are recognized just fine from what I see here.

Yes it is, but again, that's a desktop brand new computer. I did not upgrade the kernel or do any things with the core of linux since I installed it. Using that fresh install, the driver does not make it work and I have a freeze effect. I think this is the most common setup people will have.

AdnanHodzic, 2018-11-24:
@kopax did you pull latest changes (as of today), and run reinstall?

kopax, 2018-11-24:
You mean this one ? https://github.com/AdnanHodzic/displaylink-debian/pull/175/files

Basically, that's the revert of what you already asked me to do before. I can pull but looking at the operation, you have not changed anything so there is no reason to see it working. Do you want me to still do it, film me and post it on youtube? Maybe that would help see you how the screen freeze.

AdnanHodzic, 2018-11-24:
Sorry, I got mixed up with a different issue I saw your name in.

Let's start from the beginning, can you please verify that you went through [post installation guide](https://github.com/AdnanHodzic/displaylink-debian)? 

To be 100% if possible I would like if you could "Uninstall" the driver reboot, run "Install" again reboot and see if you still have the same problem.

Please paste output of whole "Install" process here.

kopax, 2018-11-24:
> Let's start from the beginning, can you please verify that you went through post installation guide?

Yes I did a few times.

> To be 100% if possible I would like if you could "Uninstall" the driver reboot, run "Install" again reboot and see if you still have the same problem.

1. Sure, I've updated my branch:

```
From github.com:AdnanHodzic/displaylink-debian
   a236153..56b72e6  master     -> origin/master
```

1. I've stopped the `dlm.service` with `systemctl stop dlm.service`. 
1. Run the script as root and choose **Uninstall**, I got `Uninstall complete` message
1. Reboot
1. Run the script as root and choose **Install** (log below)

```bash
root@dev-12:/home/dka/workspace/github.com/AdnanHodzic/displaylink-debian# ./displaylink-debian.sh 

--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux
* Deepin

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.9.0-8-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement
available at https://www.displaylink.com/downloads/file?id=1261
and accept here: [Y]es or [N]o: y

Downloading DisplayLink Ubuntu driver:

--2018-11-24 22:08:26--  https://www.displaylink.com/downloads/file?id=1261
Resolving www.displaylink.com (www.displaylink.com)... 52.208.143.240, 34.252.223.12, 54.72.7.162, ...
Connecting to www.displaylink.com (www.displaylink.com)|52.208.143.240|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: http://assets.displaylink.com/live/downloads/software/f1261_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%204.4.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1543072882&Signature=n7BbLe9VuKiz51JXY4YsN31bPxU%3D [following]
--2018-11-24 22:08:28--  http://assets.displaylink.com/live/downloads/software/f1261_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%204.4.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1543072882&Signature=n7BbLe9VuKiz51JXY4YsN31bPxU%3D
Resolving assets.displaylink.com (assets.displaylink.com)... 52.218.80.194
Connecting to assets.displaylink.com (assets.displaylink.com)|52.218.80.194|:80... connected.
HTTP request sent, awaiting response... 200 OK
[... 135 lines cut ...]
[   76.246320] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3
[   77.013113] evdi: [D] evdi_detect:76 (dev=1) Painter is connected
[   77.013121] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
[  131.684563] evdi: [D] evdi_detect:76 (dev=1) Painter is connected
[  131.684567] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
[  171.378376] evdi: [D] evdi_detect:76 (dev=1) Painter is connected
[  171.378378] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
[  187.156327] evdi: [D] evdi_painter_mode_changed_notify:507 (dev=1) Notifying mode changed: 1366x768@60; bpp 32; 
[  187.156331] evdi: [D] evdi_painter_mode_changed_notify:508 pixel format 875713112
[  187.156346] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 0
[  187.381156] evdi: [D] evdi_detect:76 (dev=1) Painter is connected
[  187.381167] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
[  204.872930] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3
[  205.183933] evdi: [D] evdi_detect:76 (dev=1) Painter is connected
[  205.183935] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
[  214.685089] evdi: [D] evdi_painter_mode_changed_notify:507 (dev=1) Notifying mode changed: 1366x768@60; bpp 32; 
[  214.685093] evdi: [D] evdi_painter_mode_changed_notify:508 pixel format 875713112
[  214.685141] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 0
[  214.685146] evdi: [D] evdi_painter_dpms_notify:470 (dev=1) Notifying dpms mode: 0
[  214.780702] evdi: [D] evdi_detect:76 (dev=1) Painter is connected
[  214.780704] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
[  225.778808] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3
[  226.029311] evdi: [D] evdi_detect:76 (dev=1) Painter is connected
[  226.029313] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
```

AdnanHodzic, 2018-11-24:
1. I remember your graphics card was i915. Because if you look at latest master: https://github.com/AdnanHodzic/displaylink-debian/blob/master/displaylink-debian.sh#L304

Your `/etc/X11/xorg.conf.d/20-displaylink.conf` file should contain following:

```
Section "Device"
  Identifier  "Intel"
  Driver      "intel"
EndSection

```
Can you confirm this is the case? If it is, what results do you get with this being as part of  `20-displaylink.conf`?

2. What were the contents of  `/etc/X11/xorg.conf.d/20-displaylink.conf` in that Youtube video?

3. Are changes better or worse with latest changes from master?

kopax, 2018-11-24:
> I remember your graphics card was i915. Because if you look at latest master: https://github.com/AdnanHodzic/displaylink-debian/blob/master/displaylink-debian.sh#L304

I have seen that merge, here is the content of `20-displaylink.conf`:

```bash
root@dev-12:/home/dka# cat /etc/X11/xorg.conf.d/20-displaylink.conf
Section "Device"
  Identifier  "DisplayLink"
  Driver      "modesetting"
  Option      "PageFlip" "false"
EndSection
```

1. No it is different content, I am trying with the expected.
2. The same as the one posted in this comment
3. Test are in progress

NoNeedForNames, 2018-11-24:
Wait, you ran the script and it didn't create the 20-displaylink.conf at all?

kopax, 2018-11-24:
It was here. Just wrong driver.

NoNeedForNames, 2018-11-24:
sorry for not letting this go, but which version are you using?
the "modesetting" or the "intel"-driver version?
and which one did the script install?

kopax, 2018-11-24:
- I am using the intel driver version
- the script installed the mode setting driver version

kopax, 2018-11-24:
Sorry for asking you that, but we're on Linux, do you know a way to have it ON on boot ?

After each reboot I must do: 

```bash
$ root@dev-12:/home/dka# xrandr --setprovideroutputsource 1 0
```

Then manually do:

- **Open Menu**
- **Type search for `Display`**,
- **Open Display**
- **Select my screen**
- **Toggle `ON`**
- **Positionate my screen appropriatly**
- **Click on Apply**

AdnanHodzic, 2018-11-24:
@kopax awesome, glad it's working as expected now. Spread the word :)

Regarding having to configure displays each time this is a problem with hotplug on Debian and Kali Linux, and you can adress this by following [Automated (persistent) display configuration](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#automated-persistent-display-configuration)

@NoNeedForNames I looked into code again as part of [last commit I fixed the problem](https://github.com/AdnanHodzic/displaylink-debian/commit/1345f8afd50da00d4deb34ff20f4fb1788995850) I addressed the issue.

Problem was that check (`if ([ "$drv" == "i915" ] && [ "$cardsub" == "530" ]);`) was too specific.

Even if `lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'` returned `i915`, it wouldn't matter because in conjunction with `lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5` which if didn't return `530` it wouldn never validate as true. 

Reason was `&&` which means both conditions have to be true. That's why "intel" part was never used, and it always went for "modesetting" driver.

So in the end I removed `cardsub` part as it's too specific, for example on Lenovo X1 Carbon which I'm typing this on, although  `drv` would return `i915`, `cardsub` part would return `Core`. Which would also go for "modesetting" driver although it should've went for "intel".

Either way, I should've noticed this as part of your second MR but I missed it, it happens :)

kopax, 2018-11-24:
@AdnanHodzic Happy to help to fix bugs. I'm also interested to read that you are using a Lenovo X1 Carbon. Which generation do you use?

I have myself a X1 Carbon Lenovo with Debian Stretch, it is my main working tool and I often travel, that's why I have a USB screen at the first place. 
Since I am notified from every change on GitHub, I have tried many times to reinstall your driver. I was still not able to use it with my Lenovo X1 Carbon, even after formatting my OS. 

Do you mind if I re-open this and to have a second look if I go through the same diag?

NoNeedForNames, 2018-11-24:
I thought this was the plan. in my first commit i was too wide (only grepping for not beeing the i915 driver). and for some users with different intel-graphics than mine (HD 530), the script didn't work properly, so i added a second statement "530" to exclude the other intel-graphics which had problems with my script-version... was there a mistake then in my first commit i didn't see?

AdnanHodzic, 2018-11-24:
@kopax Let's close this story as it's getting out of its original scope. Could you please open new story, i.e: "Doesn't work on Debian Stretch" and just paste install output, and as first two steps of [post installation](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md) guide suggest, make sure you have UEFI/secure boot disabled and you're logging in to X server and not Wayland. Because IIRC Wayland has been set as default in Stretch. Also include which kernel/X version you're using.

DisplayLink works fine on my Lenovo X1 Carbon which I don't use that often as I use for private use. For work I use [MacBook Pro (2017 with TouchBar) on which I also run Ubuntu](https://foolcontrol.org/?p=2952) :)

@NoNeedForNames right, I honestly don't remember, but here's an idea. If you wanted and still want to exclude `530` you can put it back, and just use something like `if ([ "$drv" == "i915" ] && [ "$cardsub" != "530" ]);` or maybe even add a seperate `elif` block which would explicitly say what do in case ` [ "$cardsub" != "530" ]`

I apologize for confusion.

kopax, 2018-11-24:
Ok @AdnanHodzic I close this and will open a new issue. Thanks in advance.

closed 2018-11-24 by kopax

AdnanHodzic, 2018-12-04:
@kopax @NoNeedForNames JFYI, In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

AdnanHodzic, 2019-01-30:
@kopax I remember a [comment you made about saving $100](https://github.com/AdnanHodzic/displaylink-debian/issues/172#issuecomment-441384936), so I've just added [project donate button](https://github.com/AdnanHodzic/displaylink-debian/blob/master/README.md#donate-paypal) in case you or anybody else is feeling like contributing :)

stratemeyerjw, 2019-09-23:
In case anyone else finds this useful. 
I got my AOC usb monitor to work near perfect after many hours of futzing with the lag/choppiness. 
System is -
   Lenovo E545, 
   Debian 10 (testing, Mate, updated), 
   Kernel 5.2.0-2-amd64
   AMD A10 with Radeon APU
Monitor is - (lsusb) DisplayLink E1659Fwu

1. used the displaylink debian installer
2. changed /etc/X11/xorg.conf.d/20-displaylink.conf
             Section "OutputClass"
                   Identifier  "DisplayLink"
                   #    MatchDriver "evdi"
                   #    Driver      "modesetting"
                   Driver "amdgpu"
                   Option  "AccelMethod" "none"
                   Option "Tiling" "true"
                   Option "Pageflip" "true"
                   Option "ExaNoComposite" "false"
             EndSection
3. run at login as user (or make bash script to do so for you). 
         xrandr --setprovideroutputsource 1 0 && mate-display-properties

hope this helps someone with this!

jl2035, 2020-07-25:
> Wait, you ran the script and it didn't create the 20-displaylink.conf at all?

@NoNeedForNames that happened to me on 20.04. I even tried to manually create a directory /etc/X11/xorg.conf.d and ran the installation again, but no success.