AdnanHodzic/displaylink-debian #173: Debian Testing - XPS13  - kernel 4.18.0-2-amd64

nllz, 2018-11-20:
Thanks a lot for this great work. Unfortunately I have some issues getting this to work.

My setup:

Dell XPS13 -  HP 3005pr
```
$ cat /proc/version
Linux version 4.18.0-2-amd64 (debian-kernel@lists.debian.org) (gcc version 7.3.0 (Debian 7.3.0-30)) #1 SMP Debian 4.18.10-2 (2018-11-02)
```
Installation went fine, and process is running:
```
$ systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enab
   Active: active (running) since Tue 2018-11-20 13:49:22 CET; 1h 15min ago
  Process: 712 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/4.4.2
 Main PID: 716 (DisplayLinkMana)
    Tasks: 35 (limit: 4915)
   Memory: 36.4M
   CGroup: /system.slice/dlm.service
           └─716 /opt/displaylink/DisplayLinkManager

```
but unortunately I get:
```
$ xrandr --listproviders
Providers: number : 0

```
Which probably has to do with (upon unplug-replug):

```
$ sudo dmesg | tail
[sudo] password for ****: 
[ 4401.661577] cdc_ncm 2-1.1:1.5 enx001fb532d0cf: renamed from usb0
[ 4401.684671] IPv6: ADDRCONF(NETDEV_UP): enx001fb532d0cf: link is not ready
[ 4401.684765] IPv6: ADDRCONF(NETDEV_UP): enx001fb532d0cf: link is not ready
[ 4402.185860] evdi: [D] evdi_painter_connect:574 (dev=1) Connected with 00000000ca4d71ad
[ 4402.185866] evdi: [D] evdi_detect:76 (dev=1) Painter is connected
[ 4402.185945] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3
[ 4402.239758] evdi: [D] evdi_painter_connect:574 (dev=2) Connected with 000000005c64ce21
[ 4402.239761] evdi: [D] evdi_detect:76 (dev=2) Painter is connected
[ 4402.239786] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=2) Notifying crtc state: 3
[ 4404.706868] cdc_ncm 2-1.1:1.5 enx001fb532d0cf: network connection: disconnected
```
All pointers would be very welcome. Thanks in advance.

kopax, 2018-11-20:
Could you try another USB port / reboot ? I got the same results and after doing that I was able to see 2 providers.

nllz, 2018-11-20:
I also tried re-plugging and running
`$ sudo systemctl restart dlm.service`

kopax, 2018-11-20:
@AdnanHodzic recommended here to try with : https://github.com/AdnanHodzic/displaylink-debian/issues/172#issuecomment-440173405 and to post your debian version `uname -a`

nllz, 2018-11-20:
working! thanks for all the fish!

closed 2018-11-20 by nllz

kopax, 2018-11-20:
Hey, do you mind to give some details for others ? Mine isn't working. What do you mean it work ? What is your version of Debian ? What screen are you using.

Thanks to you too.

nllz, 2018-11-20:
Heya,

This is what I am on:
```
$ cat /proc/version
Linux version 4.18.0-2-amd64 (debian-kernel@lists.debian.org) (gcc version 7.3.0 (Debian 7.3.0-30)) #1 SMP Debian 4.18.10-2 (2018-11-02)

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux testing (buster)
Release:	testing
Codename:	buster
```
As you suggested I did this:

`$ sudo mv /etc/X11/xorg.conf.d/20-displaylink.conf ~/20-displaylink.conf`

and when I then logged back into my xsession (rebooting or loggin out and in again) I got the following output:

```
$ xrandr --listproviders
Providers: number : 3
Provider 0: id: 0xa9 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:modesetting
Provider 1: id: 0x7a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

```

Hope this helps!

nllz, 2018-11-20:
Make sure that when you login, you are selecting from X11, not Gnome.

Click on the cogwheel drop-down menu next to the login form and select X11.

kopax, 2018-11-20:
I don't have such kind of form on cinnamon when I login, but still, thanks a lot for the clarification.
I am confident that this will help others. @AdnanHodzic I am curious why did you recommend to move that file.

Can we know more about that fix?

What is it supposed to fix exactly? What's expected?

@nllz seems to have a different behaviour than mine, when is it appropriate to do that? 

Also, it may be because of cinnamon that I can't display using my screen don't you think?

nllz, 2018-11-20:
@kopax Are you sure you are on Debian and not on Linux Mint?

If you use Cinnamon I think you are using Wayland as display server, and DisplayLink currently only works on X11 afaik.

AdnanHodzic, 2018-11-21:
@kopax reason I recommended to remove that file is after last [pull request](https://github.com/AdnanHodzic/displaylink-debian/pull/171/commits/87744f38c166a5926e8dab2ee416495d77663a31) by @NoNeedForNames. I realized that adding this won't work for everybody, but I merged it with master anyways, as I wanted to see how it plays out. 

In meantime I also wanted to add a new implementation which would add this file only if it's needed, but didn't have time to work on it yet.

Can you both (@nllz @kopax) please tell me what paste back the outputs of both:

`lspci | grep -i vga` 
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`

nllz, 2018-11-21:
```
$ lspci | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 620 (rev 02)
```
```
$ lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
i915
```

kopax, 2018-11-22:
```
$ lspci | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation Device 5912 (rev 04)
```

```
$ lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
i915
```

kopax, 2018-11-24:
@AdnanHodzic thanks, I've reinstalled and again, the screen is displaying, the mouse can move but can't add window to that screen. It is still "frozen" but working.

How can I help troubleshoot?

AdnanHodzic, 2018-11-24:
You could try by [Monitoring for errors](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#monitoring-for-errors)

You could also look into these, as they are Debian related problems, which sound a lot like yours, maybe you can find some useful information: 
https://github.com/AdnanHodzic/displaylink-debian/issues/68
https://github.com/AdnanHodzic/displaylink-debian/issues/130

NoNeedForNames, 2018-11-24:
@kopax, your behavior is exactly the reason why I have built in the driver-statement in the first place. It was exactly the same behavior on my  setup. What's the content of your etc/X11/xorg.conf.d/20-displaylink.conf ?

NoNeedForNames, 2018-11-24:
@nllz which one of the driver-versions in 20-displaylink.conf is working for you ?

AdnanHodzic, 2018-11-24:
@nllz ok problem has been [fixed as part of different issue](https://github.com/AdnanHodzic/displaylink-debian/issues/172#issuecomment-441396956). 

Please pull the latest changes and run "reinstall".

nllz, 2018-11-25:
Everything is working for me*, which is why I closed the thread :)

see: https://github.com/AdnanHodzic/displaylink-debian/issues/173#issuecomment-440304817