AdnanHodzic/displaylink-debian #174: Ubuntu 18.04 - Need to reinstall every time I reboot

YannGrosjean, 2018-11-23:
Hey, 
First, thank for your work. This script is pretty awesome. 

I just have a problem. My screen is now detected and work fine (an ASUS MB169B+), but every time I reboot my computer the screen stay black. 

xrandr didn't detect it, and the only think I could to to make it work again is to uninstall the driver, reboot my computer and then reinstall it....

Someone is experiencing the same issue ? 

Regards.

YannGrosjean, 2018-11-24:
Ok, I come with update. 

If I disconnect my screen, reboot, logging and then replug the screen it work well.

Very strange ...

AdnanHodzic, 2018-11-24:
Ok, could you please pull the latest changes and run "reinstall". If problem persists, could you also try connecting displaylink to different port?

YannGrosjean, 2018-11-27:
I tried both of this solution but no one work. Every time I reboot, I have to be sure than the screen is unplugged.

AdnanHodzic, 2018-11-29:
Maybe you should simply setup [Automated (persistent) display configuration](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#automated-persistent-display-configuration) as workaround. 

This is necessary to do on Debian and Kali, but maybe for whatever reason you have same issue with hotplug.

AdnanHodzic, 2018-12-04:
I'm closing this issue for inactivity. 

Please re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2018-12-04 by AdnanHodzic