AdnanHodzic/displaylink-debian #176: Debian 9.6 Crash boot after Displaylink installing

igoricegit, 2018-11-25:
Hi Adnan!

I tried to install DL on my Debian. And after reboot I can not load OS. It freeze when I have my ASUS plugged in while booting. If I unplug it I see
SOMETEXT tty1
compname login:

And I can login only via command line. 
When I uninstall it everything works fine and I can see my visual environment.

I also tried to create config files under X server. It gives the same effect as installing Displaylink. It so does not work with nvidia drivers installed and does not installed.

Some info
OS: Debian 9.6
Desktop: KDE Plasma
Computer type: PC Desktop
Graphic: intel Intel UHD Graphics 630 + nVidia GTX 1060 
MB: ASUS z370-I

Please provide any suggestions.
Thanks!

igoricegit, 2018-11-25:
Also UEFI is turned off

igoricegit, 2018-11-25:
Monitor is ASUS MB169C+

kopax, 2018-11-25:
Can you post the logs for all theses commands : 
```
$ dmesg | grep Display 
$ dmesg | grep evdi
$ lscpu
$ lspci -nnk | grep -i vga
$ uname -a
$ lsb_release -a
$ systemctl status dlm.service
$ cat /etc/X11/xorg.conf.d/20-displaylink.conf 
```

igoricegit, 2018-11-25:
Hello Kopax!
Please see attached

# dmesg | grep Display 
[    2.892168] usb 2-6.2: Manufacturer: DisplayLink
[    5.600268] usb 2-6.2: Manufacturer: DisplayLink

# dmesg | grep evdi
[    2.599978] evdi: loading out-of-tree module taints kernel.
[    2.600690] evdi: [I] Initialising logging on level 5
[    2.600691] evdi: [I] Atomic driver:no
[    5.981516] evdi: [D] evdi_add_devices:134 Increasing device count to 1
[    5.982074] evdi: [D] evdi_crtc_init:353 drm_crtc_init: 0
[    5.982385] evdi: [W] evdi_painter_crtc_state_notify:486 Painter does not exist!
[    5.982390] evdi: [D] evdi_detect:79 Painter is disconnected
[    5.982397] evdi evdi.0: No connectors reported connected with modes
[    5.985686] evdi evdi.0: fb0: evdidrmfb frame buffer device
[    6.019656] [drm] evdi: evdi_stats_init
[    6.019673] [drm] Initialized evdi 1.5.1 20180731 on minor 0
[    6.019800] evdi: [W] evdi_painter_disconnect:595 (dev=0) An unknown connection to ffff96588d54f200 tries to close us
[    6.019804] evdi: [W] evdi_painter_disconnect:596  - ignoring
[    6.020412] evdi: [W] evdi_painter_disconnect:595 (dev=0) An unknown connection to ffff96588d54f200 tries to close us
[    6.020414] evdi: [W] evdi_painter_disconnect:596  - ignoring
[    6.038302] evdi: [D] evdi_detect:79 Painter is disconnected
[    6.038371] evdi: [D] evdi_detect:79 Painter is disconnected
[    6.038653] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=0) Notifying crtc state: 3
[    6.038654] evdi: [W] evdi_painter_send_crtc_state:389 Painter is not connected!
[    6.039897] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=0) Notifying crtc state: 3
[    6.039898] evdi: [W] evdi_painter_send_crtc_state:389 Painter is not connected!
[    7.050821] evdi: [D] evdi_detect:79 Painter is disconnected
[   11.542805] evdi: [D] evdi_painter_connect:574 (dev=0) Connected with ffff965870daea00
[   11.542807] evdi: [D] evdi_detect:76 (dev=0) Painter is connected
[   11.542827] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=0) Notifying crtc state: 3
[   11.544696] evdi: [D] evdi_detect:76 (dev=0) Painter is connected
[   11.544698] evdi: [D] evdi_painter_get_edid_copy:217 (dev=0) 00 ff ff

# lscpu
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                12
On-line CPU(s) list:   0-11
Thread(s) per core:    2
Core(s) per socket:    6
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 158
Model name:            Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz
Stepping:              10
CPU MHz:               799.890
CPU max MHz:           4700.0000
CPU min MHz:           800.0000
BogoMIPS:              7392.00
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              12288K
NUMA node0 CPU(s):     0-11
Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch epb invpcid_single ssbd ibrs ibpb stibp kaiser tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp flush_l1d

# lspci -nnk | grep -i vga
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] [10de:1c03] (rev a1)

# uname -a
Linux sunderstorm 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux

# lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 9.6 (stretch)
Release:        9.6
Codename:       stretch

# systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-11-25 11:57:20 EET; 8min ago
  Process: 750 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/4.4.24 && mod
 Main PID: 753 (DisplayLinkMana)
    Tasks: 32 (limit: 4915)
   CGroup: /system.slice/dlm.service
           └─753 /opt/displaylink/DisplayLinkManager

Nov 25 11:57:20 sunderstorm systemd[1]: Starting DisplayLink Manager Service...
Nov 25 11:57:20 sunderstorm systemd[1]: Started DisplayLink Manager Service.

> cat /etc/X11/xorg.conf.d/20-displaylink.conf 

I removed any additional configuration files from xorg because if I add something related to displaylink I have the same situation with the boot problem.
I tried to use this recommendations
https://support.displaylink.com/knowledgebase/articles/1181623-displaylink-ubuntu-driver-after-recent-x-upgrades
https://github.com/DisplayLink/evdi/issues/133

Also as I said before I can run Debian only with Displaylink removed.

AdnanHodzic, 2018-11-25:
@igoricegit what about contents of `cat /etc/X11/xorg.conf.d/20-displaylink.conf `

igoricegit, 2018-11-25:
Hello Adanan!

I am very new in Linux. I tried only add this

/usr/share/X11/xorg.conf.d/20-displaylink.conf 
with the following content: 

Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection  

and this
/usr/share/X11/xorg.conf.d/evdidevice.conf

Section "OutputClass"
	Identifier "DisplayLink"
	MatchDriver "evdi"
	Driver "modesetting"
	Option "AccelMethod" "none"
EndSection

All provides the same effect

ungentilgarcon, 2018-11-30:
do you have a fix for this yet?

igoricegit, 2018-11-30:
No I have no idea how to. I spent 3 days for this issue and just use 2 other hdmi and displayport monitors.

ungentilgarcon, 2018-11-30:
Ok, it works fine for me using only 
/etc/X11/xorg.conf.d/20-displaylink.conf
Section "Device"
Identifier "DisplayLink"
EndSection

no  /usr/share/X11/xorg.conf.d/evdidevice.conf

igoricegit, 2018-11-30:
tried one by one and 2 config files together. Or you think Driver "modesetting" can crash everything?

ungentilgarcon, 2018-11-30:
:)) oh and nomodeset  (though I don't know if it does something here....
I'm on debian stretch, kernel 4.18 with nvidia graphics drivers, using a tegra adapter.
no tearing or else

ungentilgarcon, 2018-11-30:
yep , I first commented page flip, NO GO
then modesetting =>GOOOOD

ungentilgarcon, 2018-11-30:
Try it, 
oh, and BIG THANKS to adnan, 
now my xen issues ;)

igoricegit, 2018-11-30:
Hello Guys!
What I tried.

1 I have created file /etc/X11/xorg.conf.d/20-displaylink.conf
with 

Section "Device"
Identifier "DisplayLink"
EndSection

on board ad installed Display Link.

Everything loads fine but monitor still not work. I tried to unplug and plug it. also turn off unplug - turn on and plug. It loads without any problems but display does not wokk((

ungentilgarcon, 2018-11-30:
Do you have Arandr?

igoricegit, 2018-11-30:
No Do you mean it can help?

AdnanHodzic, 2018-12-01:
@ungentilgarcon most welcome, btw could you please tell me what's the output of this command for you?

`lspci -nnk | grep -i vga`

Please correct me if I'm wrong, but having only this in: /etc/X11/xorg.conf.d/20-displaylink.conf

```
Section "Device"
  Identifier "DisplayLink"
EndSection
```

and removing `/usr/share/X11/xorg.conf.d/evdidevice.conf`

Work great for you? If this is correct I would like to incorporate these changes to be made when displaylink-debian detects user has an Nvidia card.

@igoricegit yes, you should definitely try and see what happens with `arandr`, for some people it's like easier to do using GUI tools, please see ["Screen Layout Configuration" part for more details.](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#screen-layout-configuration) 

Also could you confirm what I wrote above for @ungentilgarcon also worked for you?

ungentilgarcon, 2018-12-01:
@adnan Yes, that's the way to go with my nvidia at least, with debian stretch and quite many backports (like kernel 4.18) and testing packages -apt pinning.

@igoricegit green is great ;), now you have to run : 
xrandr --setprovideroutputsource 1 0 
(or 2 0 instead of 1 0, or 2 1...depending on your config)

and install arandr because it is great and easy, then you should be able to activate the screens.

ungentilgarcon, 2018-12-01:
@AdnanHodzic  lspci output
0a:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] [10de:1c03] (rev a1)

AdnanHodzic, 2018-12-02:
@ungentilgarcon I'm about to changes to `displaylink-debian` could you please tell me what the output of `lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'` is? Thanks!

AdnanHodzic, 2018-12-02:
What happens if you add this as your: `/etc/X11/xorg.conf.d/20-displaylink.conf` and restart X/logout > login

```
Section "Device"
  Identifier  "DisplayLink"
  Driver      "modesetting"
  Option      "PageFlip" "false"
EndSection
```

If your `lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'` output is nvidia, what is the output of: `lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`?

igoricegit, 2018-12-02:
Hello!
After I reload PC s shows this  screen
[image]
I used Debian Recovery mode and freeze. Next time I reloaded PC It loads fine.

$ lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5
GP106

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

igoricegit, 2018-12-04:
Hello!
Thank you for the tips.

I have completely uninstantiated DL with conf file and reinstalled it again from new repository.

System loads fine but monitor still black ((

igorice@sunderstorm:~/Downloads/displaylink-debian$ cat /etc/X11/xorg.conf.d/20-displaylink.conf
Section "Device"
  Identifier "DisplayLink"
EndSection

igorice@sunderstorm:~/Downloads/displaylink-debian$ lspci | grep -i vga
01:00.0 VGA compatible controller: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] (rev a1)

igorice@sunderstorm:~/Downloads/displaylink-debian$ lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
nvidia

igorice@sunderstorm:~/Downloads/displaylink-debian$ lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5GP106

igorice@sunderstorm:~/Downloads/displaylink-debian$

igoricegit, 2018-12-04:
Also I can not reopen issue

AdnanHodzic, 2018-12-04:
What happens if you add following to `/etc/X11/xorg.conf.d/20-displaylink.conf` file:

```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

logout and re-login?

igoricegit, 2018-12-04:
Nothing. Still black and stand by mode

AdnanHodzic, 2018-12-29:
@igoricegit sorry for such a delayed reply, but I just released a new debug feature, with that said could you include the output of: `sudo ./displaylink-debian.sh --debug`

igoricegit, 2019-01-05:
Hello Adnan! No problems. Thank you for your work. Please find the output 

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: stretch
Kernel: 4.9.0-8-amd64

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24
                                                                                                             
------------------ Graphics card -----------------                                                           
                                                                                                             
Vendor: nvidia                                                                                               
Subsystem: GP106                                                                                             
VGA: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] (rev a1)

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x1b9 cap: 0x1, Source Output crtcs: 4 outputs: 7 associated providers: 0 name:NVIDIA-0
Provider 1: id: 0x21b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

AdnanHodzic, 2019-01-05:
I'm pretty sure the problem is with the contents of `/etc/X11/xorg.conf.d/20-displaylink.conf` file. From what I know it's supposed to be `modesetting` for Nvidia cards ,but I could be wrong as I don't have an Nvidia card and no proper way of testing this.

From what I found on Google, apparently some people have both Intel and Nvidia X11 configs, or it's a problem with Nvidia driver, i.e:

- https://ubuntuforums.org/showthread.php?t=2391150
- https://devtalk.nvidia.com/default/topic/1033140/kubuntu-18-04-gives-black-screen-when-sddm-starts-no-login-/?offset=6

I know last user who reported he had Nvidia managed to sort it out himself, but I'm not sure what he used for as his xorg.config. Feel free to ask him: https://github.com/AdnanHodzic/displaylink-debian/issues/188

I also of this other case but IIRC user switched to an Intel card: https://github.com/AdnanHodzic/displaylink-debian/issues/179

Btw, there's lot more [Nvidia mentions under issues.](https://github.com/AdnanHodzic/displaylink-debian/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aall+nvidia). Either way let me know if you come across anything useful, so I can add it to the tool itself so it works out of box.

AdnanHodzic, 2019-01-06:
@igoricegit out of sheer curiosity what happens if you set following as contents for: `/etc/X11/xorg.conf.d/20-displaylink.conf` followed by logout/login or reboot.

```
Section "Device"
	Identifier "DisplayLink"
	MatchDriver "evdi"
	Driver "modesetting"
	Option  "AccelMethod" "none"
EndSection
```

igoricegit, 2019-01-06:
Thank you. It crashes the system. I can login into command line.

AdnanHodzic, 2019-01-06:
ok and what happens if you tried something like this:

```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

Section "Device"
    Identifier     "nvidia"
    Driver         "nvidia"
    VendorName     "NVIDIA Corporation"
    Option         "AllowEmptyInitialConfiguration"
EndSection
```

for `/etc/X11/xorg.conf.d/20-displaylink.conf`

Also what do you get when you run `dpkg -l | grep "ii  xserver-xorg-core" | awk '{print $3}' | sed 's/[^,:]*://g'` (I'm about to add this as an extra option for debug)

igoricegit, 2019-01-06:
Hello Adnan!
I tried to do some things and returend to the beginning. I am very new in Linux maybe somewhere is really my fault. So what I have now.

During my investigation I found that I have 2 config files on my PC:

/etc/X11/xorg.conf.d/20-displaylink.conf

Section "Device"
Identifier "DisplayLink"
Driver "modesetting"
Option "PageFlip" "false"
EndSection

and this
/usr/share/X11/xorg.conf.d/20-displaylink.conf

Section "Device"
Identifier "DisplayLink"
EndSection

Looks like this is the same configuration. And conf file on **/usr/share/X11/xorg.conf.d/** has higher priority then **/etc/X11/xorg.conf.d/**.

I think that is why I had no any chages. So I tried to remove one file and use different configurations.

so when I get.

If I have one file and does not care where with next lines:
Section "Device"
Identifier "DisplayLink"
EndSection

Nothing happens but I can load the system.

If I have more additional lines like **Driver "modesetting"** my systeem can not load. 

But when I have portable screen connected it shows 
SOMETEXT tty1
compname login:

and is freezed. Only helps CTR+C

If device is unplugged I see the same but can login via command line and remove 20-displaylink.conf file and reload the system. After this system loads fine.

FYI
I have totally 3 monitors
1 DP via nvidia
2 HDMI via nvidia
3 Portable screen via USB-C MotherBoard based

I tried to remove one monitor. It provides the same results.

Also I changed priority of device loading to modesetting via xrandr --setprovideroutputsource

Thank you!

AdnanHodzic, 2019-01-06:
Okay you did good, we definitely want to have only one `xorg.conf` which we'll use. But I think in your case there even might be a third xorg.conf file which is: `/etc/x11/xorg.conf`. This one is generated by `nvidia-xsetup` command which is there from the driver itself, I assume you installed the driver by installing `nvidia-driver` package, correct me if I'm wrong.

Regardless, if it's there, make sure take a backup of it first, i.e: `sudo cp /etc/x11/xorg.conf /etc/x11/xorg.conf.org.bak`

Also make sure to temporary remove even the only xorg.conf that you had, i.e: `sudo mv /usr/share/X11/xorg.conf.d/20-displaylink.conf /home/your_username`. So that this time we truly use only a single xorg.conf.

Then, add following two lines (if they are not there) to the end of `/usr/share/sddm/scripts/Xsetup`

```
xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --auto
```

Whole file should look something like this:

```
#!/bin/sh
# Xsetup - run as root before the login dialog appears

if [ -e /sbin/prime-offload ]; then
    echo running NVIDIA Prime setup /sbin/prime-offload
    /sbin/prime-offload
fi

xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --auto
```

add following to: `/etc/X11/xorg.conf`

```
Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    BusID "PCI:0@0:2:0"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    BusID "PCI:1@0:0:0"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection
```

After this reboot and see what happens. If it doesn't work, try removing both occurrences of `BusID "PCI:0@0:2:0"` lines and restart again. 

In case of failure, restore your original xorg.conf by running: `sudo mv /etc/x11/xorg.conf.org.bak /etc/x11/xorg.conf` or re-create it by running `nvidia-xconfig` tool.

igoricegit, 2019-01-07:
[image]
Thank you Adnan!

Just did what you said to me and I see this.

Now I faced other problem. Now I have delays with updating elements on the screen. Looks like all works via Intel not sure. Because it works like on very slow graphicc card. And I see old textures sometimes near mouse.

Thanks!

AdnanHodzic, 2019-01-07:
Nice! That's awesome. 

With that said, I'm going to revamp how Nvidia configuration is added and push the code changes in one the next couple of days. So this is done automatically for you and anyone else with Nvidia cards.

I need a bit of your input before I do this, what would happen if moved contents of xorg.conf to 20-displaylink.conf, i.e:

```
sudo mv /etc/X11/xorg.conf /etc/X11/xorg.conf.d/20-displaylink.conf 
```

and restarted/logged out? Is it still working the same, as in code implementation I would like all of the DisplayLink stuff to have its own dedicated xorg.conf.

Second, in your xorg.conf did you leave in the `BusID` lines or you had them removed as well? If it's easier just paste the whole xorg.conf or 20-displaylink.conf if it worked out in first step.

igoricegit, 2019-01-07:
Sure I will do. But what to do with the slow. It is not comfort to operate the system? Where to dig with this?

Thanks!

AdnanHodzic, 2019-01-07:
Do this, for "Intel" part replace,

```
Section "Device"
    Identifier "intel"
    Driver "modesetting"
    BusID "PCI:0@0:2:0"
    Option "AccelMethod" "None"
EndSection
```

with:

```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

But then also update this section:

```
Section "Screen"
    Identifier "DisplayLink"
    Device "DisplayLink"
EndSection
```

Reboot, then see if works any better, if not also try try removing `Option "ConstrainCursor" "off"` out of:

```
Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    BusID "PCI:1@0:0:0"
    Option "ConstrainCursor" "off"
EndSection
```

and reboot.

If you had it, what are the contents of: `/etc/x11/xorg.conf.org.bak` file?

igoricegit, 2019-01-07:
Hello Adnan!
Thank you for your works. Lets share my investigation.

About lust message.
It is crash and freeze during the loading.

I found the issue. There is  the problems with Intel and OpenGL rendering of Desktop environment. The issue is square artifacts around the mouse. There are 2 solutions:
1st - after loading restard sddm `sudo systemctl restart sddm` but you need to do this each time.
2d change rendering from OpenGL to xRender

I have updated nvidia driver to 390.xxx but still the same. And this problem with portable monitor enabled ((

About your questions:
1 sudo mv /etc/X11/xorg.conf /etc/X11/xorg.conf.d/20-displaylink.conf 
works perfect I removed xorg.conf totally.
Also I removed next lines:
all BusID
Option "ConstrainCursor" "off"
was no any changes.

So want to say many thanks for your time. Only still want to return to OpenGL rendering because it works better.

AdnanHodzic, 2019-01-07:
In regards to my questions, thanks I know everything I need to know to make these changes in code now.

You're welcome and I'm glad you managed to figure it by changing renders from OpenGL to xRender. I believe this is related to Nvidia itself, so just try to Google it, answer has to be somewhere. If you find the problem to be related to X11 let me know and I'll make additional changes.

I'll close the issue once I've pushed the code changes, after I do this I'll ask you do "re-install" just to confirm everything still works as expected.

igoricegit, 2019-01-07:
this crush

Please update and I will reinstall

about cursor issue there is very well known bug with nvidia driver. so many people have this bug. the solution to make logout/login

AdnanHodzic, 2019-01-07:
Okay, changes pushed. Pull the latest changes and run re-install where all changes necessary for Nvidia will happen automatically.

I'm closing the issue, in case re-install breaks anything in your case, please re-open the issue and let me know what happened.

closed 2019-01-07 by AdnanHodzic

igoricegit, 2019-01-08:
Hello Adnan!
Thank you for your help! I have reinstalled driver all works fine. I have only 2 small bugs

1 Cursor lagging.
2 Sometimes display is black and when you reboot the system it works fine.

Any way it works! Many thanks again!

AdnanHodzic, 2019-01-09:
You're welcome.

Regarding cursor problem, another user had (maybe) similar problem and he managed to resolve it as part of: https://github.com/AdnanHodzic/displaylink-debian/issues/192

Could you also give it a try? In case it works it should probably be added to: [Troubleshooting most common issues](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#troubleshooting-most-common-issues) 

In case you removed it, maybe you should put `Option "ConstrainCursor" "off"` line back as part of your `/etc/X11/xorg.conf.d/20-displaylink.conf` nvidia section.

Regarding other problem with black screen, you should probably report this as an [upstream driver](https://github.com/DisplayLink/evdi) problem and if you look [under their issues](https://github.com/DisplayLink/evdi/issues) you'll find similar reports.

Since this original issue was solved here and it's really getting of the scope and I would like to keep it closed. If you continue to experience similar problems open a new issue and report a problem there.

igoricegit, 2019-01-09:
I have found some piece if code in /etc/X11/xorg.conf.d/20-displaylink.conf after i removed it dp works better

AdnanHodzic, 2019-01-09:
Which piece fixed it exactly? I want to make changes in tool itself.

igoricegit, 2019-01-10:
Hello Adnan!
Sorry forgot to make a screenshoot.

There was something like this 

Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    BusID "PCI:0@0:2:0"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    BusID "PCI:1@0:0:0"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection
**Section "Device"
Identifier "DisplayLink"
EndSection**

in file /etc/X11/xorg.conf.d/20-displaylink.conf 
Section "Device"
Identifier "DisplayLink"
EndSection

looks like some rudimentary piece of code.

AdnanHodzic, 2019-01-10:
Cool thanks, I've also removed it from code as part of last commit.

igoricegit, 2019-01-11:
I worked yesterday whole day on Linux and it is fine. Still has a problem with cursor but  one relogin fix it for one session.

Sometimes after reboot the display is dark and system loosing this remote monitor. But after rebbot it works fine. Like not unnoying bug.

igoricegit, 2019-01-12:
Hello Adnan!

Sorry for bothering you, but I have the same issue. I have crashed totally my debian and reinstalled it. after I have the same issue just dark screen. Next  I decided  to upgrade the kernel to 4.18 and used your nexgen-kernel. the same no luck (((

igorice@sunderstorm:~/Downloads/displaylink-debian$ sudo ./displaylink-debian.sh --debug                                                                            
                                                                                                                                                                    
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: stretch
Kernel: 4.18.0-0.bpo.3-amd64

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4:40
4.4.24
4:40
4:40

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: GP106
VGA: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] (rev a1)
X11 version: 1.19.2-1+deb9u5

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x1b9 cap: 0x1, Source Output crtcs: 4 outputs: 7 associated providers: 0 name:NVIDIA-0
Provider 1: id: 0x241 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)