AdnanHodzic/displaylink-debian #177: is not work en  Kali 2018.4

jokeralood, 2018-11-25:
help me pliz is not work in my kali linux 2018.4

AdnanHodzic, 2018-11-26:
@jokeralood what doesn't work, can you provide the output of what happens when you run "install"

jokeralood, 2018-11-26:
the instal install complet but after i reboot but the scren in display not work

AdnanHodzic, 2018-11-26:
what is the output of these 3 commands:

`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`cat /etc/X11/xorg.conf.d/20-displaylink.conf`

jokeralood, 2018-11-26:
# lspci | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 520 (rev 07)
# lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
i915
# cat /etc/X11/xorg.conf.d/20-displaylink.conf
Section "Device"
  Identifier  "Intel"
  Driver      "intel"
EndSection

jokeralood, 2018-11-26:
# echo $XDG_SESSION_TYPE
x11
:~# dmesg | grep Display
[    3.458304] usb 2-4.1: Manufacturer: DisplayLink

jokeralood, 2018-11-29:
help me plizz

AdnanHodzic, 2018-12-03:
@jokeralood ok please add following as your `/etc/X11/xorg.conf.d/20-displaylink.conf` contents:

```
Section "Device"
  Identifier  "Intel"
  Driver      "intel"
EndSection
```

and logout and re-login to X11 session and everything should be working as expected.

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic