AdnanHodzic/displaylink-debian #178: evdi module is not removed upon uninstall on Ubuntu 18.10

jphein, 2018-11-28:
I recently removed my Nvidia card from my computer. Since then this Display link does not work with the built in intel chipset. I'm getting a blank screen even though the modesetting tweak has been set to intel in the xorg conf file. 

When I run this script to uninstall displaylink drivers it says that it removes the evdi module, but in fact it's still loading on every boot. 
```sh
lsb_release --all
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.10
Release:	18.10
Codename:	cosmic
```

```sh
uname -r
4.18.0-11-lowlatency
```
```sh
sudo ./displaylink-debian.sh 

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: u

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.18.0-11-lowlatency is installed
dkms is installed
lsb-release is installed
linux-source is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

Removing evdi module

-------------------------------------------------------------------

Performing clean-up

-------------------------------------------------------------------

Uninstall complete

-------------------------------------------------------------------
```

```sh
reboot
```

```sh
lsmod|grep evdi
evdi                   53248  0
drm_kms_helper        172032  2 evdi,i915
drm                   454656  7 drm_kms_helper,evdi,i915
syscopyarea            16384  2 drm_kms_helper,evdi
sysfillrect            16384  2 drm_kms_helper,evdi
sysimgblt              16384  2 drm_kms_helper,evdi
```

jphein, 2018-11-28:
I had to run
```sh
sudo dkms remove evdi/4.4.24 --all
```

AdnanHodzic, 2018-11-29:
After you ran `sudo dkms remove evdi/4.4.24 --all` did it end up working or? I'm confused why you did that.

jphein, 2018-11-29:
I had to run that command to finish the uninstall. Isn't the script supposed uninstall the module?

AdnanHodzic, 2018-12-01:
Ok that's good to know, as part of "uninstall" procedure, `displaylink-debian` will remove kernel modules by running `rmmod evdi` and that seemed to be enough in most cases. But in next update I'll also include option to run `dkms remove evdi/$driver_version --all` to make sure it's all gone.

jphein, 2018-12-01:
Great!!! Thank you so much. =]

AdnanHodzic, 2018-12-31:
feature implemented as part of [last commit](https://github.com/AdnanHodzic/displaylink-debian/commit/9a219d5075ae1bd58577ea5c933794ed1cc3ca22)

closed 2018-12-31 by AdnanHodzic