AdnanHodzic/displaylink-debian #179: Black screen in Ubuntu 18.10 with Intel i915. It worked with Nouveau.

jphein, 2018-11-28:
Took out my Nvidia card, and now Display link just gives me a blank screen. Even with the new modesetting tweak being automatically set in xorg.conf. 

```sh
lsb_release --all
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.10
Release:	18.10
Codename:	cosmic
```

```sh
uname -r
4.18.0-11-lowlatency
```
```sh
lspci | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
```

```sh
lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
i915
```
```sh
sudo ./displaylink-debian.sh 

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.18.0-11-lowlatency is installed
dkms is installed
lsb-release is installed
linux-source is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement
available at https://www.displaylink.com/downloads/file?id=1261
and accept here: [Y]es or [N]o: y

Downloading DisplayLink Ubuntu driver:

--2018-11-27 18:05:13--  https://www.displaylink.com/downloads/file?id=1261
Resolving www.displaylink.com (www.displaylink.com)... 54.76.51.192, 52.208.143.240, 54.72.7.162, ...
Connecting to www.displaylink.com (www.displaylink.com)|54.76.51.192|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: http://assets.displaylink.com/live/downloads/software/f1261_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%204.4.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1543371493&Signature=BbhWADrQljC9Bxt%2BlyHx4YGI6Ps%3D [following]
--2018-11-27 18:05:14--  http://assets.displaylink.com/live/downloads/software/f1261_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%204.4.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1543371493&Signature=BbhWADrQljC9Bxt%2BlyHx4YGI6Ps%3D
Resolving assets.displaylink.com (assets.displaylink.com)... 52.218.104.122
Connecting to assets.displaylink.com (assets.displaylink.com)|52.218.104.122|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 17626360 (17M) [application/octet-stream]
Saving to: ‘DisplayLink_Ubuntu_4.4.zip’

DisplayLink_Ubuntu_4.4. 100%[=============================>]  16.81M   291KB/s    in 58s     

2018-11-27 18:06:13 (297 KB/s) - ‘DisplayLink_Ubuntu_4.4.zip’ saved [17626360/17626360]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_4.4.zip
  inflating: 4.4/LICENSE             
  inflating: 4.4/displaylink-driver-4.4.24.run  
Creating directory displaylink-driver-4.4.24
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 4.4.24  100%  

-------------------------------------------------------------------

Installing driver version: 4.4

DisplayLink Linux Software 4.4.24 install script called: install
Distribution discovered: Ubuntu 18.10
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS
Installing EVDI kernel module to kernel tree
EVDI kernel module built successfully
Installing x64-ubuntu-1604/DisplayLinkManager
Installing libraries
Installing firmware packages
Installing licence file
Adding udev rule for DisplayLink DL-3xxx/5xxx devices

Please read the FAQ
http://support.displaylink.com/knowledgebase/topics/103927-troubleshooting-ubuntu

-------------------------------------------------------------------

Performing post install steps

Disabling PageFlip for modesetting

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_4.4.zip" file
Removing redundant: "4.4" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes

-------------------------------------------------------------------
```
```sh
cat /etc/X11/xorg.conf.d/20-displaylink.conf
Section "Device"
  Identifier  "Intel"
  Driver      "intel"
EndSection
```
reboot

AdnanHodzic, 2018-11-29:
What would happen if you replaced existing: `/etc/X11/xorg.conf.d/20-displaylink.conf` contents with:

```
Section "Device"
  Identifier  "DisplayLink"
  Driver      "modesetting"
  Option      "PageFlip" "false"
EndSection
```

and restarted the X?

Also out of sheer curiosity, why are you using lowlatency kernel?

jphein, 2018-11-29:
I'll try it! =] 
I'm running the lowlatency kernel for recording audio in OBS. It does seem to reduce delay in my monitors.

AdnanHodzic, 2018-12-01:
Any updates? In case it doesn't work, I would be interested if you could run driver re-install in `displaylink-debian` on "generic" driver just to see if it would make any difference in terms of your monitor displaying any different results.

jphein, 2018-12-01:
Thank you for all your work! Everything is working with the changed Xorg conf file.  
I actually upgraded my computer. So it is now using the Intel 3rd gen chipset, instead of the 2nd gen my old computer had. 
```sh
lspci | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor Graphics Controller (rev 09)
```
I didn't have time to test the change on my old computer. I just swapped my SSD, so all software is the same. It's using the same driver:
```sh
lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
i915
```
I'll reboot, and see if the old Xorg settings give me trouble with the new computer.

jphein, 2018-12-01:
Yes! That fixed it. The old Xorg conf file displays only a black screen.
```sh
cat /etc/X11/xorg.conf.d/20-displaylink.conf
Section "Device"
  Identifier  "Intel"
  Driver      "intel"
EndSection
```

The new xorg conf file works perfectly!

```sh
cat /etc/X11/xorg.conf.d/20-displaylink.conf
Section "Device"
  Identifier  "DisplayLink"
  Driver      "modesetting"
  Option      "PageFlip" "false"
EndSection
```

Thank you again! =]

AdnanHodzic, 2018-12-02:
@jphein before Great! 

However, before I can make changes to the tool itself, can you confirm output you get from: `lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'` is i915?

If that's the case, can you also give me the output of following command: `lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

Thanks

jphein, 2018-12-02:
```sh
lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
i915
```
```sh
lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5
v2/3rd
```

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic