AdnanHodzic/displaylink-debian #181: Adjusting brightness

alejandrosame, 2018-12-04:
Hello!

I must say that out-of-the-box, the script worked wonderfully for my Ubuntu 18.04 laptop machine. 

However, I am not able to adjust the brightness of the external screen, neither through its physical buttons, nor does it share the brightness values of the main laptop screen (it always stays with the same brightness level).

Is there already a known workaround? I wouldn't mind trying to help fixing this issue for other users, but I'm clueless as to how to proceed with this. Any pointers to where to start looking would be also appreciated.

Thank you!

AdnanHodzic, 2018-12-04:
Glad to hear that.

Don't think this is related to `displaylink-debian`. 

Not sure which desktop environment you use but what would happen if you simply adjusted brightness through your DE brightness controls on that (external) desktop?

alejandrosame, 2018-12-05:
I see what you mean. I am using the standard Ubuntu 18.04 installation with Gnome3. In the power settings it actually only appears one slide bar for controlling the brightness settings, so it seems to not recognize that another screen exists for brightness purposes.

[image]

I will check what workaround exists out there.

Once again, thanks!

AdnanHodzic, 2018-12-29:
I was curious, could you include the output of:
`sudo ./displaylink-debian.sh --debug`

alejandrosame, 2019-01-12:
Hi again, I'm sorry that I missed this reply.

Here's the output that I get:
--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.15.0-43-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 09)
X11 version: 1.19.6-1ubuntu4.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x2fa cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 2 name:NVIDIA-0
Provider 1: id: 0x346 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x44 cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 2 associated providers: 1 name:modesetting

AdnanHodzic, 2019-01-30:
As part of https://github.com/AdnanHodzic/displaylink-debian/commit/b57cea89cc2e97de071e417bdd39ddd374072dbd commit udl/udlf will be added to blacklist.conf automatically if >= 4.14.9 is detected. 

I'm not sure if this will fix your problem, but I'd definitely advise you to pull the latest changes and perform reinstall and report back if there any changes.

AdnanHodzic, 2019-02-03:
In last few days I made a lot of changes in regards to automated NVIDIA card detection and configuration. 

Make sure to pull the latest changes and run "sudo ./displaylink-debian --reinstall". Let me know if this changes anything for you.

alejandrosame, 2019-02-12:
I made the reinstall but nothing changed regarding the brightness adjustment. However, I noticed that the slight cursor blink that I was experiencing went away :)

Here is the new debug output, just in case:

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.15.0-45-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 09)
VGA (3D): NVIDIA Corporation GK208M [GeForce GT 730M] (rev a1)
X11 version: 1.19.6-1ubuntu4.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x322 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 2 name:NVIDIA-0
Provider 1: id: 0xc2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x44 cap: 0x2, Sink Output crtcs: 3 outputs: 2 associated providers: 1 name:modesetting

AdnanHodzic, 2019-02-16:
This is definitely an upstream issue so feel free to report the problem there: https://github.com/DisplayLink/evdi/issues

Since the installer works just as it's supposed to, I'm going to close this issue. If you have any updates regarding this later on feel free to re-open this issue.

closed 2019-02-16 by AdnanHodzic