AdnanHodzic/displaylink-debian #182: Doesn't work on Ubuntu 18.04 with ast2400 bmc device

syuriks, 2018-12-04:
Hello Adnan,

I'm having problems bringing up DisplayLink device on the server having AST2400 device with embedded VGA controller. The monitor is not being detected. 
Please see relevant logs:

dmesg:

_[   23.787943] [drm] Initialized evdi 1.5.1 20180731 for evdi.0 on minor 1
[   23.787992] [drm] evdi: evdi_stats_init
[   23.795361] evdi: [D] evdi_detect:79 Painter is disconnected
[   23.822822] evdi: [D] evdi_painter_connect:574 (dev=1) Connected with 0000000086ed9b84
[   23.822825] evdi: [D] evdi_detect:76 (dev=1) Painter is connected
[   23.822848] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3
[   24.318334] igb 0000:0a:00.0 enp10s0: igb: enp10s0 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX/TX
[   24.318539] IPv6: ADDRCONF(NETDEV_CHANGE): enp10s0: link becomes ready
[   54.693053] evdi: [D] evdi_detect:76 (dev=1) Painter is connected
[   54.693057] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
[   54.693324] evdi: [D] evdi_detect:76 (dev=1) Painter is connected
[   54.693326] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
[   54.721822] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3
[   54.865576] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3_

_~$lspci  -v | grep VGA
09:00.0 VGA compatible controller: ASPEED Technology, Inc. ASPEED Graphics Family (rev 30) (prog-if 00 [VGA controller])

~$ xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x78 cap: 0x0 crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 1: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

~$xrandr --listmonitors
Monitors: 1
 0: +*VGA-1 1920/509x1080/286+0+0  VGA-1

~$ lsusb
Bus 001 Device 002: ID 8087:8000 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 17e9:4301 DisplayLink 
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 03f0:0024 Hewlett-Packard KU-0316 Keyboard
Bus 002 Device 002: ID 046d:c016 Logitech, Inc. Optical Wheel Mouse
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

~$ service dlm status
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-12-04 11:43:29 IST; 25min ago
  Process: 1058 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/4.4.24 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 1065 (DisplayLinkMana)
    Tasks: 32 (limit: 4915)
   CGroup: /system.slice/dlm.service
           └─1065 /opt/displaylink/DisplayLinkManager_

Thank you in advance,
Yuri

AdnanHodzic, 2018-12-04:
What is the output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

syuriks, 2018-12-04:
cat /etc/X11/xorg.conf.d/20-displaylink.conf 
Section "Device"
  Identifier  "DisplayLink"
  Driver      "modesetting"
  Option      "PageFlip" "false"
EndSection

$ lspci | grep -i vga09:00.0 VGA compatible controller: ASPEED Technology, Inc. ASPEED Graphics Family (rev 30)

$ lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
ast

lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5
ASPEED

AdnanHodzic, 2018-12-04:
I honestly don't know anything about this card, so I have no idea how your Xorg config should look like.

But out of sheer curiosity what happens if you do `mv  /etc/X11/xorg.conf.d/20-displaylink.conf /tmp/20-displaylink.conf` logout and then re-login again?

syuriks, 2018-12-04:
I don't see any difference after moving `20-displaylink.conf` and re-login.

syuriks, 2018-12-05:
Please look at Xorg.0.log output. DVI output is identified but not connected.

```
~$ cat ~/.local/share/xorg/Xorg.0.log
[  1169.330] 
X.Org X Server 1.19.6
Release Date: 2017-12-20
[  1169.331] X Protocol Version 11, Revision 0
[  1169.331] Build Operating System: Linux 4.4.0-138-generic x86_64 Ubuntu
[  1169.331] Current Operating System: Linux celare-tsg-xeon 4.15.0-42-generic #45-Ubuntu SMP Thu Nov 15 19:32:57 UTC 2018 x86_64
[  1169.331] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-42-generic root=UUID=8e737323-fd5a-470a-9922-be0b1c4a36d4 ro quiet splash vt.handoff=1
[  1169.331] Build Date: 25 October 2018  04:11:27PM
[  1169.331] xorg-server 2:1.19.6-1ubuntu4.2 (For technical support please see http://www.ubuntu.com/support) 
[  1169.331] Current version of pixman: 0.34.0
[  1169.331] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[  1169.331] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[  1169.331] (==) Log file: "/home/celare/.local/share/xorg/Xorg.0.log", Time: Wed Dec  5 17:48:03 2018
[  1169.331] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[  1169.332] (==) No Layout section.  Using the first Screen section.
[  1169.332] (==) No screen section available. Using defaults.
[  1169.332] (**) |-->Screen "Default Screen Section" (0)
[  1169.332] (**) |   |-->Monitor "<default monitor>"
[  1169.332] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[  1169.332] (==) Automatically adding devices
[  1169.332] (==) Automatically enabling devices
[  1169.332] (==) Automatically adding GPU devices
[  1169.332] (==) Automatically binding GPU devices
[  1169.332] (==) Max clients allowed: 256, resource mask: 0x1fffff
[  1169.332] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[  1169.332] 	Entry deleted from font path.
[  1169.332] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[  1169.332] 	Entry deleted from font path.
[  1169.332] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[  1169.332] 	Entry deleted from font path.
[  1169.332] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[  1169.332] 	Entry deleted from font path.
[  1169.332] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[  1169.332] 	Entry deleted from font path.
[  1169.332] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	/usr/share/fonts/X11/Type1,
	built-ins
[  1169.332] (==) ModulePath set to "/usr/lib/xorg/modules"
[  1169.332] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[  1169.332] (II) Loader magic: 0x559cf6b69020
[  1169.332] (II) Module ABI versions:
[  1169.332] 	X.Org ANSI C Emulation: 0.4
[  1169.332] 	X.Org Video Driver: 23.0
[  1169.332] 	X.Org XInput driver : 24.1
[  1169.332] 	X.Org Server Extension : 10.0
[  1169.333] (++) using VT number 2

[  1169.336] (II) systemd-logind: took control of session /org/freedesktop/login1/session/_35
[  1169.336] (II) xfree86: Adding drm device (/dev/dri/card0)
[  1169.337] (II) systemd-logind: got fd for /dev/dri/card0 226:0 fd 12 paused 0
[  1169.337] (II) xfree86: Adding drm device (/dev/dri/card1)
[  1169.338] (II) systemd-logind: got fd for /dev/dri/card1 226:1 fd 13 paused 0
[  1169.349] (--) PCI:*(0:9:0:0) 1a03:2000:1a03:2000 rev 48, Mem @ 0xf4000000/67108864, 0xf8000000/131072, I/O @ 0x0000e000/128, BIOS @ 0x????????/131072
[  1169.349] (II) LoadModule: "glx"
[  1169.349] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[  1169.350] (II) Module glx: vendor="X.Org Foundation"
[  1169.350] 	compiled for 1.19.6, module version = 1.0.0
[  1169.350] 	ABI class: X.Org Server Extension, version 10.0
[  1169.350] (==) Matched ast as autoconfigured driver 0
[  1169.350] (==) Matched ast as autoconfigured driver 1
[... 542 lines cut ...]
[  2615.875] (II) modeset(0): Using hsync ranges from config file
[  2615.875] (II) modeset(0): Using vrefresh ranges from config file
[  2615.875] (II) modeset(0): Printing DDC gathered Modelines:
[  2615.875] (II) modeset(0): Modeline "1920x1080"x0.0  148.50  1920 2008 2052 2200  1080 1084 1089 1125 +hsync +vsync (67.5 kHz eP)
[  2615.875] (II) modeset(0): Modeline "1680x1050"x0.0  146.25  1680 1784 1960 2240  1050 1053 1059 1089 -hsync +vsync (65.3 kHz e)
[  2615.875] (II) modeset(0): Modeline "800x600"x0.0   40.00  800 840 968 1056  600 601 605 628 +hsync +vsync (37.9 kHz e)
[  2615.875] (II) modeset(0): Modeline "800x600"x0.0   36.00  800 824 896 1024  600 601 603 625 +hsync +vsync (35.2 kHz e)
[  2615.875] (II) modeset(0): Modeline "640x480"x0.0   31.50  640 656 720 840  480 481 484 500 -hsync -vsync (37.5 kHz e)
[  2615.875] (II) modeset(0): Modeline "640x480"x0.0   31.50  640 664 704 832  480 489 492 520 -hsync -vsync (37.9 kHz e)
[  2615.875] (II) modeset(0): Modeline "640x480"x0.0   25.18  640 656 752 800  480 490 492 525 -hsync -vsync (31.5 kHz e)
[  2615.875] (II) modeset(0): Modeline "720x400"x0.0   28.32  720 738 846 900  400 412 414 449 -hsync +vsync (31.5 kHz e)
[  2615.875] (II) modeset(0): Modeline "1280x1024"x0.0  135.00  1280 1296 1440 1688  1024 1025 1028 1066 +hsync +vsync (80.0 kHz e)
[... cut ...]
[  2615.875] (II) modeset(0): Modeline "1280x800"x72.0  102.80  1280 1360 1496 1712  800 801 804 834 -hsync +vsync (60.0 kHz e)
[  2615.875] (II) modeset(0): Modeline "1280x960"x70.0  120.84  1280 1368 1504 1728  960 961 964 999 -hsync +vsync (69.9 kHz e)
[  2615.875] (II) modeset(0): Modeline "1280x960"x0.0  108.00  1280 1376 1488 1800  960 961 964 1000 +hsync +vsync (60.0 kHz e)
[  2615.875] (II) modeset(0): Modeline "1280x720"x60.0   74.48  1280 1336 1472 1664  720 721 724 746 -hsync +vsync (44.8 kHz e)
```

AdnanHodzic, 2018-12-29:
@syuriks could you include the output of:
sudo ./displaylink-debian.sh --debug

syuriks, 2018-12-30:
Hello Adnan,

I don't have this board anymore. Anyway, thank you very much for your help.

Best regards,
Yuri

closed 2018-12-30 by AdnanHodzic