AdnanHodzic/displaylink-debian #185: service won't start Kubuntu 18.04 :

Yanux56, 2018-12-12:
Hello, 

I try to install displaylink-debian to connect a screen MB169B+ to my laptop using USB3. 
but the service won't start, and because I'm a noob, i don't understand why :( 

this are somme command returns,if someone understand my problem :) 

`yanux@GPCP ~> systemctl status dlm.service 
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2018-12-12 12:35:51 CET; 4s ago
  Process: 3541 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/4.2.29 && modprobe evdi) (code=exited, status=200/CHDIR)`

`yanux@GPCP ~> journalctl -xe
-- 
-- L'unité (unit) dlm.service a terminé son arrêt.
déc. 12 12:33:19 GPCP systemd[1]: Starting DisplayLink Manager Service...
-- Subject: L'unité (unit) dlm.service a commencé à démarrer
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- L'unité (unit) dlm.service a commencé à démarrer.
déc. 12 12:33:19 GPCP systemd[3061]: dlm.service: Changing to the requested working directory failed: No such file or directory
déc. 12 12:33:19 GPCP systemd[3061]: dlm.service: Failed at step CHDIR spawning /bin/sh: No such file or directory
-- Subject: Le processus /bin/sh n'a pas pu être exécuté
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Le processus /bin/sh n'a pas pu être exécuté, et a donc échoué.
-- 
-- Le code d'erreur renvoyé est 2.
déc. 12 12:33:19 GPCP systemd[1]: dlm.service: Control process exited, code=exited status=200
déc. 12 12:33:19 GPCP systemd[1]: dlm.service: Failed with result 'exit-code'.
déc. 12 12:33:19 GPCP systemd[1]: Failed to start DisplayLink Manager Service.
-- Subject: L'unité (unit) dlm.service a échoué
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- L'unité (unit) dlm.service a échoué, avec le résultat RESULT.`

`yanux@GPCP ~> sudo /bin/sh
[sudo] Mot de passe de yanux : 
# modprobe evdi || (dkms install evdi/4.2.29 && modprobe evdi)
modprobe: FATAL: Module evdi not found in directory /lib/modules/4.19.2-041902-generic

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j8 KERNELRELEASE=4.19.2-041902-generic all INCLUDEDIR=/lib/modules/4.19.2-041902-generic/build/include KVERSION=4.19.2-041902-generic DKMS_BUILD=1...(bad exit status: 2)
ERROR (dkms apport): binary package for evdi: 4.2.29 not found
Error! Bad return status for module build on kernel: 4.19.2-041902-generic (x86_64)
Consult /var/lib/dkms/evdi/4.2.29/build/make.log for more information. `

result of dmesg command when I disconnect/reconnect the USB screen : 
`[déc.12 12:39] usb 2-1: USB disconnect, device number 2
[  +4,011857] usb 2-1: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[  +0,020640] usb 2-1: New USB device found, idVendor=17e9, idProduct=ff0b, bcdDevice=31.13
[  +0,000007] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000005] usb 2-1: Product: ASUSTEK MB169B+
[  +0,000004] usb 2-1: Manufacturer: DisplayLink
[  +0,000004] usb 2-1: SerialNumber: J3LMTF044986`

`yanux@GPCP ~> uname -a
Linux GPCP 4.19.2-041902-generic #201811132032 SMP Tue Nov 13 20:34:19 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`

Thanks a lot :)

AdnanHodzic, 2018-12-12:
Have you checked [post-installation guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md) and [troubleshoot most common issues](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#troubleshooting-most-common-issues)?

AdnanHodzic, 2018-12-29:
@Yanux56 since I haven't heard from you in awhile, I'll close this issue due to inactvity. 

If you still have the same problem, feel free to re-open the issue and include the output of:
`sudo ./displaylink-debian.sh --debug`

Thanks

closed 2018-12-29 by AdnanHodzic