AdnanHodzic/displaylink-debian #190: Distribution Support Request: Devuan ascii

azfirefighter, 2018-12-28:
Devuan ascii is Debian 9 using sysvinit instead of systemd.  It shouldn't be too difficult to tweak the shell file a bit.

AdnanHodzic, 2018-12-29:
There was another request like this to add support for `sysvinit`. With that said I'll add [feature request] prefix to this issue title to keep it on my radar.

Since I'm not sure when I'll have time to do this, if you feel like giving it a try a submitting a merge request, I'll be glad to review your code.

Darcidride, 2019-03-14:
Hi @azfirefighter 

I understand why you choose to use sysvinit, maybe you will appreciate to watch this video : https://hooktube.com/o_AIw9bGogo

This code snippet is for you, this sysvinit service should work as similar - more or less - with the actual systemd service :

```
#!/bin/sh
### BEGIN INIT INFO
# Provides: displaylink-driver
# Required-Start:	$syslog $local_fs 
# Required-Stop:	$syslog $local_fs 
# Default-Start:	2 3 4 5
# Default-Stop:		0 1 6
# Short-Description: DisplayLink Driver Service
### END INIT INFO

. /lib/lsb/init-functions
prog=displaylink-driver
PIDFILE=/var/run/$prog.pid
DESC="DisplayLink Driver Service"
start() {
	log_daemon_msg "Starting $DESC" "$prog"
	start_daemon /bin/sh -c 'modprobe evdi || (dkms install evdi/5.1.26 && modprobe evdi)'
	if [ $? -ne 0 ]; then
		log_end_msg 1
		exit 1
	fi
	start_daemon -p $PIDFILE /opt/displaylink/DisplayLinkManager
	if [ $? -ne 0 ]; then
		log_end_msg 1
		exit 1
	fi
	if [ $? -eq 0 ]; then
		log_end_msg 0
	fi
	exit 0
}

stop() {
	log_daemon_msg "Stopping $DESC" "$prog"
	killproc -p $PIDFILE /opt/displaylink/DisplayLinkManager
	if [ $? -ne 0 ]; then
		log_end_msg 1
		exit 1
	fi
	if [ $? -eq 0 ]; then
		log_end_msg 0
	fi
}

force_reload() {
	stop
	start

}

case "$1" in
	start)
		start
		;;
	stop)
		stop
		;;
	force-reload)
		force_reload
		;;
	restart)
		stop
		start
		;;

	*)
		echo "$Usage: $prog {start|stop|force-reload|restart}"
		exit 2
esac
```
License code : WTFPL, use at your own risks :)

@AdnanHodzic I didn't create a PR, but i guess you will appreciate to add sysvinit support to your script, i let you do the changes :)

AdnanHodzic, 2019-03-17:
@Darcidride I really encourage people to make changes themselves.

Change seems simple enough, so I'd really encourage you to make the change yourself, submit a PR and I'll be glad to review it and merge with master. Go for it!

AdnanHodzic, 2019-09-17:
Changes have been implemented as part of https://github.com/AdnanHodzic/displaylink-debian/pull/300 MR

closed 2019-09-17 by AdnanHodzic