AdnanHodzic/displaylink-debian #192: Second monitor only detected after dock plugged into USB

matthewhegarty, 2018-12-30:
Hi

Thanks for writing this software.

I had the installation working no problem but this weekend I re-installed from scratch so that I could use Ubuntu full disk encryption.  All the hardware and OS remains the same from when it was working prior to re-install.

I have a second monitor connected directly to HDMI which works fine.

I have a third monitor which connects via DVI to a Lenovo ThinkPad Pro USB 3.0 Dock (keyboard, ethernet and mouse also go through the dock).  The monitor is not enabled following boot (doesn't show up in `xrandr`).  The only way to have the monitor show up is to disconnect the dock USB, and then reboot, and then connect the dock USB.  After about 5 seconds, then the monitor appears correctly.

Any ideas as to how I can have the monitor display without having to plug the dock USB in after boot?

#### Spec

- Lenovo ThinkPad T470
- Lenovo Pro Dock USB 3.0
- Ubuntu 18.04.1
- Linux 4.15.0-43-generic
- 1 Samsung SyncMaster 2233 22" monitor (connected to Dock)
- 1 Samsung S22C300 (connected to laptop via HDMI)

#### Logs

- [dmesg log after plugging in dock](https://github.com/AdnanHodzic/displaylink-debian/files/2717334/usb_log.txt)
- [xrandr.txt](https://github.com/AdnanHodzic/displaylink-debian/files/2717338/xrandr.txt)
- [lsusb.txt](https://github.com/AdnanHodzic/displaylink-debian/files/2717339/lsusb.txt)

AdnanHodzic, 2018-12-30:
Most welcome.

What's the output of `sudo ./displaylink-debian.sh --debug`?

matthewhegarty, 2018-12-30:
Hi Adnan
Thanks for swift response.  I tried to uninstall / reinstall using `displaylink-debian.sh` and got the following:

```
$ sudo ./displaylink-debian.sh 

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: u

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.15.0-43-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

Removing evdi module
rmmod: ERROR: Module evdi is in use
Removing disabled PageFlip for modesetting

-------------------------------------------------------------------

Performing clean-up

-------------------------------------------------------------------

Uninstall complete

-------------------------------------------------------------------

$ sudo ./displaylink-debian.sh --debug

Wrong key, aborting ...

```

AdnanHodzic, 2018-12-31:
Ok, so I just implemented feature which will resolve your uninstall problem, this was originally mentioned in: [issue 178](https://github.com/AdnanHodzic/displaylink-debian/issues/178)

Regarding debug option not working, it's because you're running old code, I had this feature implemented just a couple of days ago. Reason I know this is because under options you listed there's no mention of `[D]ebug` and in latest version it's there, i.e:

````
[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit
````

Regardless, with all this said do the following:

* In displaylink-debian dir, run `git pull` to get latest code
* Run uninstall: `sudo ./displaylink-debian.sh --uninstall`
* Since I just pushed code, run this manually as well to be sure: `sudo dkms remove evdi/4.4.24 --all`
* Reboot
* Install again `sudo ./displaylink-debian.sh --install`
* Reboot
* If problem still exists, paste the output of `sudo ./displaylink-debian.sh --debug` command

AdnanHodzic, 2018-12-31:
It's not your kernel version, I think it's your Xorg config, what if you replaced this: 

```
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
```

with this:

```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```
as your `/etc/X11/xorg.conf.d/20-displaylink.conf` and logout/login or just reboot to apply the changes?

matthewhegarty, 2018-12-31:
Yes - that has fixed it!
Many thanks for your input.
(closing issue)

AdnanHodzic, 2018-12-31:
Most welcome, I've just pushed changes so from now on this change will be automatically made when your type of graphic cards is detected.

matthewhegarty, 2019-01-04:
One thing I notice is that the DVI connected monitor has an annoying 'mouse flicker' issue, where the mouse pointer flickers very rapidly when using the monitor.  I didn't notice it before my system upgrade so I think it is since I re-installed.

I notice there are similar issues reported.  I can open a separate issue if you like?

```
$ cat /etc/X11/xorg.conf.d/20-displaylink.conf
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
$ lspci | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 620 (rev 02)
$ lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
i915
$ lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5
620

```

AdnanHodzic, 2019-01-04:
Let's move this issue to: https://github.com/AdnanHodzic/displaylink-debian/issues/126 

Just re-open it, feel free to copy paste problem you described here and include output of `sudo ./displaylink-debian.sh --debug` command.

closed 2019-01-04 by AdnanHodzic

matthewhegarty, 2019-01-08:
I was able to solve the 'mouse flicker issue' by following the instructions described [here](https://support.displaylink.com/knowledgebase/articles/1843660-screen-freezes-after-opening-an-application-only).  See also [this thread](https://bugs.launchpad.net/ubuntu/+source/xorg/+bug/1278223).

dkebler, 2019-04-02:
I too was having mouse trails and other refresh problems and had to manually make this change.  If you install @AdnanHodzic's driver without any displaylink hardware attached and then like me later get some displaylink hardware this conf file has already been created with your onboard device.  So I suppose per @AdnanHodzic if you do have the hardware (and connected) and then load then build/load this driver it should make this change for you.    Either way seems necessary for stable refresh.

> It's not your kernel version, I think it's your Xorg config, what if you replaced this:
> 
> ```
> Section "Device"
>     Identifier  "Intel"
>     Driver      "intel"
> EndSection
> ```
> 
> with this:
> 
> ```
> Section "Device"
>     Identifier  "DisplayLink"
>     Driver      "modesetting"
>     Option      "PageFlip" "false"
> EndSection
> ```
> 
> in your `/etc/X11/xorg.conf.d/20-displaylink.conf` and logout/login or just reboot to apply the changes?