AdnanHodzic/displaylink-debian #194: Ubuntu 18.10 AMD Radeon no monitors detected

dschulten, 2019-01-05:
I have an Acer Nitro AN515-42 with RX-560 and Vega 8, kubuntu 18.10. No providers are detected, screen remains dark. I am using the latest script.

Strangely, immediately after installation, kubuntu asked me to setup my screens, but that was before the reboot. I tried to unify the screens, but only on the notebook screen I saw the mouse sprite on a dark background without desktop, the external monitor remained dark.

Double checked with a Windows 10 notebook, the itec dock is working fine there.
After rebooting my  kubuntu notebook, I get the following debug output:

```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: cosmic
Kernel: 4.18.0-13-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Raven
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Raven Ridge [Radeon Vega Series / Radeon Vega Mobile Series] (rev c3)

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x7a cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 1 name:Unknown AMD Radeon GPU @ pci:0000:04:00.0
Provider 1: id: 0x44 cap: 0x6, Sink Output, Source Offload crtcs: 5 outputs: 0 associated providers: 1 name:AMD Radeon (TM) RX Graphics @ pci:0000:01:00.0

s@ds-Nitro-AN515-42:~/Dokumente/ws/displaylink-debian$ cat /etc/X11/xorg.conf.d/20-displaylink.conf
Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection
ds@ds-Nitro-AN515-42:~/Dokumente/ws/displaylink-debian$ lspci | grep -i vga
04:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Raven Ridge [Radeon Vega Series / Radeon Vega Mobile Series] (rev c3)
ds@ds-Nitro-AN515-42:~/Dokumente/ws/displaylink-debian$ lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
amdgpu
ds@ds-Nitro-AN515-42:~/Dokumente/ws/displaylink-debian$ lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5
Raven
--------------- Linux system info ----------------

Distro: Ubuntu
Release: cosmic
Kernel: 4.18.0-13-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Raven
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Raven Ridge [Radeon Vega Series / Radeon Vega Mobile Series] (rev c3)

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x7a cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 1 name:Unknown AMD Radeon GPU @ pci:0000:04:00.0
Provider 1: id: 0x44 cap: 0x6, Sink Output, Source Offload crtcs: 5 outputs: 0 associated providers: 1 name:AMD Radeon (TM) RX Graphics @ pci:0000:01:00.0

```

AdnanHodzic, 2019-01-06:
What happens if you set following as contents for: `/etc/X11/xorg.conf.d/20-displaylink.conf` followed by logout/login or reboot.

```
Section "Device"
	Identifier "DisplayLink"
	MatchDriver "evdi"
	Driver "modesetting"
	Option  "AccelMethod" "none"
EndSection
```

or if that doesn't work:

```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

matthewhegarty, 2019-01-08:
I had issues with Ubuntu 18.10 and re-installed 18.04 which I knew was working for me and after that it was fine.

AdnanHodzic, 2019-01-08:
@matthewhegarty can you give me the output of `sudo ./displaylink-debian.sh --debug`? I want to compare it to one @dschulten provided.

matthewhegarty, 2019-01-08:
Attached.
[debug1.txt](https://github.com/AdnanHodzic/displaylink-debian/files/2736152/debug1.txt)

AdnanHodzic, 2019-01-10:
I don't think there should be a problem between running on Ubuntu 18.04/18.10. I think you experienced that problem with 18.10 because until few days ago I made many changes to how xorg.conf are generated.

Since he's using an AMD/ATI graphics card I think this is all related to xorg.conf, but until I get an [answer to question I asked](https://github.com/AdnanHodzic/displaylink-debian/issues/194#issuecomment-451722453) I can't say for sure.

matthewhegarty, 2019-01-21:
Hi @resquivel24 - I suggest following the [troubleshooting guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md) (if not already), and create a new issue with debug logs if you think it is an issue with this installation script.

AdnanHodzic, 2019-02-03:
I'm closing this issue due to inactivity.

Feel free re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2019-02-03 by AdnanHodzic