AdnanHodzic/displaylink-debian #196: High CPU ussage on Kubuntu 18.10?

killmasta93, 2019-01-08:
Hi,
I was wondering if someone could shed some light on the issue im having. Sometimes when i connect i get high CPU usage not sure why. Here is a screen shot. Only happens when i connect the plugable

[image]

AdnanHodzic, 2019-01-09:
It's seems like a problem with [upstream driver](https://github.com/DisplayLink/evdi), especially since even I remember [reporting one last year](https://github.com/DisplayLink/evdi/issues/65). 

Hence I would suggest your [report an issue there](https://github.com/DisplayLink/evdi/issues).

MrMEEE, 2019-01-25:
Same problem here...

AdnanHodzic, 2019-01-30:
As part of https://github.com/AdnanHodzic/displaylink-debian/commit/b57cea89cc2e97de071e417bdd39ddd374072dbd cmmit udl/udlf will be added to blacklist.conf automatically if >= 4.14.9 is detected. 

Since I never got output of `sudo ./displaylink-debian.sh --debug` I'm not sure if this will fix your problem, but I'd definetely advise you to pull the latest changes and perform reinstall.

AdnanHodzic, 2019-02-20:
I've just merged `displaylink-5` branch with `master` which introduces big changes, among which are support for yesterday released DisplayLink 5.1 driver. Disabling modesetting as part of `displaylink-debian` as now all of it should be happening on EVDI level. I've included few more changes.

Regardless, please pull latest (`git pull`) changes on `master` and run `sudo ./displaylink-debian.sh --install`

After this everything should be working as expected and with this said I'm closing the issue. If the issue still persists, re-open the issue and include output of `sudo ./displaylink-debian.sh --debug` command.

closed 2019-02-20 by AdnanHodzic

hookumsnivy, 2020-01-04:
I'm having a similar issue on Kubuntu 18.04

Just having some windows open and running top, I typically see DisplayLinkManager using between 20 and 25% of a cpu. If I start making circles with my mouse, it jumps to over 70%. If there's a video playing, it jumps to over 100% (obviously meaning it's using more than 1). The more I do, the more CPU it demands. This is after a fresh install and reboot. Simple things such as opening a new browser tab spikes the cpu usage of DisplayLinkManager. It seems anything I do is causing the CPU to go wild. I will be creating a ticket with evdi as well, but since I found this first I thought I would share.
Also, every once in a while, it will just fail. The monitors will just go blank and I have unplug the usb and reconnect it. Sometimes that works just fine and other times it messes up my desktop config.

This is the result of running the script with the debug flag:

```Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 5.0.0-37-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.20.4-1ubuntu3~18.04.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------                                                                                                                         
                                                                                                                                                                           
File: /etc/X11/xorg.conf.d/20-displaylink.conf                                                                                                                             
Contents:                                                                                                                                                                  
 Section "OutputClass"                                                                                                                                                     
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0xa1 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 2 name:modesetting
Provider 1: id: 0x79 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting```

killmasta93, 2020-01-04:
what i did was downgrade to ubuntu 18.04 but with kernel 4.15.0-72-generic is working flawless