AdnanHodzic/displaylink-debian #198: Display not detected

sossalemaire, 2019-01-09:
Hi,
I'm using a debian 9 on a Dell XPS 9550 with an i-tec 3x 4K dock station; everything works fine except display. The result of sudo ./displaylink-debian.sh --debug is as follows:

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: stretch
Kernel: 4.9.0-8-amd64

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 530
VGA: Intel Corporation HD Graphics 530 (rev 06)
X11 version: 1.19.2-1+deb9u5

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
#Section "Device"
#	Identifier "DisplayLink"
#	MatchDriver "evdi"
#	Driver "modesetting"
#	Option  "AccelMethod" "none"
#EndSection
#Section "Device"
#    Identifier  "DisplayLink"
#    Driver      "modesetting"
#    Option      "PageFlip" "false"
#EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
I'm logging in using X11; as you can see, I tried several configurations in /etc/X11/xorg.conf.d/20-displaylink.conf, to no avail, nothing changed.

xrandr --listproviders gives me:
Providers: number : 1
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 0 name:Intel

The display works when I plug it directly into the HDMI port of the laptop, which is a good enough workaround since I only have one more screen for the moment, but I'd really like to have it working as intended!

AdnanHodzic, 2019-01-09:
That's very weird, with right xorg.conf I never had this case before, especially with Intel cards.

Suggestions, since quite some changes were pushed in last hour or so, could you pull the changes and run reinstall just to see if it does anything.

Do you have any other xorg.conf running somewhere besides `/etc/X11/xorg.conf.d/20-displaylink.conf` by any chance I.e: 
`/etc/X11/xorg.conf` or any other config under: 
`/usr/share/X11/xorg.conf.d/`

If yes remove it so you only have `/etc/X11/xorg.conf.d/20-displaylink.conf`

sossalemaire, 2019-01-10:
Thank you for your answer. I pulled, reinstalled, rebooted, it still doesn't work. I don't have a /etc/X11/xorg.conf or anything in /etc/X11/xorg.conf.d (except 20-displaylink.conf) but I have 4 files under /usr/share/X11/xorg.conf.d/; if I remove them, the following happens: the computer boots OK, the login screen appears, and all input methods freeze within thirty seconds at most (both the keyboard of the laptop and the keyboard I attached to my docking station, same thing with the trackpad and mouse). I managed to log in once, and the second remained blank (I didn't have time to try xrandr --listproviders before the keyboard froze). The files are the following:
```
cat 10-amdgpu.conf 
Section "OutputClass"
	Identifier "AMDgpu"
	MatchDriver "amdgpu"
	Driver "amdgpu"
EndSection
```
```
cat 10-quirks.conf 
# Collection of quirks and blacklist/whitelists for specific devices.

# Accelerometer device, posts data through ABS_X/ABS_Y, making X unusable
# http://bugs.freedesktop.org/show_bug.cgi?id=22442 
Section "InputClass"
        Identifier "ThinkPad HDAPS accelerometer blacklist"
        MatchProduct "ThinkPad HDAPS accelerometer data"
        Option "Ignore" "on"
EndSection

# https://bugzilla.redhat.com/show_bug.cgi?id=523914
# Mouse does not move in PV Xen guest
# Explicitly tell evdev to not ignore the absolute axes.
Section "InputClass"
        Identifier "Xen Virtual Pointer axis blacklist"
        MatchProduct "Xen Virtual Pointer"
        Option "IgnoreAbsoluteAxes" "off"
        Option "IgnoreRelativeAxes" "off"
EndSection

# https://bugs.freedesktop.org/show_bug.cgi?id=55867
# Bug 55867 - Doesn't know how to tag XI_TRACKBALL
Section "InputClass"
        Identifier "Tag trackballs as XI_TRACKBALL"
        MatchProduct "trackball"
        MatchDriver "evdev"
        Option "TypeName" "TRACKBALL"
EndSection

# https://bugs.freedesktop.org/show_bug.cgi?id=62831
# Bug 62831 - Mionix Naos 5000 mouse detected incorrectly
Section "InputClass"
        Identifier "Tag Mionix Naos 5000 mouse XI_MOUSE"
        MatchProduct "La-VIEW Technology Naos 5000 Mouse"
        MatchDriver "evdev"
        Option "TypeName" "MOUSE"
EndSection
```
```
cat 40-libinput.conf 
# Match on all types of devices but tablet devices and joysticks
Section "InputClass"
        Identifier "libinput pointer catchall"
        MatchIsPointer "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
EndSection

Section "InputClass"
        Identifier "libinput keyboard catchall"
        MatchIsKeyboard "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
EndSection

Section "InputClass"
        Identifier "libinput touchpad catchall"
        MatchIsTouchpad "on"
        MatchDevicePath "/dev/input/event*"
[... 106 lines cut ...]
Section "InputClass"
	Identifier "Waltop class"
	MatchProduct "WALTOP"
	MatchIsTablet "on"
	MatchDevicePath "/dev/input/event*"
	Driver "wacom"
EndSection

# N-Trig Duosense Electromagnetic Digitizer
Section "InputClass"
	Identifier "Wacom N-Trig class"
	MatchProduct "HID 1b96:0001|N-Trig Pen|N-Trig DuoSense"
	MatchDevicePath "/dev/input/event*"
	Driver "wacom"
	Option "Button2" "3"
EndSection
```
The content of 20-displaylink.conf is now:
```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
```
I don't know if anything from those 4 files can interfere. I hope you're not out of ideas :)

sossalemaire, 2019-01-10:
I tried removing /etc/X11/xorg.conf.d/20-displaylink.conf, here is the output of xrandr --listproviders:
Providers: number : 2
Provider 0: id: 0x77 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting
Provider 1: id: 0x3f cap: 0x5, Source Output, Source Offload crtcs: 0 outputs: 0 associated providers: 0 name:modesetting
The screen remains totally blank, it doesn't detect any signal.

coroa, 2019-01-28:
As I just commented over at #207, your problem might be the same as mine and the `udlfb` module is somehow grabbing the screen for itself. Anyway, if `lsmod | grep udl` lists both `udl` and `udlfb`, you might try blacklisting `udlfb` (by adding a new line `blacklist udlfb` to `/etc/modprobe.d/blacklist.conf`) and restarting.

coroa, 2019-01-29:
> Thanks for your message, but nothing comes up when I run `lsmod | grep udl` (neither udl, nor udlfb). Maybe that's my problem!

It's not your problem. They are only introduced in a later kernel than the version you are using and only interfere with the commercial display link drivers. I did not check for your kernel version earlier. Sorry for the noise.

AdnanHodzic, 2019-01-30:
@sossalemaire have you managed to make the changes I suggested as part of? https://github.com/AdnanHodzic/displaylink-debian/issues/198#issuecomment-458434363?

As part of https://github.com/AdnanHodzic/displaylink-debian/commit/b57cea89cc2e97de071e417bdd39ddd374072dbd commit udl/udlf will be added to blacklist.conf automatically if >= 4.14.9 is detected. 

I don't think these changes will fix your problem, but I'd definitely advise you to do `displaylink-debian` operations using latest changes.

sossalemaire, 2019-01-30:
Now the screen is detected, I can see it in xrandr or gnome's screen parameters, but the secondary screen remains frozen ; it displays the first image sent to it, then remains perfectly still. My computer sort of knows it's there, since if I move the mouse towards the screen, it disappears, then reappears if I move the mouse back towards the main screen. Maybe it's another known issue? I don't know if it is relevant, but I tried HDMI and DisplayPort, with HDMI the screen is not detected and with DisplayPort it is detected as I described above.

AdnanHodzic, 2019-01-30:
I've just added changes which will make automatic changes necessary for you card, so you don't have to do anything manually after install.

If you haven't take a look at [Troubleshooting most common issues](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#troubleshooting-most-common-issues). As following could help you find an answer to your issue: https://github.com/AdnanHodzic/displaylink-debian/issues/68 https://github.com/AdnanHodzic/displaylink-debian/issues/130 https://github.com/AdnanHodzic/displaylink-debian/issues/192

Also as part of https://github.com/AdnanHodzic/displaylink-debian/commit/b57cea89cc2e97de071e417bdd39ddd374072dbd commit udl/udlf will be added to blacklist.conf automatically if >= 4.14.9 is detected. 

This doesn't affect you since your kernel is older then that, but I'd definitely advise you to pull the latest changes and perform reinstall and report back if there any changes.

AdnanHodzic, 2019-02-20:
I've just merged `displaylink-5` branch with `master` which introduces big changes, among which are support for yesterday released DisplayLink 5.1 driver. Disabling modesetting as part of `displaylink-debian` as now all of it should be happening on EVDI level. I've included few more changes.

Regardless, please pull latest (`git pull`) changes on `master` and run `sudo ./displaylink-debian.sh --install`

After this everything should be working as expected and with this said I'm closing the issue. If the issue still persists, re-open the issue and include output of `sudo ./displaylink-debian.sh --debug` command.

closed 2019-02-20 by AdnanHodzic

Darcidride, 2019-03-12:
Hi, 

First of all, your work on this Displaylink-driver is amazing, keep up the great work! :)

Can you reopen this issue ? I'm available to try to fix it with you :)

I've the same configuration as the creator of this issue and the same issue as him. Also i can tell it still doesn't work after your branch merge and with the new code improvement you told before closing this issue.

Here is my debug output : 

systemctl status displaylink-driver.service 
● displaylink-driver.service - DisplayLink Driver Service
   Loaded: loaded (/lib/systemd/system/displaylink-driver.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-03-13 00:16:34 CET; 12min ago
  Process: 1360 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/5.1.26 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 1376 (DisplayLinkMana)
    Tasks: 33 (limit: 4915)
   Memory: 31.2M
   CGroup: /system.slice/displaylink-driver.service
           └─1376 /opt/displaylink/DisplayLinkManager

cat /etc/modprobe.d/blacklist.conf    
blacklist udlfb
blacklist udl

sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-3-amd64

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Gen
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.20.3-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x4c cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 9 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

My hardware : Macbook Pro mid-2012 (intel integrated graphics from a i5-3210M) --> Very common laptop

I'm on Xorg for the test :+1: 
loginctl show-session 2 -p Type                        ✘
Type=x11

My OS : 
$ uname -a   
Linux LUNE 4.19.0-3-amd64 #1 SMP Debian 4.19.20-1 (2019-02-11) x86_64 GNU/Linux
$ cat /etc/debian_version
buster/sid
--> Yes it's a unstable Debian

By the way i noticed a bug in your script, for the latest kernel (4.19.0-3-amd64) to work with your script, you need "linux-kbuild-4.19" in its latest version with the command line "sudo apt install -t unstable linux-kbuild-4.19" (if you don't have the "4.19.20-1" version of "linux-kbuild-4.19" you will have dependencies issues and a crash :)):

apt-cache policy linux-kbuild-4.19
linux-kbuild-4.19:
  Installed: 4.19.20-1
  Candidate: 4.19.20-1
  Version table:
 *** 4.19.20-1 100
        100 https://deb.debian.org/debian unstable/main amd64 Packages
        100 /var/lib/dpkg/status
     4.19.16-1 900
        900 https://deb.debian.org/debian buster/main amd64 Packages
     4.19.16-1~bpo9+1 100
        100 https://deb.debian.org/debian stretch-backports/main amd64 Packages

It's very frustrating this pain in the ass to work on this to get it working on our favorite OS :/
Will we be able to use an external display through USB3 on GNU/Linux one day... ? :S

oscar-reyes-gs, 2019-12-15:
I am also having the issue of displays not being detected.. This is the debug output:

```
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-6-amd64

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.4-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0x1f2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x1b5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```