AdnanHodzic/displaylink-debian #200: Rendering errors with i915 (branch newgen_xorg_pt2)

adangel, 2019-01-09:
While testing #191, I used the branch nextgen_kernel, which includes the changes on newgen_xorg_pt2.

It seems, for my graphics card, I still need to use the old Xorg config, otherwise, the output is not rendered correctly (screen is on, but not usable).

Here's my setting right now:
```
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-1-amd64

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 540
VGA: Intel Corporation Iris Graphics 540 (rev 0a)
X11 version: 1.20.3-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    #MatchDriver "evdi"
    Driver      "modesetting"
    #Option      "AccelMethod" "none"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 1 name:modesetting
Provider 1: id: 0x217 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```

toddsnider-epc, 2019-01-09:
I'm having the same issue as adangel.  I temporarily reverted back to the 4.18 kernel and f4a9a45 as a workaround.  Here are my settings:
```
--------------- Linux system info ----------------

Distro: Debian
Release: sid
Kernel: 4.18.0-3-amd64

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [HD
VGA: Intel Corporation HD Graphics 520 (rev 07)
X11 version: 1.20.3-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x90 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 1 name:modesetting
Provider 1: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```

AdnanHodzic, 2019-01-09:
Interesting, okay good to know.

I saw this problem on:

- https://github.com/DisplayLink/evdi/issues/133 
- https://support.displaylink.com/knowledgebase/articles/1181623-displaylink-ubuntu-driver-after-recent-x-upgrades

 and thought to make these changes preemptively. But in this case it's only causing problems, so I've just disabled it as part of last commit.

Please run reinstall and in case problem is still there re-open the issue.

closed 2019-01-09 by AdnanHodzic

adangel, 2019-01-10:
@AdnanHodzic I did some more tests and clean install - and now it works in both cases.
I had a file `/etc/X11/xorg.conf` with the following content:

```
Section "Device"
	Identifier "Intel"
	Driver "intel"
#	Option "AccelMethod" "uxa"
#	Option "VSync" "false"
EndSection
```

I've removed this file and did a fresh install from master branch: 
[2018-01-10-master-debug.txt](https://github.com/AdnanHodzic/displaylink-debian/files/2746641/2018-01-10-master-debug.txt) - this works.

Then I deinstalled displaylink and installed it again with commenting in the new xorg configuration: 
[2018-01-10-master-newx-debug.txt](https://github.com/AdnanHodzic/displaylink-debian/files/2746644/2018-01-10-master-newx-debug.txt) - this still works!

Maybe my problem was only caused by the leftover old xorg.conf?

AdnanHodzic, 2019-01-10:
@adangel I'd definitely say the problem was a leftover xorg.conf, as I seen this in couple of other cases. 

If you don't see any performance differences between these 2, let's keep using:

```
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

as the default for now and then we can re-evaluate option to use: 

```
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
``` 

Although it might only be an Ubuntu issue. Not sure, I'll have to keep an eye on this one.

adangel, 2019-01-12:
So far, I don't see a performance difference between the two configs.

AdnanHodzic, 2019-01-15:
Ok, in that case as part of last push I've re-enabled execution of `xorg_modesetting_newgen` in case xorg >= 1.19.6 is detected.