AdnanHodzic/displaylink-debian #201: Display is black on debian 9

igoricegit, 2019-01-12:
Hello Adnan!

Sorry for bothering you, but I have the same issue. I have crashed totally my debian and reinstalled it. after I have the same issue just dark screen. Next I decided to upgrade the kernel to 4.18 and used your nexgen-kernel. the same no luck (((

igorice@sunderstorm:~/Downloads/displaylink-debian$ sudo ./displaylink-debian.sh --debug

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: stretch
Kernel: 4.18.0-0.bpo.3-amd64

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4:40
4.4.24
4:40
4:40

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: GP106
VGA: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] (rev a1)
X11 version: 1.19.2-1+deb9u5

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
Section "ServerLayout"
Identifier "layout"
Screen 0 "nvidia"
Inactive "intel"
EndSection

Section "Device"
Identifier "intel"
Driver "modesetting"
Option "AccelMethod" "None"
EndSection

Section "Screen"
Identifier "intel"
Device "intel"
EndSection

Section "Device"
Identifier "nvidia"
Driver "nvidia"
Option "ConstrainCursor" "off"
EndSection

Section "Screen"
Identifier "nvidia"
Device "nvidia"
Option "AllowEmptyInitialConfiguration" "on"
Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x1b9 cap: 0x1, Source Output crtcs: 4 outputs: 7 associated providers: 0 name:NVIDIA-0
Provider 1: id: 0x241 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

AdnanHodzic, 2019-01-15:
Don't use any other branch then `master`, all changes have been merged to it, make sure you're using latest master. With that said, what if you:

**1:** check you don't have multiple xorg.conf files like you did last time: https://github.com/AdnanHodzic/displaylink-debian/issues/176#issuecomment-451777141

If you do make sure to remove them all except: `/etc/X11/xorg.conf.d/20-displaylink.conf`

If after restart problem persists, move to **2**

**2:** put back this section at the end of your: `/etc/X11/xorg.conf.d/20-displaylink.conf` which you requested to have removed as part of https://github.com/AdnanHodzic/displaylink-debian/issues/176#issuecomment-453018193 

```
Section "Device"
    Identifier "DisplayLink"
EndSection
```

igoricegit, 2019-01-15:
Hello Adnan.

No any additional config files.
I uninstalled and installed again DL from master
Added piece of code.

No any changes

Debug output
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = "en_US:ru",
        LC_ALL = (unset),
        LC_MEASUREMENT = "ru_UA.UTF-8",
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_US.UTF-8").
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = "en_US:ru",
        LC_ALL = (unset),
        LC_MEASUREMENT = "ru_UA.UTF-8",
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_US.UTF-8").

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: stretch
Kernel: 4.18.0-0.bpo.3-amd64

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------
                                                                                                                                                                    
Vendor: nvidia                                                                                                                                                      
Subsystem: GP106                                                                                                                                                    
VGA: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] (rev a1)
X11 version: 1.19.2-1+deb9u5

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

Section "Device"
    Identifier "DisplayLink"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x1b9 cap: 0x1, Source Output crtcs: 4 outputs: 7 associated providers: 0 name:NVIDIA-0
Provider 1: id: 0x241 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

AdnanHodzic, 2019-01-15:
Okay, what if you put this as your `/etc/X11/xorg.conf.d/20-displaylink.conf` contents:

```
Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "DisplayLink"
EndSection

Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

Section "Screen"
    Identifier "DisplayLink"
    Device "DisplayLink"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection
```

and reboot?

Also could you tell me what are the contents of `/usr/share/sddm/scripts/Xsetup` file?

igoricegit, 2019-01-16:
Hello!

My Xsetup

#!/bin/sh
# Xsetup - run as root before the login dialog appears

xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --auto

With this piece of code for /etc/X11/xorg.conf.d/20-displaylink.conf it crush the boot

Thanks!

AdnanHodzic, 2019-01-16:
Right, but it seems like your`/usr/share/sddm/scripts/Xsetup` file is incomplete, put following contents in it:

```
#!/bin/sh
# Xsetup - run as root before the login dialog appears

if [ -e /sbin/prime-offload ]; then
    echo running NVIDIA Prime setup /sbin/prime-offload
    /sbin/prime-offload
fi

xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --auto
```

Reboot and see what happens. 

If nothing happens, make sure that's still the contents of `Xsetup` file and pull the latest changes from master and run re-install, and see if anything changes after that.

igoricegit, 2019-01-16:
I added and reistalled DL again. All the same (( Black screen

igoricegit, 2019-01-23:
Maybe there is soem third party prooblem. Like some driver?

AdnanHodzic, 2019-01-26:
Not sure, if it worked with all changes made in https://github.com/AdnanHodzic/displaylink-debian/issues/176 there's no reason why it wouldn't now unless something is wrong with your install/config.

So only thing I can suggest is for you to check if there any deviations between files made by installer and what was done https://github.com/AdnanHodzic/displaylink-debian/issues/176 

Also make sure to check [Post install guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md)

AdnanHodzic, 2019-01-30:
As part of https://github.com/AdnanHodzic/displaylink-debian/commit/b57cea89cc2e97de071e417bdd39ddd374072dbd commit udl/udlf will be added to blacklist.conf automatically if >= 4.14.9 is detected. 

I'm not sure if this will fix your problem, but I'd definitely advise you to pull the latest changes and perform reinstall and report back if there any changes.

igoricegit, 2019-01-30:
Hello Adnan!

I have some progress. I hope you help me to finish this.

So when I run commands

xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --auto

in console mobile monitor starting to work. I have checked my /usr/share/sddm/scripts/Xsetup

it has #!/bin/sh
# Xsetup - run as root before the login dialog appears

if [ -e /sbin/prime-offload ]; then
    echo running NVIDIA Prime setup /sbin/prime-offload
    /sbin/prime-offload
fi

xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --auto

igoricegit, 2019-01-30:
Finally I figured it out. There was way stupid problem. To make changes in /usr/share/sddm/scripts/Xsetup you need to make in executable))

sudo chmod +x /usr/share/sddm/scripts/Xsetup

it helped

closed 2019-01-30 by igoricegit

AdnanHodzic, 2019-01-30:
Those are great news. So what did you do differently then what it's done with `displaylink-debian`? I want to make necessary code changes on my side.

Original contents of `/usr/share/sddm/scripts/Xsetup` file before you run `displaylink-debian` are:

```
if [ -e /sbin/prime-offload ]; then
echo running NVIDIA Prime setup /sbin/prime-offload
/sbin/prime-offload
fi
```
After you run `displaylink-debian` install and Nvidia card is detected among other things, changes which will be made to `/usr/share/sddm/scripts/Xsetup` file so it looks like this:

```
if [ -e /sbin/prime-offload ]; then
echo running NVIDIA Prime setup /sbin/prime-offload
/sbin/prime-offload
fi

xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --auto
```

So what are you saying is that only thing you had to make was to give executable permissions to Xsetup file by running: `sudo chmod +x /usr/share/sddm/scripts/Xsetup`?

Is that the only change you did outside of `displaylink-debian` installer?

igoricegit, 2019-01-30:
Hello Adnan!

Yes I have completely reinstalled DL with no result. Next I found this issue via command line and I decided that

xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --auto

are not executable. I firstly checked ownership and next permissions. and I decided to add it and reboot. Only this one thing.

And yes my file looks like this 

if [ -e /sbin/prime-offload ]; then
echo running NVIDIA Prime setup /sbin/prime-offload
/sbin/prime-offload
fi

xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --auto

Thanks!

igoricegit, 2019-01-31:
I think the problam was in case when xsetup creates during DL installation as new file and just do not have +x permittion.

AdnanHodzic, 2019-01-31:
Don't think that was the case. As `displaylink-debian` will only appends necessary content to the existing (xsetup) file.

But regardless, re-adding executable permissions can't hurt.

AdnanHodzic, 2019-01-31:
What do you mean, in this comment here you said you do and you confirmed it's contents are as they are supposed to be: https://github.com/AdnanHodzic/displaylink-debian/issues/201#issuecomment-459127959

igoricegit, 2019-02-01:
No I men did not have before installing. Now of cource I have

igoricegit, 2019-02-01:
After installing clean debian