AdnanHodzic/displaylink-debian #202: Kali linux 2018.4. Problematic repositorys.

jf6921668, 2019-01-14:
Hi! I recently got a adapter and decided to use this script. It works great on Linux mint but when i run the script on Kali Linux i get errors with the apt installments. I'm currently not near my computer so i cant give the exact errors but it says theirs a problem with dkpg and glob an regex. I have done apt update upgrade and dist upgrade to no help. I tried a fresh image installation as well and i've edited my sources.list. Any help would be appreciated. Thanks very much!

AdnanHodzic, 2019-01-14:
If you want me to help you then you'll need to provide exact error you're getting.

Also make sure to include output of:

```
sudo ./displaylink-debian.sh --debug
```

jf6921668, 2019-01-14:
I will as soon as i get home tonight thank you so much!

jf6921668, 2019-01-14:
ok @AdnanHodzic  here it is.
ISSUE:
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
dpkg-query: package 'linux-headers-4.18.0-kali2-amd64' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
linux-headers-4.18.0-kali2-amd64 not found! Install? [Y/n] y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.18.0-kali2-amd64
E: Couldn't find any package by glob 'linux-headers-4.18.0-kali2-amd64'
E: Couldn't find any package by regex 'linux-headers-4.18.0-kali2-amd64'
linux-headers-4.18.0-kali2-amd64 installation failed.  Aborting

DEBUG:

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: d

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

Unit dlm.service could not be found.
--------------- Linux system info ----------------

Distro: Kali
Release: kali-rolling
Kernel: 4.18.0-kali2-amd64

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Processor
VGA: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Graphics & Display (rev 0e)
X11 version: 1.20.1-5

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 4 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

AdnanHodzic, 2019-01-15:
Seems like this is a Kali related issue, maybe [try doing the same thing that was done here](https://forums.kali.org/showthread.php?42939-Unable-to-locate-package-linux-headers-4-18-0-kali2-amd64) and install headers manually for your kernel version.

jf6921668, 2019-01-15:
I have run dist upgrade. Once its done i will give you my results...

AdnanHodzic, 2019-01-30:
As part of https://github.com/AdnanHodzic/displaylink-debian/commit/b57cea89cc2e97de071e417bdd39ddd374072dbd commit udl/udlf will be added to blacklist.conf automatically if >= 4.14.9 is detected. 

I'm not sure if this will fix your problem, but I'd definitely advise you to pull the latest changes and perform reinstall and report back if there any changes.

AdnanHodzic, 2019-02-03:
I'm closing this issue due to inactivity.

Feel free re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2019-02-03 by AdnanHodzic