AdnanHodzic/displaylink-debian #203: Debug File Mint 19.1

ChristiaanVansteenkiste, 2019-01-16:
I'm no programmer, just a nOOb user :-). I followd after installation guide, but I am not able to find the process asked for. SO i just went on an posted this dubug. Thankk!

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

Unit dlm.service could not be found.
--------------- Linux system info ----------------

Distro: LinuxMint
Release: tessa
Kernel: 4.15.0-43-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 530
VGA: Intel Corporation HD Graphics 530 (rev 06)
X11 version: 1.19.6-1ubuntu4.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x23a cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 1 name:NVIDIA-0
Provider 1: id: 0x44 cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 2 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

jf6921668, 2019-01-16:
What is the error youre getting?

AdnanHodzic, 2019-01-16:
@ChristiaanVansteenkiste I've just made a change that added support for Mint 19.1 Tessa, not sure how you didn't get a message about unsupported platforms. 

So pull the latest changes and run "re-install".

After this just as @jf6921668 said I would also be interested in what error you're getting during install.

PedroFernandoLima, 2019-01-23:
**@AdnanHodzic  pls help. not working monitor** 

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y    

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: LinuxMint
Release: tara
Kernel: 4.15.0-43-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
X11 version: 1.19.6-1ubuntu4.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
#Section "Device"
#       Identifier "DisplayLink"
#       MatchDriver "evdi"
#       Driver "modesetting"
#       Option  "AccelMethod" "none"
#EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x64 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0x3f cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

pedro@pedro-Lenovo ~/Downloads $ sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: LinuxMint
Release: tara
Kernel: 4.15.0-43-generic

---------------- DisplayLink info ----------------

Driver version: 
EVDI service status: up and running
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
X11 version: 1.19.6-1ubuntu4.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
#Section "Device"
#       Identifier "DisplayLink"
#       MatchDriver "evdi"
#       Driver "modesetting"
#       Option  "AccelMethod" "none"
#EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x64 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0x3f cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

AdnanHodzic, 2019-01-26:
@PedroFernandoLima your `/etc/X11/xorg.conf.d/20-displaylink.conf` has wrong values, are you using lates version of `displaylink-debian` as I'm not sure how you're getting those values.

What if you put:

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```
 `/etc/X11/xorg.conf.d/20-displaylink.conf`?

PedroFernandoLima, 2019-01-28:
the screen turns black shortly after delivery of the system to xorg

AdnanHodzic, 2019-01-28:
and it also fails if contents of: `/etc/X11/xorg.conf.d/20-displaylink.conf` are:

```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

?

PedroFernandoLima, 2019-01-28:
https://github.com/AdnanHodzic/displaylink-debian/issues/79

AdnanHodzic, 2019-01-30:
As part of https://github.com/AdnanHodzic/displaylink-debian/commit/b57cea89cc2e97de071e417bdd39ddd374072dbd commit udl/udlf will be added to blacklist.conf automatically if >= 4.14.9 is detected. 

I'm not sure if this will fix your problem, but I'd definitely advise you to pull the latest changes and perform reinstall and report back if there any changes.

AdnanHodzic, 2019-02-03:
I'm closing this issue due to inactivity.

Feel free re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2019-02-03 by AdnanHodzic

AdnanHodzic, 2019-02-20:
I've just merged `displaylink-5` branch with `master` which introduces big changes, among which are support for yesterday released DisplayLink 5.1 driver. Disabling modesetting as part of `displaylink-debian` as now all of it should be happening on EVDI level. I've included few more changes.

Regardless, please pull latest (`git pull`) changes on `master` and run `sudo ./displaylink-debian.sh --install`

After this everything should be working as expected and with this said I'm closing the issue. If the issue still persists, re-open the issue and include output of `sudo ./displaylink-debian.sh --debug` command.