AdnanHodzic/displaylink-debian #204: Kali Linux doesnt detect display

jf6921668, 2019-01-17:
I finally got it to install correctly but now my issue is i cant seem to get it to detect the display. There are no errors i can give you. Any help would be appreciated 
Thanks @AdnanHodzic

AdnanHodzic, 2019-01-17:
Have you [referred to Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md)?

jf6921668, 2019-01-17:
@AdnanHodzic
I had kali kde installed i edited the script to support it which may be a problem as well as i had the uefi installation so ill have to make some adjustments then ill get back to you

AdnanHodzic, 2019-01-17:
What kind of changes did you have to make to support KDE?

As it's suggested in [referred to Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md) UEFI/secure boot should be disabled.

jf6921668, 2019-01-17:
It was detected as kali n/a so i changed 2018.4 in the script to n/a so it would let the script run.

AdnanHodzic, 2019-01-17:
Ah, okay good to know. Feel free to [create a pull request](https://github.com/AdnanHodzic/displaylink-debian/pulls) with changes you made, I'll be happy to review them and pull them to master.

jf6921668, 2019-01-17:
@AdnanHodzic
Ok i will when i get home tonight. Ill also let you know if it works when i change uefi.
I also love the idea of this project so i plan to help develope this script so others like myself can do this.

jf6921668, 2019-01-17:
@mjdisano
Did you do a non uefi install
Dont use kde try the default
Other than that im at a loss

mjdisano, 2019-01-18:
Legacy mode does not work, not using kde 
Interesting is when i list providers my man display on laptop does not show intel
root@kali:~# xrandr --listproviders
Providers: number : 3
Provider 0: id: 0xa1 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0x7a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

AdnanHodzic, 2019-01-26:
@mjdisano what happens if you put following as contents of your ` /etc/X11/xorg.conf.d/20-displaylink.conf` file:

```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

lagordon70, 2019-01-27:
I had a very similar problem and nothing seemed to work on kubuntu 18.04.1. I solved it by trashing my etc/X11/xorg.conf and installing the nvidia-driver-390 package.  Hope this helps someone.

AdnanHodzic, 2019-01-29:
@lagordon70 `displaylink-debian` will remove`/etc/X11/xorg.conf` file in case NVIDIA card is detected. But from what I see here judging by debug info here graphic card in this case is Intel.

AdnanHodzic, 2019-01-30:
As part of https://github.com/AdnanHodzic/displaylink-debian/commit/b57cea89cc2e97de071e417bdd39ddd374072dbd commit udl/udlf will be added to blacklist.conf automatically if >= 4.14.9 is detected. 

I'm not sure if this will fix your problem, but I'd definitely advise you to pull the latest changes and perform reinstall and report back if there any changes.

AdnanHodzic, 2019-02-03:
I'm closing this issue due to inactivity.

Feel free re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2019-02-03 by AdnanHodzic

ptspawn, 2019-02-13:
Had a similar problem when installing in kali 2019.1
changed the script from 2018.4 to n/a to run the install but after the reboot, i had no more gui.

Had to uninstall.
I'm running on a GeForce GTX 1050 Mobile and had to install hashcat-nivia over the nvidia driver to get the graphics card to work properly (otherwise it always freezes before shutting down). Always ran perfectly since.

debug file:

-------------------------------------------------------------------

Starting Debug ...

--------------- Linux system info ----------------

Distro: Kali
Release: n/a
Kernel: 4.19.0-kali1-amd64

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation HD Graphics 630 (rev 04)
VGA (3D): NVIDIA Corporation GP107M [GeForce GTX 1050 Mobile] (rev a1)
X11 version: 1.20.3-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

AdnanHodzic, 2019-02-16:
@ptspawn thanks, that was a valuable information. 

That's why as part of last commit (https://github.com/AdnanHodzic/displaylink-debian/commit/12cad8fed686ca16384adf79ff7faf8ec2a6bca0) I've added both Kali codename `n\a` and installation of `hashcat-nvidia` package if nvidia cards are detected.

rec1313, 2021-06-16:
> Have you [referred to Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-gui Thank you so much i spent 4 days trying figure out what was the problem. after doing everything i stumbled on your post and i fixed issue with in seconds.

IhsanMowaket, 2021-07-09:
hey you can try this tool:
https://github.com/IhsanMowaket/Kali-Parrot_Dual_Monitor