AdnanHodzic/displaylink-debian #205: Displaylink Ubuntu 18.04

resquivel24, 2019-01-19:
After installing the driver for DisplayLink, and following the tutorial online, when re-booting the monitors were very sluggishly, even though the mouse was moving with the correct velocity, viewing videos, or navigating from folders, it was very slow.  Is there any fix on this bug?  I'm installing it on an HP Pavilion, with an AMD Radeon 6 graphics card.

AdnanHodzic, 2019-01-19:
Could you give me the output of:
```
sudo ./displaylink-debian.sh --debug
```

adangel, 2019-01-19:
You'll need to execute the script `./displaylink-debian.sh` and not the displaylink driver you downloaded.
It seems, you didn't use the script at all yet?

Then please follow the install steps like outlined on the [Readme](https://github.com/AdnanHodzic/displaylink-debian/blob/master/README.md):

    git clone https://github.com/AdnanHodzic/displaylink-debian.git
    cd displaylink-debian/
    sudo ./displaylink-debian.sh --debug

The script will download the driver on its own, install it and configure you system, so that hopefully the displaylink adapter works.

resquivel24, 2019-01-19:
@AdnanHodzic, the following is the output on the "sudo ./displaylink-debian.sh --debug command":

sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.15.0-43-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: [103c:8246]
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Carrizo (rev c5)
X11 version: 1.19.6-1ubuntu4.2
14.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x54 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:AMD Radeon R6 Graphics @ pci:0000:00:01.0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

resquivel24, 2019-01-21:
@adangel , Thanks Adangel, I did as instructed, but now I'm having the issue that my two monitors are displaying everything in slow motion, any input on how to fix this bug?  I'd appreciate very much if you could assist me on this.

adangel, 2019-01-21:
@resquivel24 This sounds like #180 ? Unfortunately I don't have a AMD graphics card, so I can't help you. Maybe @Tegaf has more info on his setup? (Although he mentions that it still feels slow...)

AdnanHodzic, 2019-01-26:
@resquivel24 what happens if as part of `/usr/share/X11/xorg.conf.d/20-displaylink.conf` file you put following contents:

```
Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    #Option     "PageFlip" "false"
EndSection
```

does that make it any faster? I also don't have AMD graphics card so I can't test any of this myself.

Tegaf, 2019-01-26:
With this setting it worked better (faster) but it was't perfect (Ubuntu 18.10 with a 4.18 kernel).

Unfortunately (fortunately) I complained about the AMD laptop (fan problem) and now I have an Intel laptop... Sorry, a can't help you.

resquivel24, 2019-01-28:
@AdnanHodzic, one question how do you change permission on the file on Ubuntu?

AdnanHodzic, 2019-01-28:
Ummm ... http://lmgtfy.com/?q=how+do+you+change+permission+on+the+file+on+Ubuntu%3F ?

resquivel24, 2019-01-29:
@AdnanHodzic the 20-displaylink.confM-D, file, I already change the input to what you request to change it on the above message, it doesn't work, but I thank you very much @AdnanHodzic.

AdnanHodzic, 2019-01-29:
@resquivel24 feel free to even remove that file. As it will be re-created once `displaylink.sh --install` is run

In meantime try AMD Carrizo Displaylink related xorg configs.

AdnanHodzic, 2019-01-30:
As part of https://github.com/AdnanHodzic/displaylink-debian/commit/b57cea89cc2e97de071e417bdd39ddd374072dbd commit udl/udlf will be added to blacklist.conf automatically if >= 4.14.9 is detected. 

I'm not sure if this will fix your problem, but I'd definitely advise you to pull the latest changes and perform reinstall and report back if there any changes.

AdnanHodzic, 2019-02-03:
I'm closing this issue due to inactivity.

Feel free re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2019-02-03 by AdnanHodzic

AdnanHodzic, 2019-02-20:
I've just merged `displaylink-5` branch with `master` which introduces big changes, among which are support for yesterday released DisplayLink 5.1 driver. Disabling modesetting as part of `displaylink-debian` as now all of it should be happening on EVDI level. I've included few more changes.

Regardless, please pull latest (`git pull`) changes on `master` and run `sudo ./displaylink-debian.sh --install`

After this everything should be working as expected and with this said I'm closing the issue. If the issue still persists, re-open the issue and include output of `sudo ./displaylink-debian.sh --debug` command.

moddie666, 2019-06-07:
Hi,
my machine is experiencing severe display-performance problems and high cpu utilisation.
Also, if i do not delete /etc/X11/xorg.conf.d/20-displaylink.conf + reboot, I am always missing the two extra providers.
#-----------------------------------------------------------------------------------------------------------------------------------#
--------------- Linux system info ----------------
Distro: Debian
Release: buster
Kernel: 4.19.0-5-amd64
---------------- DisplayLink info ----------------
Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0
------------------ Graphics card -----------------
Vendor: amdgpu
Subsystem: [Radeon
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Raven Ridge [Radeon Vega Series / Radeon Vega Mobile Series] (rev c3)
VGA (3D): 
X11 version: 1.20.3-1
X11 configs: /etc/X11/Xsession.d/50-synaptics.conf
-------------- DisplayLink xorg.conf -------------
File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
-------------------- Monitors --------------------
Providers: number : 3
Provider 0: id: 0x9b cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 4 associated providers: 2 name:Unknown AMD Radeon GPU @ pci:0000:05:00.0
Provider 1: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
#-----------------------------------------------------------------------------------------------------------------------------------#

Usage while running with 2 screens @ Displaylink + eDP + HDMI (4 Total)
#-----------------------------------------------------------------------------------------------------------------------------------#
Tasks: 284 total,   6 running, 278 sleeping,   0 stopped,   0 zombie
%Cpu(s): 24.2 us, 31.2 sy,  0.0 ni, 44.5 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :  15662.4 total,  12869.9 free,   1897.9 used,    894.7 buff/cache
MiB Swap:   7628.0 total,   7628.0 free,      0.0 used.  13429.7 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                 
 1086 root      20   0 2665452  88448  10500 S 212.5   0.6   1:47.19 /opt/displaylink/DisplayLinkManager                                                                                     
 3470 XXX       20   0  671960 153932  74356 S 118.8   1.0   0:03.38 /opt/google/chrome/chrome 
 1083 root      20   0 1769472 105128  76196 S  43.8   0.7   0:52.20 /usr/lib/xorg/Xorg :0 -seat seat0 -auth /var/run/lightdm/root/:0 -nolisten tcp vt7 -novtswitch                          
 2653 XXX       20   0 1049880 268444 124612 R  37.5   1.7   0:39.81 /opt/google/chrome/chrome
 #-----------------------------------------------------------------------------------------------------------------------------------#
 processor	: 7
vendor_id	: AuthenticAMD
cpu family	: 23
model		: 17
model name	: AMD Ryzen 7 2700U with Radeon Vega Mobile Gfx
stepping	: 0
microcode	: 0x810100b
cpu MHz		: 1412.710
cache size	: 512 KB
physical id	: 0
siblings	: 8
core id		: 3
cpu cores	: 4
apicid		: 7
initial apicid	: 7
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb hw_pstate sme ssbd sev ibpb vmmcall fsgsbase bmi1 avx2 smep bmi2 rdseed adx smap clflushopt sha_ni xsaveopt xsavec xgetbv1 xsaves clzero irperf xsaveerptr arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif overflow_recov succor smca
bugs		: sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 4392.20
TLB size	: 2560 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 43 bits physical, 48 bits virtual
power management: ts ttp tm hwpstate eff_freq_ro [13] [14]
#-----------------------------------------------------------------------------------------------------------------------------------#
Disabling DisplayLink-ed Screens immediately drops cpu usage to normal.

Would an strace on the DisplayLinkManager proc be helpful? Didn't see anythind suspicious there.
The Logs are just full of encoded stuff:
#-----------------------------------------------------------------------------------------------------------------------------------#
==> /var/log/displaylink/DisplayLinkManager.log <==
r8K1cRZIx2XS3WnebKWwisIJlcSX007n0JG3u5c6tTkTwaw56Rcq1IeX4qdDeGLm9M1dDZU3rnKdGO+usIBcIt6UDhOCtW1Cin4EMdsNQE127dVktvq+GB2ef4t34gTPwYTmWOBoV5tC0QhA8WQeGEW3kbbK92qdYpY6AkdSXtVEo0qZUQ+n4xdTe8MqjmF8BdHrqZjG8IAvbnX3YNcUE15BfuaurMvEZCaG3EtFq/SnDFPoULjcTaiI9biTK2WWlt3TyBC84z9HxNpTWmFISmLpaDrvFfD3ji0FcBDigY4nh5GRGG+vuDpdBcWyk3XO

==> /var/log/displaylink/FirmwareTrace.log <==
[... cut ...]
Dn6Ai3GxyWRxXYxivB1TBna4AEjK2hod9Lnlbe5dQoKydy3N9tBOMHVjN/HBBmbaIDH5fBUmViLf/aA4udQjDqkmoRShhcC4JSnhIUfFgXFmB0ZveLb3l+5LHyDqUQoHAuOBOHwXKKBx/yCkd3B+J6uAlqfVGIpZWYZ/sa9jcg7XpKc0AW3wAs1oqOhYwJwKuFwj3U+n8qksTS1Cb276SgsXhadhBuVc0oHvUJK8sUFeIUwQHMpS8M1OT7o63BPPB2Rd0+xXMHoaaBAsYvHwPmOGrLcj5+J8qo7y3SKPd2LL5FKp7dPd8CYdURK7aQ0u0sEUiqjCpi6NBibt6Bw/BQRJVgVj3iknt36SiDWqp9RnpboTOJTSomAaxW2zqy/jhKZ2H7yDJNvj2SgQtMVBSA==
#-----------------------------------------------------------------------------------------------------------------------------------#

I am happy to provide any further info that you may need...

breytex, 2019-10-26:
@moddie666  have you been able to fix that problem? I guess I have the exact same issue with my T495 with amd graphics...