AdnanHodzic/displaylink-debian #206: DisplayLink blocks Nvidia driver

dpcat237, 2019-01-27:
System: Ubuntu 18.04, Nvidia GeForce MX150 (Nvidia driver 415), Intel i7 8550
Problem:
After installing DisplayLink, Nvidia driver doesn't work properly and Ubuntu automatically switches to Intel GPU while Nvidia prime still should Nvidia as selected GPU. I tried to install Nvidia driver 390, 396, 410, 415 (from Ubuntu ppa) with the same effect. The driver is installed without any error. During installation Nvidia driver 410 downloaded from nvidia.com displays this error `Unable to load the 'nvidia-drm' kernel module`.
PS. DisplayLink with Intel works well without any lag.

If I remove DisplayLink driver all above problems disappear.

AdnanHodzic, 2019-01-27:
Could you give me back the output of `sudo ./displaylink-debian.sh --debug` after driver is installed? 

Also make sure that there's no other xorg.conf related configs besides: `/etc/X11/xorg.conf.d/20-displaylink.conf` in `/etc/X11/` or `/usr/share/X11/xorg.conf.d/`? 

Also could you paste the contents of `/usr/share/sddm/scripts/Xsetup` file?

dpcat237, 2019-01-28:
Debugging information:
```
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.15.0-43-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24
414 
414 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: UHD
VGA: Intel Corporation UHD Graphics 620 (rev 07)
X11 version: 1.19.6-1ubuntu4.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Intel
```

How can I check about `xorg.conf` configs?
This folder `/usr/share/sddm` doesn't exist.

AdnanHodzic, 2019-01-29:
If you pull the latest changes and run debug again (and sharing it here), with the changes I've just made all related xorg.configs will be shown.

What happens if you put following as your `etc/X11/xorg.conf.d/20-displaylink.conf` contents:

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

Followed by logout/login or reboot.

dpcat237, 2019-01-29:
After pulling last changes:
```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.15.0-43-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: UHD
VGA: Intel Corporation UHD Graphics 620 (rev 07)
X11 version: 1.19.6-1ubuntu4.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Intel
```

After changing content of `etc/X11/xorg.conf.d/20-displaylink.conf` and rebooting, Nvidia works properly. So, this is manual fix. Debugging result:
```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.15.0-44-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: i915
Subsystem: UHD
VGA: Intel Corporation UHD Graphics 620 (rev 07)
X11 version: 1.19.6-1ubuntu4.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x23b cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 1 name:NVIDIA-0
Provider 1: id: 0x44 cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 2 associated providers: 1 name:modesetting
```
 
I have `nvidia-driver-415`. I searched `sddm` folder and it doesn't exist in all system. Also, this file `Xsetup` doesn't exist.

AdnanHodzic, 2019-01-30:
I just pushed the changes, so from now when your type of graphics card is detected changes you I suggested you make manually will now be done automatically for: `/etc/X11/xorg.conf.d/20-displaylink.conf`

Regarding mouse/cursor flicker issue, check [Troubleshooting most common issues](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#troubleshooting-most-common-issues) which has that problem mentioned.

Judging by Ubuntu debug output your monitors have been detected, they just need to be configured I suggest consulting [Screen layout configuration](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#screen-layout-configuration).

AdnanHodzic, 2019-01-30:
As part of https://github.com/AdnanHodzic/displaylink-debian/commit/b57cea89cc2e97de071e417bdd39ddd374072dbd commit udl/udlf will be added to blacklist.conf automatically if >= 4.14.9 is detected. 

I'm not sure if this will fix your problem, but I'd definitely advise you to pull the latest changes and perform reinstall and report back if there any changes.

dpcat237, 2019-01-31:
After re-installing last changes and changing `/etc/X11/xorg.conf.d/20-displaylink.conf` to original content, I have same problem as at the beginning. So, setting Nvidia Ubuntu uses Intel GPU.

AdnanHodzic, 2019-01-31:
Yes, from your debug output I've noticed there's no mention of Nvidia but only Intel card.

There was a similar problem with another user with Nvidia card: https://github.com/AdnanHodzic/displaylink-debian/issues/201

His problem was that Xsetup file didn't have executable permissions, which I fixed as part of last commit.

But you said you didn't have `/usr/share/sddm/scripts/Xsetup` file at all. So what would happen if you created that file manually and put following contents in it:

```
#!/bin/sh
# Xsetup - run as root before the login dialog appears

if [ -e /sbin/prime-offload ]; then
    echo running NVIDIA Prime setup /sbin/prime-offload
    /sbin/prime-offload
fi

xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --auto
```

followed by sudo `chmod +x /usr/share/sddm/scripts/Xsetup`

Of course in this case you would also have to change `/etc/X11/xorg.conf.d/20-displaylink.conf` file contents to:

```
Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection
```

Make this changes perform restart and let me know what happens then.

dpcat237, 2019-01-31:
After doing commented changes Nvidia works properly also with external display. This doesn't affect commented bugs with external display is connected.

AdnanHodzic, 2019-01-31:
What's the output of `sudo ./displaylink-debian.sh --debug` with Nvidia working as it should now? I want to make code changes so this will be done automatically during `displaylink-debian` install.

Regarding bugs you mentioned, I suggested what you can do but I haven't heard any feedback from you regarding this: https://github.com/AdnanHodzic/displaylink-debian/issues/206#issuecomment-458889014

I can help with installation of this driver, I can't help you with how to configure your system and/or resolve Nvidia and Evdi bugs, for this I would point you to report any upstream problem: https://github.com/DisplayLink/evdi/issues

dpcat237, 2019-01-31:
```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.15.0-44-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: i915
Subsystem: UHD
VGA: Intel Corporation UHD Graphics 620 (rev 07)
X11 version: 1.19.6-1ubuntu4.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x23b cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 2 name:NVIDIA-0
Provider 1: id: 0x277 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x44 cap: 0x2, Sink Output crtcs: 3 outputs: 2 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```
I have one monitor connected. So, in total are two. I don't know why are 3 providers.

Okay, thank you. I'll write them because after trying two solutions commented there made things worth.

AdnanHodzic, 2019-02-02:
Great, in that case I'm closing this issue. Feel free to re-open it in case you still have the same problem re-occurs or if you have any updates.

Also as it was pointed to me in https://github.com/AdnanHodzic/displaylink-debian/issues/201. As part of last commit https://github.com/AdnanHodzic/displaylink-debian/commit/c185aa3430a1849d0b7f700c972d543df0ff28f1 I made the changes so Xsetup file will be created with necessary contents in case it's not there.

closed 2019-02-02 by AdnanHodzic