AdnanHodzic/displaylink-debian #207: Only one xrandr provider on Debian sid with 4.19 kernel

coroa, 2019-01-28:
Hi,

first installation on an up-to-date Debian sid. Installation succeds without errors (ie. kernel module is built, `dlm.service` is registered and `20-displaylink.conf` contains a device section for my main `intel` card).

After a reboot:
- the `dlm` service is `active (running)`.
- DisplayLink is registered as a framebuffer device
 ```
➜  src sudo dmesg | grep Display
[   16.958407] usb 2-2: Manufacturer: DisplayLink
[   18.482325] usb 2-2: fb1 is DisplayLink USB device (1920x1080, 8104K framebuffer memory)
```
- But, does not show up in `Xorg.0.log`
```
➜  src grep Display /var/log/Xorg.0.log 
[    18.891] (II) intel(0): Creating default Display subsection in Screen section
[    18.899] (==) intel(0): Display hotplug detection enabled
```
- And, there is only one `xrandr` provider:
```
➜  src xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 0 name:Intel
```
- The display is on and a bright green.
- Removing `20-displaylink.conf` from `xorg.conf.d` does NOT change the outcome.
- `/var/log/displaylink` is there and fills up with info:
```
➜  displaylink-debian git:(master) tail /var/log/displaylink/*           
==> /var/log/displaylink/DisplayLinkManager.log <==
[... abbreviated ...] /BxTlP0XrEjwwJbgzD7z0QTHh+hQ5DbJUWzoiJU7hDQyJSXdQis+IvELWuSuu4V0tSzGBvtqsBz7d1GBUqn9hEUJxQsO07mlYMZHEFbaeSbUq/cjp/7lB+jayruZqPmj9uyQdKUltOWMVtmdK+Q4+ICQe3ro3Cq6VMBWGF69mdOb6dTv13IQsRtOB1HswgHMAuJCSRFYBE6M/r/jY=

==> /var/log/displaylink/W2443_GSM571C-2011.1.0006E4D8.edid <==
ÿÿÿÿÿÿmWØäê3xê®ŢWJ%PT¥K³qO6 p8@0 5þ":q8-@X,Eþ"ý8KS
      üW2443
       Ê%                                                                                                                                                           
```

The debugging output of `displaylink-debian.sh` is:
```
➜  displaylink-debian git:(master) sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: sid
Kernel: 4.19.0-2-amd64

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [17aa:2214]
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
X11 version: 1.20.3-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 0 name:Intel

-------------------------------------------------------------------
```

Any ideas, what I can try?

AdnanHodzic, 2019-01-28:
I think something went wrong with detecting your graphics card, what if you made following to be part of your `/etc/X11/xorg.conf.d/20-displaylink.conf` file:

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

or in case that doesn't work:

```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

Make sure after making either of those changes to perform a logout/login (X restart) or reboot.

Also, what is the output of: `lspci | grep VGA`

coroa, 2019-01-28:
Looks a bit like a duplicate of https://github.com/AdnanHodzic/displaylink-debian/issues/198 .

I added the `DisplayLink` configurations you suggested to `20-displaylink.conf` (i kept the `intel` section intact), no net change with either of them. The latter, where I explicitly declare the `DisplayLink` device produces a small mention in `/var/log/Xorg.0.log`:
```
[   350.248] (==) No screen section available. Using defaults.
[   350.248] (**) |-->Screen "Default Screen Section" (0)
[   350.248] (**) |   |-->Monitor "<default monitor>"
[   350.248] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[   350.248] (**) |   |-->Device "Intel"
[   350.248] (**) |   |-->GPUDevice "DisplayLink"
[   350.248] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[   350.248] (==) Automatically adding devices
[   350.248] (==) Automatically enabling devices
[   350.248] (==) Automatically adding GPU devices
```

There is still only one provider.

```
➜  ~ lspci | \grep --color VGA
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
```

Thanks for the amazing response time!

coroa, 2019-01-28:
Ok, found the helpful remark in the [Archwiki](https://wiki.archlinux.org/index.php/DisplayLink) to blacklist `udlfb` if the kernel is new enough and `udl` is available. That solved the issue completely for me! DisplayLink is running fine without any xorg configuration. I only have to issue `xrandr --setprovideroutputsource 1 0` after plugging it in.

Thanks

AdnanHodzic, 2019-01-29:
Great, glad you figured it out I also updated [post install guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md) with this information

coroa, 2019-01-29:
Since we made it into the post install guide, let's add a couple of comments:
- As described in the ArchWiki linked above, the kernel modules `udlfb` and `udl` have been added to the linux kernel around version 4.14.9. They are the open variant of the `evdi` displaylink driver, which is installed by the scripts in this repository.
- At least on my system the kernel-internal ones took precedence: the first one to grab hold of the device was `udlfb` (which exposed the displaylink as a framebuffer device), after blacklisting it (see below), `udl` was next (which gave me a working but a bit sluggish and tearing display), and after blacklisting `udl` and `udlfb`, `evdi` took over as intended and worked best.

So to get it working on Debian sid with a recent kernel, I had to add:
```
blacklist udlfb
blacklist udl
```
to `/etc/modprobe.d/blacklist.conf` to ensure the device was managed by `evdi`.

I am not using any xorg configuration anymore, ie. I removed the `/etc/X11/xorg.conf.d/20-displaylink.conf` (but it does also work if it is there).

To debug which of the drivers is used you can examine the output of `dmesg`. For example, with `evdi` it should look something like
```
➜ sudo dmesg | grep -E "udl|evdi"   
[   18.111148] evdi: loading out-of-tree module taints kernel.
[   18.111225] evdi: module verification failed: signature and/or required key missing - tainting kernel
[   18.113078] evdi: [I] Initialising logging on level 5
[   18.113080] evdi: [I] Atomic driver:no
[   18.610555] evdi: [D] evdi_add_devices:136 Increasing device count to 1
[   18.610755] evdi: [D] evdi_crtc_init:353 drm_crtc_init: 0
[   18.610837] evdi: [W] evdi_painter_crtc_state_notify:486 Painter does not exist!
[   18.610844] evdi: [D] evdi_detect:88 Painter is disconnected
[   18.610854] evdi: [W] evdi_painter_crtc_state_notify:486 Painter does not exist!
[   18.611074] [drm] Initialized evdi 1.5.1 20180731 for evdi.0 on minor 1
[   18.611077] [drm] evdi: evdi_stats_init
[   18.632599] evdi: [D] evdi_painter_connect:574 (dev=1) Connected with 0000000070affa3c
[   18.632604] evdi: [D] evdi_detect:85 (dev=1) Painter is connected
[... and so on ...]
```

AdnanHodzic, 2019-01-30:
As part of last commit https://github.com/AdnanHodzic/displaylink-debian/commit/b57cea89cc2e97de071e417bdd39ddd374072dbd, udl/udlf will be added to blacklist.conf automatically if >= 4.14.9 is detected.

Please re-open the issue if the problem persists.

closed 2019-01-30 by AdnanHodzic