AdnanHodzic/displaylink-debian #212: Deepin 15.9

sticky99, 2019-02-04:
Won't install as its seems to be looking for the wrong linux-headers package, ie its looking for linux-headers-4.15.0-29deepin-generic-deepin instead of just linux-headers-4.15.0-29deepin-generic

stevo@DeepinLinux:~$ chmod +x displaylink-debian.sh && sudo ./displaylink-debian.sh

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for stevo: 

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: I

-------------------------------------------------------------------

Checking dependencies

unzip is installed
dpkg-query: package 'linux-headers-4.15.0-29deepin-generic-deepin' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
linux-headers-4.15.0-29deepin-generic-deepin not found! Install? [Y/n] n

-------------------------------------------------------------------

Cannot continue without linux-headers-4.15.0-29deepin-generic-deepin.  Aborting.

-------------------------------------------------------------------

stevo@DeepinLinux:~$ sudo apt-get install linux-headers-4.15.0-29deepin-generic-deepin
[sudo] password for stevo: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.15.0-29deepin-generic-deepin
E: Couldn't find any package by glob 'linux-headers-4.15.0-29deepin-generic-deepin'
E: Couldn't find any package by regex 'linux-headers-4.15.0-29deepin-generic-deepin'
stevo@DeepinLinux:~$ uname -r
4.15.0-29deepin-generic
stevo@DeepinLinux:~$ sudo apt search linux-headers-$(uname -r)
Sorting... Done
Full Text Search... Done
linux-headers-4.15.0-29deepin-generic/panda,now 4.15.0-29deepin.31 amd64 [installed]
  Linux kernel headers for version 4.15.0 on 64 bit x86 SMP

stevo@DeepinLinux:~$ sudo apt-get install linux-headers-4.15.0-29deepin-generic
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-headers-4.15.0-29deepin-generic is already the newest version (4.15.0-29deepin.31).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

AdnanHodzic, 2019-02-04:
In last commit https://github.com/AdnanHodzic/displaylink-debian/commit/e933ce82abe3d5c429d3dc995e9373e6961c89ee I made the changes. If this doesn't fix your problem feel free to re-open the issue.

sticky99, 2019-02-05:
So now the installation completed, but getting the screen recognized doesn't work

`stevo@DeepinLinux:~$ lsusb | grep -i Display
Bus 001 Device 003: ID 17e9:037b DisplayLink Plugable USB-VGA-165
`

`stevo@DeepinLinux:~$ lsusb | grep -i Display
Bus 001 Device 003: ID 17e9:037b DisplayLink Plugable USB-VGA-165
`

The default 20-displaylink.conf only had the intel component, and only a single provider was shown. Adding the Section for Displaylink now results in the second provider being shown

`stevo@DeepinLinux:~$ xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x87 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 3 associated providers: 0 name:modesetting
Provider 1: id: 0x41 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
stevo@DeepinLinux:~$ 
`

`stevo@DeepinLinux:~$ sudo ./displaylink-debian.sh --debug
[sudo] password for stevo: 

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Deepin
Release: unstable
Kernel: 4.15.0-29deepin-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4744
4.4.24
4748
4748

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Generation
VGA: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.20.0-2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/75-wacom.conf /etc/X11/xorg.conf.d/50-synaptics.conf /etc/X11/imwheel/startup.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 #Section "Device"
#    Identifier  "Intel"
#    Driver      "intel"
#EndSection

Section "Device"
	Identifier "DisplayLink"
	Driver "modesetting"
	Option "PageFlip" "false"
EndSection

-------------------- Monitors --------------------
[... 1553 lines cut ...]
[    4.473137] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1b.0/sound/card0/input13
[    4.696247] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    4.698122] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    4.716588] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    4.716591] cfg80211: failed to load regulatory.db
[    5.426837] IPv6: ADDRCONF(NETDEV_UP): eno1: link is not ready
[    5.496691] evdi: [I] Initialising logging on level 5
[    5.496693] evdi: [I] Atomic driver:no
[    5.648495] IPv6: ADDRCONF(NETDEV_UP): eno1: link is not ready
[    9.198271] e1000e: eno1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[    9.198316] IPv6: ADDRCONF(NETDEV_CHANGE): eno1: link becomes ready
[    9.344175] NET: Registered protocol family 15
[    9.368973] vboxdrv: Found 8 processor cores
[    9.385177] Initializing XFRM netlink socket
[    9.388505] vboxdrv: TSC mode is Invariant, tentative frequency 2793640333 Hz
[    9.388508] vboxdrv: Successfully loaded version 5.2.12_Debian (interface 0x00290001)
[    9.397673] VBoxNetFlt: Successfully started.
[    9.404738] VBoxNetAdp: Successfully started.
[    9.415171] VBoxPciLinuxInit
[    9.420384] vboxpci: IOMMU not found (not registered)
[   14.759373] random: crng init done
[   82.672587] udl 1-1.2:1.0: swiotlb buffer is full (sz: 348160 bytes)
[   82.672589] udl 1-1.2:1.0: DMA: Out of SW-IOMMU space for 348160 bytes
[   83.076102] lightdm[6594]: segfault at 0 ip 00007f981e593086 sp 00007ffc817ab728 error 4 in libc-2.27.so[7f981e4fd000+1b1000]
`

sticky99, 2019-02-06:
I deleted those additional related configs, eg 75-wacom.conf, 50-synaptics.conf, startup.conf based on not having a Wacom device connected, it being a desktop rather than a laptop and thus not having a touchpad, and only having a standard two-button mouse with scroll wheel. Rebooted, but no change.

So changed the 20-displaylink from/to as shown in this new pastebin

https://pastebin.com/K4zi2Qaf

Rebooted, Did the debug to check everything looked ok, then ran arandr but still no change, I don't see the DVI-I-1-1 output again. I've still no idea why it worked that once, but never since.

AdnanHodzic, 2019-02-06:
Doesn't matter, I made `displaylink-debian` in such a way that it should make all changes based on detected components. 

However files from  `/usr/share/X11/xorg.conf/` shouldn't be used. However if there is a conflict then it's because of these two files, so you can either move them to your home directory or remove them completely just to be sure:

`10-amdgpu.conf`:

```
Section "OutputClass"
    Identifier "AMDgpu"
    MatchDriver "amdgpu"
    Driver "amdgpu"
EndSection
```

and `10-radeon.conf`

```
Section "OutputClass"
    Identifier "Radeon"
    MatchDriver "radeon"
    Driver "radeon"
EndSection
```

So you can move/remove these two followed by reboot and let me know if changes anything, but I doubt it.

Also judging by your `displaylink-debian --debug` output in https://pastebin.com/K4zi2Qaf your monitors are detected:

```
------------------- Monitors --------------------
 
Providers: number : 2
Provider 0: id: 0x87 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 3 associated providers: 0 name:modesetting
Provider 1: id: 0x41 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```

Not sure if you looked through [Screen Layout Configuration](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#screen-layout-configuration) but maybe you just need to properly configure the displays properly?

Also you can also try with changing contents of `/etc/X11/xorg.conf.d/20-displaylink.conf` file from:

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```
to:

```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

followed by reboot and see if it makes any difference but I don't think it will, but worth giving a try. If there are no changes revert back.

sticky99, 2019-02-07:
I probably didn't explain myself very well. I meant that while still running the Deepin OS off the SSD, I temporarily connected the HDD and mounted it so I could see if there was an xorg.conf or similar from Ubuntu that I could replicate in Deepin. I've searched, but couldn't find anything that I could see that could be replicated. So the 10-amdgpu.conf and 10-radeon.conf files don't exist within Deepin.

I actually have two screens connected at the moment that work off the embedded i915 GPU, and I've a third connected via the Displaylink USB that I'm trying to get working. As part of this I use a script to activate the third screen, and then arrange the screens in the order I want

> #!/bin/sh
>xrandr --output DVI-I-1-1 --auto
>xrandr --output VGA1 --mode 1920x1080 --pos 0x0 --rotate normal --output DVI-I-1-1 --mode 1920x1080 --pos 3840x0 --rotate normal --output HDMI1 --primary --mode 1920x1080 --pos 1920x0 --rotate normal

The problem is that when I want to activate the DVI-I-1-1 I get the following (other than that one time that I can't replicate since that actually was able to activate it successfully)

>stevo@DeepinLinux:~$ sudo xrandr --output DVI-I-1-1 --auto
>[sudo] password for stevo: 
>warning: output DVI-I-1-1 not found; ignoring

Using the xrandr command without any additional parameters just lists the two screens I have already working (HDMI-1 and VGA-1). I get the same effect in Arandr, that from the Output menu, it again only lists HDMI-1 and VGA-1.

So the disconnect as I understand it is between xrandr --listproviders detecting the GPU (since before I added the Displaylink section to the 20-displaylink.conf file it only provided a single provider) and the xrandr output not detecting the display.

Xrandr output shown at https://pastebin.com/mRGGAL1M

sticky99, 2019-02-09:
Some progress of sorts on this. I used the script to uninstall the dlm service, rebooted, reinstalled via the script etc. Brought the 20-displaylink.conf file back to what it used to be

> stevo@DeepinLinux:~$ more /etc/X11/xorg.conf.d/20-displaylink.conf
>#Section "Device"
>#    Identifier  "Intel"
>#    Driver      "intel"
>#EndSection
> 
>Section "OutputClass"
>   Identifier "DisplayLink"
>    MatchDriver "evdi"
>   Driver "modesetting"
>   Option "AccelMethod" "none"
>EndSection

Now when I use the command "xrandr --setprovideroutputsource 1 0" Xorg no longer crashes, it does however make both screens connected to the i915 GPU go completely black for 3-4 seconds. But once it comes back to normal, the DVI-I-1-1 output is seen in the xrandr output. I've done this a few times now, and it works everytime.

The problem is though that when I try to make the DVI-I-1-1 screen be active through either arandr or "sudo xrandr --output DVI-I-1-1 --auto" then Xorg does crash. I don't know if there's an alternative way around this?

Output including multiple Xorg crashes at dmesg is at https://pastebin.com/7XwLF9C9

AdnanHodzic, 2019-02-20:
I've just merged `displaylink-5` branch with `master` which introduces big changes, among which are support for yesterday released DisplayLink 5.1 driver. Disabling modesetting as part of `displaylink-debian` as now all of it should be happening on EVDI level. I've included few more changes.

Regardless, please pull latest (`git pull`) changes on `master` and run `sudo ./displaylink-debian.sh --install`

After this everything should be working as expected and with this said I'm closing the issue. If the issue still persists, re-open the issue and include output of `sudo ./displaylink-debian.sh --debug` command.

commit e933ce82 (2019-02-04): remove headers "deepin" postfix (closes: #212)
closed 2019-02-20 by AdnanHodzic