AdnanHodzic/displaylink-debian #213: Displays not detected

Zer07, 2019-02-04:
I am having trouble connecting to my two displays via plugable UD3900. The mouse and keyboard work fine but the system doesn't see the displays. I followed through post-installation steps and everything is ok till I get to the point where I am supposed to see my displays. 

Can you help?

I ran the debug and here is the information:

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.15.0-29-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Core
VGA: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.19.6-1ubuntu4.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:Intel

-------------------------------------------------------------------

AdnanHodzic, 2019-02-04:
What happens if you put following as contents of your `/etc/X11/xorg.conf.d/20-displaylink.conf` file:

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

Zer07, 2019-02-04:
I reinstalled ubuntu and downloaded the driver from displaylink directly after updating the system. It seems to have solved the issue.

AdnanHodzic, 2019-02-04:
Great, now that everything is working can you tell me what are the contents of your `/etc/X11/xorg.conf.d/20-displaylink.conf` file?

Zer07, 2019-02-07:
my /etc/X11/xorg.conf.d/ is empty. I just used the official driver. It recognizes my two screens connected to my dock, but only after I log in. Is there a way to set a persistent screen layout that loads early in the login process?  

I saw this in your post-installation guide. But I don't have /.bashrc directory.

# two
alias two="xrandr --setprovideroutputsource 1 0 && xrandr --setprovideroutputsource 2 0 && xrandr --output VIRTUAL1 --off --output DVI-1-0 --primary --auto --pos 0x0 --rotate normal --output DP1 --off --output HDMI2 --off --output HDMI1 --off --output eDP1 --off --output DVI-2-1 --auto --pos 1680x0 --rotate normal"

# one
alias one="xrandr --output VIRTUAL1 --off --output DVI-1-0 --off --output DP1 --off --output HDMI2 --off --output HDMI1 --off --output eDP1 --primary --mode 1366x768 --pos 0x0 --rotate normal --output DVI-2-1 --off"

AdnanHodzic, 2019-02-07:
Why are you submitting an issue here if you're using an official driver? You do realize what [displaylink-debian](https://github.com/AdnanHodzic/displaylink-debian) does right?

Zer07, 2019-02-15:
After I tried your contraption, which didn't work, I  installed the official version... read above:  "I reinstalled ubuntu and downloaded the driver from displaylink directly after updating the system. It seems to have solved the issue."

I understand I don't have the /.bashrc directory because it is what your script adds...

closed 2019-02-16 by AdnanHodzic