AdnanHodzic/displaylink-debian #215: chipset type supported?

zhilongjia, 2019-02-13:
which chipset types are supprted in this displaylink-debian? Thank you. 

DisplayLink has several chipset types, such as `DL-1x0`, `DL-1x` `DL-3xxx`, `DL-41xx`, `DL-5xxx`, `DL-6xxx` as shown in the [DisplayLink wiki](https://en.wikipedia.org/wiki/DisplayLink)

AdnanHodzic, 2019-02-16:
Goal of `displaylink-debian` is to support all chipsets and so far all of the chipsets are supported.

closed 2019-02-16 by AdnanHodzic

Darcidride, 2019-03-14:
@AdnanHodzic he list of supported shipset is definitely something to add in the main readme file, i was looking for this information before i bought my displaylink device and checked if it was working with your great work :)

AdnanHodzic, 2019-03-17:
@Darcidride there's another story in regards to this: https://github.com/AdnanHodzic/displaylink-debian/issues/226

If you're up for it, please do what I suggested in: https://github.com/AdnanHodzic/displaylink-debian/issues/226#issuecomment-467439973 (make changes in ReadME yourself and make a Pull Request and I'll be glad to review the changes and merge with master).

I'm closing this story, as let's continue any further discussion in #226