AdnanHodzic/displaylink-debian #216: Kali 19.1 + Nvidia GeForce GTX 1050M

ptspawn, 2019-02-14:
Changed the script from '2018.4' to 'n/a' to run the install but after the reboot, i had no more gui.
I'm running on Kali with KDE.
I was unfortunately forced to uninstall.

I'm running on a GeForce GTX 1050 Mobile and had to install hashcat-nivia over the nvidia driver to get the graphics card to work properly (otherwise it always freezes before shutting down). Always ran perfectly since.

debug file:

Starting Debug ...

--------------- Linux system info ----------------

Distro: Kali
Release: n/a
Kernel: 4.19.0-kali1-amd64

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation HD Graphics 630 (rev 04)
VGA (3D): NVIDIA Corporation GP107M [GeForce GTX 1050 Mobile] (rev a1)
X11 version: 1.20.3-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
Section "ServerLayout"
Identifier "layout"
Screen 0 "nvidia"
Inactive "intel"
EndSection

Section "Device"
Identifier "intel"
Driver "modesetting"
Option "AccelMethod" "None"
EndSection

Section "Screen"
Identifier "intel"
Device "intel"
EndSection

Section "Device"
Identifier "nvidia"
Driver "nvidia"
Option "ConstrainCursor" "off"
EndSection

Section "Screen"
Identifier "nvidia"
Device "nvidia"
Option "AllowEmptyInitialConfiguration" "on"
Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

AdnanHodzic, 2019-02-16:
Thanks for reporting this, I made these changes as part of https://github.com/AdnanHodzic/displaylink-debian/issues/216 and as part of last commit (https://github.com/AdnanHodzic/displaylink-debian/commit/12cad8fed686ca16384adf79ff7faf8ec2a6bca0).

Feel free to pull the latest changes and do re-install. If the problem persists after this, let me know so I can make any additional necessary changes.

Thanks

ptspawn, 2019-02-16:
Thank you for the quick reply and fix.
Sadly it didn't work. After installing and rebooting i still get stuck in terminal mode, after the login prompt flashes a few times like it's trying to get the gui running... :(

AdnanHodzic, 2019-02-17:
As I said in comments of (https://github.com/AdnanHodzic/displaylink-debian/commit/12cad8fed686ca16384adf79ff7faf8ec2a6bca0#commitcomment-32345283).

I did all the changes you mentioned as part of: #204. Are you sure you didn't anything else besides these two changes?

I'd even encourage you to make a "Pull request" based on `displaylink-debian` changes you made to make it work. I'd love to review these changes and merge them with master.

ptspawn, 2019-02-18:
Hey Adnan.

Those were the only two changed I made to run the script yes, but I haven't been able to get it working so far.
I think I'll try to compare the changes the original script makes in Ubunto to the ones your script is making in Kali, but not being sure what's breaking it (my money is on KDE, but using a non standard NVIDIA driver (hashcat) like I was forced to use to get kali running is also a big contender in my opinion), won't make my job easy. Hoped someone else with the hashcat install of nvidia would show up here with displaylink up and running so I could compare.

Regardless, If I do figure this out, I'll be more that happy to share it with you. Kudos by the way, on the script!

AdnanHodzic, 2019-02-18:
Ah, okay in that case I've just disabled execution of `nvidia_hashcat` function in `displaylink-debian` as I thought if fixed it for you. Sorry for the misunderstanding. 

Since Kali is based on Debian I'd advise you to refer to [Debian Nvidia wiki page](https://wiki.debian.org/NvidiaGraphicsDrivers).

But since I don't use KDE or have a NVIDIA card I really can't be much more of help. So good luck and let me know if you have any updates.

Btw thanks for the compliments :)

lhy0403, 2019-02-20:
I accidentally removed the `20-DisplayLink.conf` earlier today, and it did the trick.
I don't know why it's like that, but I think it worths a try...
Don't forget to back up before `rm`

AdnanHodzic, 2019-02-20:
I've just merged `displaylink-5` branch with `master` which introduces big changes, among which are support for yesterday released DisplayLink 5.1 driver. Disabling modesetting as part of `displaylink-debian` as now all of it should be happening on EVDI level. I've included few more changes.

Regardless, please pull latest (`git pull`) changes on `master` and run `sudo ./displaylink-debian.sh --install`

After this everything should be working as expected and with this said I'm closing the issue. If the issue still persists, re-open the issue and include output of `sudo ./displaylink-debian.sh --debug` command.

closed 2019-02-20 by AdnanHodzic