AdnanHodzic/displaylink-debian #217: Ubuntu 18.04 No Monitors detected

henry-aj-jones, 2019-02-14:
In relation to #137 

When using display link and having the dock plugged in no monitors are detected. 

Output from: xrandr --listproviders
Providers: number : 0

Output from Debug: 

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.15.0-44-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: 
[1180]:
GTX
VGA: Intel Corporation Device 591b (rev 04)
NVIDIA Corporation GP107M [GeForce GTX 1050 Ti Mobile] (rev a1)
VGA (3D): 
X11 version: 1.19.6-1ubuntu4.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Output from: lspci | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation Device 591b (rev 04)
01:00.0 VGA compatible controller: NVIDIA Corporation GP107M [GeForce GTX 1050 Ti Mobile] (rev a1)

Output From: lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
nvidia

Output From :lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5

[1180]:
GTX

henry-aj-jones, 2019-02-14:
Output from: echo $XDG_SESSION_TYPE
x11

AdnanHodzic, 2019-02-16:
Contents of `/etc/X11/xorg.conf.d/20-displaylink.conf` file weren't populated properly.

I just made changes as part of last commit (`https://github.com/AdnanHodzic/displaylink-debian/commit/6936c2c7d437ce04205f99a0b11817b9dd36f9c7`)

You need to pull latest changes (`git pull`) followed by re-install, let me know what happens after that.

henry-aj-jones, 2019-02-18:
Thank you for your response, I have got it working with the normal display link drivers for now. I appreciate this isn't as effective as the solution you are providing. Thank you for all you help however!

closed 2019-02-18 by henry-aj-jones