AdnanHodzic/displaylink-debian #218: Request support for Mx Linux 18

cloudostrich, 2019-02-14:
Please add support for mx linux 18. When I try to install, I get the message 'linux-source' not found. I am assuming this got something to do with the kernel for mx linux being 4.19.0-1-amd64 (am using 64 bit). mx linux is now easily the top dstro in distrowatch, mostly due to mx linux 18!! ( i am guessing). If there is support for 18, it will just explode in popularity!.... ok, extrapolating... but still, please support 18!!!

AdnanHodzic, 2019-02-16:
Sure, no problem :)

To help me do this, I need output of `sudo ./displaylink-debian.sh --debug`?

Also what output do you get if you run `apt search linux-source`?

AdnanHodzic, 2019-02-20:
I've just merged `displaylink-5` branch with `master` which introduces big changes, among which are support for yesterday released DisplayLink 5.1 driver. Disabling modesetting as part of `displaylink-debian` as now all of it should be happening on EVDI level. I've included few more changes.

Regardless, please pull latest (`git pull`) changes on `master` and run `sudo ./displaylink-debian.sh --install`

After this everything should be working as expected and with this said I'm closing the issue. If the issue still persists, re-open the issue and include output of `sudo ./displaylink-debian.sh --debug` command.

AdnanHodzic, 2019-02-20:
@cloudostrich JFYI, if I don't get the information I requested, I'll close this issue in new few days due to inactivity.

jjxxzz, 2019-02-25:
Hi,

I'm also interested in implementation, so here's my debug info! :)

$ sudo ./displaylink-debian.sh --debug
Failed to retrieve unit: No such method 'GetUnit'

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: Nie ma takiego pliku ani katalogu
--------------- Linux system info ----------------

Distro: MX
Release: Continuum
Kernel: 4.15.0-1-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 5500
VGA: Intel Corporation HD Graphics 5500 (rev 09)
VGA (3D): 
X11 version: 1.19.2-1+deb9u5
X11 configs: /etc/X11/xorg.conf.d/10-evdev.conf /etc/X11/xorg.conf.d/synaptics.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Nie ma takiego pliku ani katalogu
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
BR,
J

dkebler, 2019-02-25:
Given that mint 18 and 19 have ubuntu 18.04 under the hood and this is supported it should work.  I have found more times than not that install scripts make use of `lsb_release -a` and  linux mint folk mess with what that returns.  

```
lsb_release -a
No LSB modules are available.
Distributor ID:	LinuxMint
Description:	Linux Mint 19.1 Tessa
Release:	19.1
Codename:	tessa
```

this file below on the other hand gives back the underlying ubuntu release.
I haven't looked at your script but it may need to parse this file if uses `lsb_release` and finds it says linux-mint.  I have edited a number of generic install scripts to fix this issue. 
```
cat /etc/upstream-release/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04 LTS"
 ```

btw the `uname -a` does report the right kernel 4.15.x (at least for my updated 19.3 linux mint install)

AdnanHodzic, 2019-02-25:
@jjxxzz added support for MX Continuum in last commit. Pull the latest changes and run install.

@dkebler support for Mint 19.1 Tessa is already in `displaylink-debian`.  Let me know if it doesn't work for you

closed 2019-02-25 by AdnanHodzic

jjxxzz, 2019-02-25:
@AdnanHodzic Thanks for fast reaction time!

Sorry, but install would not finish properly, please check the message below

Installing driver version: 5.1

DisplayLink Linux Software 5.1.26 install script called: install
/sbin/init: invalid option -- '-'
Usage: init {-e VAR[=VAL] | [-t SECONDS] {0|1|2|3|4|5|6|S|s|Q|q|A|a|B|b|C|c|U|u}}
Distribution discovered: MX 18.1 Continuum
Unsatisfied dependencies. Missing component: Linux headers for running kernel, 4.15.0-1-amd64.
This is a fatal error, cannot install DisplayLink Linux Software.
Removing Displaylink Xorg config file

-------------------------------------------------------------------