AdnanHodzic/displaylink-debian #219: Ubuntu 18.04 cannot activate the monitor

AngeloTrotta85, 2019-02-15:
Dear all,
when using display link and having the dock plugged I see that the system detects the monitor, but I'm not able to use it.
I have an HP-Elitebook-8440p laptop with an internal NVIDIA graphic card and one external USB2VGAPRO2. The NVIDIA manages two monitor (the on-board monitor and one external) and a third one managed by the external card.

I followed the steps in "https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md" but I have problems when setting provider sources:
- xrandr --setprovideroutputsource 1 0    ---> OK
- xrandr --setprovideroutputsource 2 0    ---> OK

In attachment there is the requested log. Here also 'listmonitors':
```
xrandr --listmonitors
xrandr: Output DVI-I-1-1 is not disconnected but has no modes
Monitors: 2
 0: +*DP-3 1600/310x900/174+0+0  DP-3
 1: +VGA-0 1280/338x1024/270+1600+0  VGA-0
```
Output DVI-I-1-1 is the monitor that I'm not able to activate.

Thank you
  Best

[log-displaylink.log](https://github.com/AdnanHodzic/displaylink-debian/files/2869215/log-displaylink.log)

AdnanHodzic, 2019-02-16:
To help you resolve this problem I'll need output of `sudo ./displaylink-debian.sh --debug` command.

As part of last couple of commits (https://github.com/AdnanHodzic/displaylink-debian/commit/12cad8fed686ca16384adf79ff7faf8ec2a6bca0 and https://github.com/AdnanHodzic/displaylink-debian/commit/6936c2c7d437ce04205f99a0b11817b9dd36f9c7) I made some changes for Nvidia cards which could possibly resolve your issue.

Make sure to pull the latest changes on `master` and run re-install.

AdnanHodzic, 2019-02-20:
I've just merged `displaylink-5` branch with `master` which introduces big changes, among which are support for yesterday released DisplayLink 5.1 driver. Disabling modesetting as part of `displaylink-debian` as now all of it should be happening on EVDI level. I've included few more changes.

Regardless, please pull latest (`git pull`) changes on `master` and run `sudo ./displaylink-debian.sh --install`

After this everything should be working as expected and with this said I'm closing the issue. If the issue still persists, re-open the issue and include output of `sudo ./displaylink-debian.sh --debug` command.

closed 2019-02-20 by AdnanHodzic