AdnanHodzic/displaylink-debian #220: Rendering Issues on Second External Monitor with Ubuntu 18.10

matthew-davis, 2019-02-15:
Working on a Dell XPS 13 Model 9380 using Ubuntu 18.10 and an Intel graphics card.

I have successfully installed the display link drivers via your awesome script, I had a few squabbles, but found everything I needed in your Post-Install Guide or your Troubleshooting. I am now working with the laptop shut on one external monitor. The problem is I have two external monitors. The second external monitor is being recognized and connected to, however it appears to be displaying the issues being discussed 

I have however seemed to run into the same issue being discussed in Issue 68: https://github.com/AdnanHodzic/displaylink-debian/issues/68 Black Screen, mouse visible. There seems to be some sort of redraw issue. The tips in issue 68 don't appear to be working for me.

--------------- Linux system info ----------------

Distro: Ubuntu
Release: cosmic
Kernel: 4.18.0-15-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 3ea0
VGA (3D): 
X11 version: 1.20.1-3ubuntu2.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
     Identifier  "DisplayLink"
     MatchDriver "evdi"
     Driver      "modesetting"
     Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0xa3 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 2 name:modesetting
Provider 1: id: 0x7d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

explanation

I seem to have reached the end of what xrandr can do for me. My gut is telling me this is a xorg config file issue. Unfortunately I can't find a good enough (read dumbed down enough) explanation of this config to begin exploring how to fix this. Any suggestions of what to try and or places to go to get more info on this issue. I'd love to be working off of two screens in the near future.

AdnanHodzic, 2019-02-16:
Great to hear you have installed it successfully. I also think problem is with your xorg.conf, with this said, what would happen if you made put following as contents of your `/etc/X11/xorg.conf.d/20-displaylink.conf` file followed by reboot?

```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

matthew-davis, 2019-02-18:
Thanks for getting back to me. Your advice, however, didn't have an effect. The same issues were presenting themselves. So I decided to really work through issue 68 and try every little mention of an idea. And went through a series of attempted config file contents.

I eventually got to a working state of affairs, and along the way learned a few bits. First, perhaps it is just me and I'm not exactly sure why as it should just be using xrandr commands through a gui, but Arandr just took a dump in the bed every time I used it to apply changes. I'm not good enough to say that it is broken, but in my case, it was a big set back.

I eventually settled on the initial 20-displaylink.conf setting:
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```
There seemed to be an issue where the xrandr commands I had set in my alias crossed with the 20-display.conf file setup and the native Gnome settings for the displays that I eventually saw they were all connected and a combination of going back and forth through all of them got me to a state of not only the monitors being recognized, but being used correctly.

I think the key missing ingredient was that the system settings wasn't set to display the monitors as extensions of a desktop (despite setting --right-of or --pos commands in xrandr.

Thanks for your help.

matthew-davis, 2019-02-18:
[Section "OutputClass"Identifier "DisplayLink"MatchDriver "evdi"Driver "modesetting"Option "AccelMethod" "none"EndSection](url)

closed 2019-02-18 by matthew-davis