AdnanHodzic/displaylink-debian #221: Not working to  Product: BILLBOARD DEVICE   displaylink DA200

PedroFernandoLima, 2019-02-16:
Dmesg not displaylink is : Product: BILLBOARD DEVICE !!!????

HELP!

[  596.663424] usb 1-2.3: Product: BILLBOARD DEVICE
[  596.663428] usb 1-2.3: Manufacturer: BILLBOARD CLASS
[  596.663432] usb 1-2.3: SerialNumber: DCF96B0000000000E26B474000000000

-------------- Linux system info ----------------

Distro: LinuxMint
Release: tessa
Kernel: 4.15.0-20-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.19.6-1ubuntu4.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x64 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0x3f cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

AdnanHodzic, 2019-02-17:
Based on what I see your monitors have been detected, have you checked [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md)? In particular prat about "Setting up providers" and "Screen layout configuration"

PedroFernandoLima, 2019-02-18:
Setting up providers" and "Screen layout configuration".

NOT WORKING, is AMD RADEON 535.

[ IMAGE ERRO](https://uploaddeimagens.com.br/imagens/captura_de_tela_em_2019-02-18_11-48-22-png)

AdnanHodzic, 2019-02-18:
Well that's the reason why it's not working, as it `displaylink-debian` only detected your Intel card. Could you give me whole output of ` lspci` command so I can make improvements in detection of AMD cards.

Also, you need to change contents of `/etc/X11/xorg.conf.d/20-displaylink.conf` file to:

```
Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection
```

Followed by reboot and let me know what that does for you?

AdnanHodzic, 2019-02-20:
I've just merged `displaylink-5` branch with `master` which introduces big changes, among which are support for yesterday released DisplayLink 5.1 driver. Disabling modesetting as part of `displaylink-debian` as now all of it should be happening on EVDI level. I've included few more changes.

Regardless, please pull latest (`git pull`) changes on `master` and run `sudo ./displaylink-debian.sh --install`

After this everything should be working as expected and with this said I'm closing the issue. If the issue still persists, re-open the issue and include output of `sudo ./displaylink-debian.sh --debug` command.

closed 2019-02-20 by AdnanHodzic