AdnanHodzic/displaylink-debian #222: Parrot sec support

jf6921668, 2019-02-18:
Please add support for Parrot security

AdnanHodzic, 2019-02-18:
Sure, could you please add output of following command: `sudo ./displaylink-debian.sh --debug`

AdnanHodzic, 2019-02-20:
@jf6921668 JFYI, if I don't get the information I requested, I'll close this issue in new few days due to inactivity.

jf6921668, 2019-02-20:
@AdnanHodzic one min ill get it for you

jf6921668, 2019-02-20:
@AdnanHodzic im going to do a quick pull on ur new release and edit it to support parrot if it works ill let u know

jf6921668, 2019-02-20:
it didnt work so here you go

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: No such file or directory
--------------- Linux system info ----------------

Distro: Parrot
Release: n/a
Kernel: 4.19.0-parrot1-13t-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Processor
VGA: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Graphics & Display (rev 0e)
VGA (3D): 
X11 version: 1.20.3-1
X11 configs:

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 4 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

AdnanHodzic, 2019-02-20:
@jf6921668 support for Parrot added, try running reinstall and let me know if you're still experiencing same issue.

commit d4788e46 (2019-02-20): added support for Parrot (close: #222)
closed 2019-02-20 by AdnanHodzic