AdnanHodzic/displaylink-debian #223: Display Not Detected Ubuntu 18.04 (Unity?)

alexhallam, 2019-02-18:
When running. `xrandr --listproviders`.  I should see two external monitors instead I see the following:

```
Providers: number : 1
Provider 0: id: 0x48 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 6 associated providers: 0 name:modesetting
```

here is the debug info
```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.15.0-45-generic

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 530
VGA: Intel Corporation HD Graphics 530 (rev 06)
VGA (3D): 
X11 version: 1.19.6-1ubuntu4.2
14.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x48 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 6 associated providers: 0 name:modesetting

-------------------------------------------------------------------
```

Following the Post-Install guide, I do have a running service
```
systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-02-18 09:23:53 EST; 11min ago
  Process: 1008 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/4.4.24 && modprobe ev
 Main PID: 1012 (DisplayLinkMana)
    Tasks: 11 (limit: 4915)
   CGroup: /system.slice/dlm.service
           └─1012 /opt/displaylink/DisplayLinkManager
```

I am running `Unity:Unity7` I am wondering if this driver is compatible with Unity.

AdnanHodzic, 2019-02-18:
Yes it's compatible with any desktop environment.

Usually it seamlessly works with Intel cards. Do you have any graphics card besides Intel one? If yes can I get output of `lspci` command?

alexhallam, 2019-02-18:
I do not have any additional graphics cards.

AdnanHodzic, 2019-02-18:
Are there any changes if you make following as contents of your `/etc/X11/xorg.conf.d/20-displaylink.conf` file, followed by reboot?

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

Also, you have consulted the [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md) right?

alexhallam, 2019-02-18:
I tried your suggestion. No luck. Also, I have gone through the following Post Installation tasks.

- [X] Make sure that UEFI/secure boot is disabled!

- [X] When you are logging into your session you are using X server (X11) and not Wayland.

- [X] After install you must reboot to apply the changes.

- [X] After reboot, make sure DispayLink is running, i.e:

- [ ] Check providers `xrandr --listproviders`

AdnanHodzic, 2019-02-20:
I've just merged `displaylink-5` branch with `master` which introduces big changes, among which are support for yesterday released DisplayLink 5.1 driver. Disabling modesetting as part of `displaylink-debian` as now all of it should be happening on EVDI level. I've included few more changes.

Regardless, please pull latest (`git pull`) changes on `master` and run `sudo ./displaylink-debian.sh --install`

After this everything should be working as expected and with this said I'm closing the issue. If the issue still persists, re-open the issue and include output of `sudo ./displaylink-debian.sh --debug` command.

closed 2019-02-20 by AdnanHodzic