AdnanHodzic/displaylink-debian #224: New Version of Displaylink  driver 5.1 and no dlm.service found

wszychta, 2019-02-19:
Hi,

For your information there is new version of ubuntu driver. It was released today.
https://www.displaylink.com/downloads/ubuntu

After using your script I had error `dlm.service not found`, but driver works well after install. I just cannot manage dlm service

AdnanHodzic, 2019-02-19:
I'm testing these on `displaylink-5` branch changes before merging them with master, you can latest changes by running:

```
git fetch
git checkout displaylink-5
sudo ./displaylink-debian.sh
```

It's still bit rough around the edges, i.e: after reboot you may need to run `systemctl start displaylink-driver` in case driver is not started and working. Either way, let me me know how it's working for you.

Thanks

lhy0403, 2019-02-20:
After my test, I have Nvidia-driver package and bumblebee-Nvidia installed, and then this tool generated a `20-DisplayLink.conf` under my `/etc/X11/xorg.conf.d`
Removing the file made my device work, but with the file, I received a crash when starting x11.
I don't know how I can help, but it finally works,
Is this strange thing happening because of my wired package installations?

wszychta, 2019-02-20:
@AdnanHodzic - I have tested your changes and my problem is solved.

Thank you for your fast support

closed 2019-02-20 by wszychta

AdnanHodzic, 2019-02-20:
I've just merged `displaylink-5` branch with `master` which introduces big changes, among which are support for yesterday released DisplayLink 5.1 driver. Disabling modesetting as part of `displaylink-debian` as now all of it should be happening on EVDI level. I've included few more changes.

Regardless, please pull latest (`git pull`) changes on `master` and run `sudo ./displaylink-debian.sh --install`

After this everything should be working as expected and with this said I'm closing the issue. If the issue still persists, re-open the issue and include output of `sudo ./displaylink-debian.sh --debug` command.