AdnanHodzic/displaylink-debian #225: issue with enabling screen

killmasta93, 2019-02-21:
Hi,
So with the recent update i was still not able to get working the screen. Steps i took
First install 
`git clone https://github.com/AdnanHodzic/displaylink-debian.git
`
then cd the folder and gave chmod +x displaylink-debian.sh
`./displaylink-debian.sh `
after that reboot
then went to the website https://www.displaylink.com/downloads/ubuntu
and ran the script and reboot

i try to enable the screen but wont enable it these are the error logs

`Feb 20 21:22:18 HA55AN org.kde.KScreen[2986]: kscreen.xcb.helper: #011Root: 415
Feb 20 21:22:18 HA55AN org.kde.KScreen[2986]: kscreen.xcb.helper: #011Rotation:  "Rotate_0"
Feb 20 21:22:18 HA55AN org.kde.KScreen[2986]: kscreen.xcb.helper: #011Size ID: 65535
Feb 20 21:22:18 HA55AN org.kde.KScreen[2986]: kscreen.xcb.helper: #011Size:  1920 1080
Feb 20 21:22:18 HA55AN org.kde.KScreen[2986]: kscreen.xcb.helper: #011SizeMM:  506 285
Feb 20 21:22:18 HA55AN kernel: [  650.919176] evdi: [D] evdi_detect:88 (dev=2) poll connector state: connected
Feb 20 21:22:18 HA55AN kernel: [  650.919178] evdi: [D] evdi_painter_get_edid_copy:227 (dev=2) EDID valid
Feb 20 21:22:18 HA55AN kernel: [  650.919431] evdi: [D] evdi_detect:88 (dev=1) poll connector state: connected
Feb 20 21:22:18 HA55AN kernel: [  650.919432] evdi: [D] evdi_painter_get_edid_copy:227 (dev=1) EDID valid
Feb 20 21:22:18 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: Emitting configChanged()
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: XRandR::setConfig
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: Requested screen size is QSize(3840, 1148)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: Needed CRTCs:  2
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: Actions to perform:
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Primary Output: false
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Change Screen Size: true
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011#011Old: QSize(1920, 1080)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011#011Intermediate: QSize(3840, 1148)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011#011New: QSize(3840, 1148)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Disable outputs: false
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Change outputs: false
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Enable outputs: true
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011#011 (560)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: RRSetScreenSize
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011DPI: 96.2526
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Size: QSize(3840, 1148)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011SizeMM: QSize(1013, 302)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: QMap((63, XRandRCrtc(0x55fa35be37e0))(64, XRandRCrtc(0x55fa35be4260))(65, XRandRCrtc(0x55fa35be42b0)))
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: Testing CRTC 63
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Free: false
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Mode: 72
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Possible outputs: QVector(66, 67, 68, 69, 70)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Connected outputs: QVector(66)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Geometry: QRect(0,0 1366x768)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: Testing CRTC 64
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Free: true
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Mode: 0
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Possible outputs: QVector(66, 67, 68, 69, 70)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Connected outputs: QVector()
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Geometry: QRect(1920,0 0x0)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: Testing CRTC 65
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Free: true
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Mode: 0
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Possible outputs: QVector(66, 67, 68, 69, 70)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Connected outputs: QVector()
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Geometry: QRect(0,0 0x0)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: Failed to get free CRTC for output 560
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: Output failed to be Enabled:  "DVI-I-2-2"
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: Requested screen size is QSize(3840, 1148)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: Forced to change screen size:  QSize(3840, 1148)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: RRSetScreenSize
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011DPI: 96.2526
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011Size: QSize(3840, 1148)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: #011SizeMM: QSize(1013, 302)
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: XRandR::setConfig done!
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xcb.helper: RRScreenChangeNotify
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xcb.helper: #011Window: 44040197
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xcb.helper: #011Root: 415
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xcb.helper: #011Rotation:  "Rotate_0"
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xcb.helper: #011Size ID: 65535
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xcb.helper: #011Size:  3840 1148
Feb 20 21:22:40 HA55AN org.kde.KScreen[2986]: kscreen.xcb.helper: #011SizeMM:  1013 302
Feb 20 21:22:40 HA55AN kernel: [  673.322788] evdi: [D] evdi_detect:88 (dev=2) poll connector state: connected
Feb 20 21:22:40 HA55AN kernel: [  673.322790] evdi: [D] evdi_painter_get_edid_copy:227 (dev=2) EDID valid
Feb 20 21:22:40 HA55AN kernel: [  673.323059] evdi: [D] evdi_detect:88 (dev=1) poll connector state: connected
Feb 20 21:22:40 HA55AN kernel: [  673.323060] evdi: [D] evdi_painter_get_edid_copy:227 (dev=1) EDID valid
Feb 20 21:22:41 HA55AN org.kde.KScreen[2986]: kscreen.xrandr: Emitting configChanged()
`
and this is the debug mode
`--------------- Linux system info ----------------

Distro: Ubuntu
Release: cosmic
Kernel: 4.18.0-15-generic

---------------- DisplayLink info ----------------

Driver version: 4.4.24
5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [HD
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): 
X11 version: 1.20.1-3ubuntu2.1
X11 configs: /etc/X11/xrdp/xorg.conf

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 2 name:modesetting
Provider 1: id: 0x231 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x20e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

`

Thank you

AdnanHodzic, 2019-02-21:
In your original issue you first ran `displaylink-debian` and then downloaded script from DisplayLink website?

> git clone https://github.com/AdnanHodzic/displaylink-debian.git 
> then cd the folder and gave chmod +x displaylink-debian.sh
> ./displaylink-debian.sh 
> after that reboot
> then went to the website https://www.displaylink.com/downloads/ubuntu
> and ran the script and reboot

This makes no sense, as point of `displaylink-debian` is that it does everything for you: https://github.com/AdnanHodzic/displaylink-debian#technical

Regardless, judging from your debug output, you have 2 driver versions running now: `Driver version: 4.4.24
5.1.26`

I also see `/etc/X11/xrdp/xorg.conf` file under your X configs, since this is now all done on EVDI level, if you don't need it remove this file, if you do need it make sure to back it up.

Run these line by line, if one of them fails just proceed to next one:

```
sudo dkms remove evdi/1.6.0 --all -q 
sudo systemctl stop dlm.service 
sudo systemctl disable dlm.service 
sudo rm -f /lib/systemd/system/dlm.service
sudo rmmod evdi 
sudo ./displaylink-debian.sh --uninstall 
```

Make sure to reboot after uninstall is finished.

After reboot run debug (`sudo displaylink-debian.sh --debug`) and make sure nothing under "DisplayLink info" is installing/up and running. After this install again and you should be set:
`sudo displaylink-debian.sh --install`

killmasta93, 2019-02-22:
Thanks for the reply, sorry what i meant is that i tried the other version on the website before trying yours and maybe this is why its happening this glitch. 
So unistalled everything i reran the debug i got this 
`--------------- Linux system info ----------------

Distro: Ubuntu
Release: cosmic
Kernel: 4.18.0-15-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [HD
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): 
X11 version: 1.20.1-3ubuntu2.1
X11 configs:

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)
`

Then ran 
displaylink-debian.sh --install information when submitting an issue (http://bit.ly/2GLDlpY)
`

Then ran 
displaylink-debian.sh --install

reboot and tried to connect and completely freezes my computer i have to reboot and disconnect

AdnanHodzic, 2019-02-22:
In one of the last commits I disabled automated graphic cards configuration done by `displaylink-debian`. 

What would happen once you install it and, before you reboot you created this file: `/etc/X11/xorg.conf.d/20-displaylink.conf` with following contents:

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

Followed by reboot, what does this do?

killmasta93, 2019-02-25:
Thanks that did the trick going to test it to see how it works thank you

AdnanHodzic, 2019-02-25:
Thanks for confirming this for me, I've just re-enabled modesetting to be set by `displaylink-debian` based on detected graphics card and pushed the changes.

closed 2019-02-25 by AdnanHodzic