AdnanHodzic/displaylink-debian #226: Supported Hardware List

dkebler, 2019-02-25:
Since I haven't found any definitive list maybe as part of this project/repo you could maintain a list the hardware that users have working with your displaylink distribution.  There are SO many brands/models out there with little or no information on which work with displaylink on linux let alone this distribution.  One could waste $$ and time buying and trying.   

Could you have a notice in the readme and then separate md file or wiki page and an open issue where folks can list their successfully running hub/hardware with maybe a link for more info/purchase.  

Or if you don't want to go to that effort just keep this issue open and put a link to it in your readme.md.   Thx.

dkebler, 2019-02-25:
Here is link to what the displaylink folks list as compatible hardware.  Anyone have any experience with any of these? https://www.displaylink.com/products

AdnanHodzic, 2019-02-26:
There's a [list of supported Linux platforms/distros](https://github.com/AdnanHodzic/displaylink-debian/#supported-platforms-are), and if I user wants their distro/release supported they can submit an issue and I'll look into it. In ToDo for this project, I also have a task for automated distro release detection so I wouldn't have to do this manually. I also had plans to extend support to RedHat derivatives, but since I do this in my free time, this just item doesn't pop up as a priority.

Supported chipsets are `DL-1x0`, `DL-1x` `DL-3xxx`, `DL-41xx`, `DL-5xxx`, `DL-6xxx` as it was mentioned mentioned in: https://github.com/AdnanHodzic/displaylink-debian/issues/215

When it comes to any particular chipsets or any other compatible hardware. I haven't been reached by anyone from DisplayLink, and I don't have any contact with them. Outside of work I don't own any of those devices and don't intend on buying them to add support for their product, which is something they should probably be doing in the first place. 

Regarding "success story" list, honestly I'm not 100% sure about this idea, but I'm open to reconsider it. If you want, I'd encourage you to change the "README" and point it to this issue or however you thought would be the best approach. Submit a "pull request" and I'll be glad to review your changes and merge it with master.

Until then I'll leave this issue open.

dkebler, 2019-03-02:
@AdnanHodzic So you are indicating (despite description saying linux not supported in the description) the hardware linked below should work as it uses a DL-6950 chipset? -- given I already have your driver installed successfully. 

https://www.amazon.com/gp/product/B07476R6BP 

At $90 (ouch) one can see the dilemma of "tying" hardware.  Well, at least amazon is good about returns.

AdnanHodzic, 2019-03-09:
Bit confused now, are you saying that `displaylink-debian` successfully installed but doesn't work on your DL-6950?

Darcidride, 2019-03-14:
@dkebler ping, it's an useful information for others :)

dkebler, 2019-03-19:
This DL6950 device just arrived

[image]

https://www.amazon.com/gp/product/B07476R6BP/ref=ox_sc_act_title_2?smid=A3HIHADV23VGU1&psc=1

and I can confirm on Linux Mint 19+ (Ubuntu 18.04) that this device and @AdnanHodzic 's module/driver work great.  I am using the displaylink monitor to write this!  So yes it works with Linux despite the verbiage on the Amazon page.  FYI I am running a box similar to a NUC with a 8th gen I5 chip. 

https://www.aliexpress.com/item/HYSTOU-i7-8550U-DDR4-32GB-RAM-Windows-10-pro-Mini-PC-with-Fan-Type-C-UHD/32954596951.html 

The two onboard video ports and the two from the displaylink device all appear in the display utility and the two displaylink displays can be manipulated just like the built in hdmi and DP ones. 

Thx! @AdnanHodzic now I have a wall of 4 monitors for development work. 

[image]

https://www.amazon.com/gp/product/B07476R6BP/ref=ox_sc_act_title_2?smid=A3HIHADV23VGU1&psc=1

closed 2020-07-19 by Barabazs