AdnanHodzic/displaylink-debian #227: Script on MX 18 does not work properly

jjxxzz, 2019-02-25:
Hi,

As mentioned in original issue, the script does not work for MX18

Error from script:
DisplayLink Linux Software 5.1.26 install script called: install
/sbin/init: invalid option -- '-'
Usage: init {-e VAR[=VAL] | [-t SECONDS] {0|1|2|3|4|5|6|S|s|Q|q|A|a|B|b|C|c|U|u}}
Distribution discovered: MX 18.1 Continuum
Unsatisfied dependencies. Missing component: Linux headers for running kernel, 4.15.0-1-amd64.
This is a fatal error, cannot install DisplayLink Linux Software.
Adding udlfb to blacklist
Adding udl to blacklist

Debug script output:

cat: /sys/devices/evdi/version: Nie ma takiego pliku ani katalogu
--------------- Linux system info ----------------

Distro: MX
Release: Continuum
Kernel: 4.15.0-1-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 5500
VGA: Intel Corporation HD Graphics 5500 (rev 09)
VGA (3D): 
X11 version: 1.19.2-1+deb9u5
X11 configs: /etc/X11/xorg.conf.d/10-evdev.conf /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/synaptics.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 0 name:Intel

Result from 'apt-get install linux-headers-4.15.0-1-amd64'
linux-headers-4.15.0-1-amd64 is already the newest version (4.15.4-1~mx17+1).

jjxxzz, 2019-02-25:
Fixed, solution found in topic below
https://github.com/AdnanHodzic/displaylink-debian/issues/28

Sorry!

closed 2019-02-25 by jjxxzz

AdnanHodzic, 2019-02-26:
@jjxxzz ah it's good you pointed this out, this was done automatically for Debian/Kali/Deepin but not for MX hence as part of last commit I added necessary MX support: https://github.com/AdnanHodzic/displaylink-debian/commit/f62893a3ab9f85b20f090c24b5e4382471397ab8

So from now on this will be done automatically for MX.