AdnanHodzic/displaylink-debian #228: Ubuntu 18.10 - HP spectre 13 x360

Tegaf, 2019-02-27:
Hi,

I have a new HP spectre 13 x360 and I have a problem with dispaylink on Ubuntu 18.10.

After installing the script, I can not run the dlm.service

```
$ sudo systemctl start dlm.service
Failed to start dlm.service: Unit dlm.service not found.
```

```
$ uname -a
Linux 4.18.0-15-generic #16-Ubuntu SMP Thu Feb 7 10:56:39 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

```
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.18.0-15-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.displaylink.com/downloads/file?id=1304
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2019-02-27 15:11:24--  https://www.displaylink.com/downloads/file?id=1304
Překládám www.displaylink.com (www.displaylink.com)… 34.246.147.251, 18.203.109.3, 54.72.7.162, ...
Navazuje se spojení s www.displaylink.com (www.displaylink.com)|34.246.147.251|:443… spojeno.
HTTP požadavek odeslán, program čeká na odpověď… 303 See Other
Přesměrováno na: http://assets.displaylink.com/live/downloads/software/f1304_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1551277593&Signature=xvX0gzdgotlJBye0EBLnuyielLg%3D [následuji]
--2019-02-27 15:11:24--  http://assets.displaylink.com/live/downloads/software/f1304_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1551277593&Signature=xvX0gzdgotlJBye0EBLnuyielLg%3D
Překládám assets.displaylink.com (assets.displaylink.com)… 52.218.80.226
Navazuje se spojení s assets.displaylink.com (assets.displaylink.com)|52.218.80.226|:80… spojeno.
HTTP požadavek odeslán, program čeká na odpověď… 200 OK
Délka: 12102697 (12M) [application/octet-stream]
Ukládám do: „DisplayLink_Ubuntu_5.1.zip“

DisplayLink_Ubuntu_ 100%[===================>]  11,54M  2,23MB/s    za 5,2s    

2019-02-27 15:11:30 (2,21 MB/s) – „DisplayLink_Ubuntu_5.1.zip“ uloženo [12102697/12102697]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.1.zip
  inflating: 5.1/displaylink-driver-5.1.26.run  
  inflating: 5.1/LICENSE             
Creating directory displaylink-driver-5.1.26
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.1.26  100%  

-------------------------------------------------------------------

Installing driver version: 5.1

DisplayLink Linux Software 5.1.26 install script called: install
Distribution discovered: Ubuntu 18.10
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS
Installing EVDI kernel module to kernel tree
EVDI kernel module built successfully
Installing x64-ubuntu-1604/DisplayLinkManager
Installing libraries
Installing firmware packages
Installing licence file
Adding udev rule for DisplayLink DL-3xxx/4xxx/5xxx/6xxx devices

Please read the FAQ
http://support.displaylink.com/knowledgebase/topics/103927-troubleshooting-ubuntu
Installation complete!
Removing Displaylink Xorg config file

-------------------------------------------------------------------

Performing post install steps

Enable and start displaylink-driver service
Created symlink /etc/systemd/system/multi-user.target.wants/displaylink-driver.service → /lib/systemd/system/displaylink-driver.service.
Setup DisplayLink xorg.conf depending on graphics card
Wrote X11 changes to: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_5.1.zip" file
Removing redundant: "5.1" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes
After reboot, make sure to consult post-install guide! http://bit.ly/2TbZleK
Reboot now? [Y/n] n

Reboot postponed, changes won't be applied until reboot

-------------------------------------------------------------------
```

```-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: cosmic
Kernel: 4.18.0-15-generic

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [103c:8514]
VGA: Intel Corporation Device 3ea0
VGA (3D): 
X11 version: 1.20.1-3ubuntu2.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

I also tried:
```
-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

```

Any idea?

Thanks!

AdnanHodzic, 2019-02-27:
Yes, because `dlm.service` doesn't exist anymore with 5.1 DisplayLink driver, see how you should check if driver is running or not: https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#prerequisites

Judging by the version of Xorg you're running your `/etc/X11/xorg.conf.d/20-displaylink.conf` file should be as follows:

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
``` 
so change it to that followed by reboot.

If that also doesn't change any anything, try completely removing `/etc/X11/xorg.conf.d/20-displaylink.conf` file followed by reboot and see if that makes any changes.

Also, your graphic card definitely wasn't detected properly, could you give me whole output of command `lspci`?

Tegaf, 2019-02-27:
The `MatchDriver "evdi" ` line blocking login to the system. But if I delete  `/etc/X11/xorg.conf.d/20-displaylink.conf`, everything is fine! 

Thank you very much!

p.s.
```
00:00.0 Host bridge: Intel Corporation Device 3e34 (rev 0b)
00:02.0 VGA compatible controller: Intel Corporation Device 3ea0
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 0b)
00:12.0 Signal processing controller: Intel Corporation Device 9df9 (rev 30)
00:13.0 Serial controller: Intel Corporation Device 9dfc (rev 30)
00:14.0 USB controller: Intel Corporation Device 9ded (rev 30)
00:14.2 RAM memory: Intel Corporation Device 9def (rev 30)
00:14.3 Network controller: Intel Corporation Device 9df0 (rev 30)
00:15.0 Serial bus controller [0c80]: Intel Corporation Device 9de8 (rev 30)
00:15.1 Serial bus controller [0c80]: Intel Corporation Device 9de9 (rev 30)
00:16.0 Communication controller: Intel Corporation Device 9de0 (rev 30)
00:1c.0 PCI bridge: Intel Corporation Device 9dba (rev f0)
00:1c.4 PCI bridge: Intel Corporation Device 9dbc (rev f0)
00:1d.0 PCI bridge: Intel Corporation Device 9db4 (rev f0)
00:1f.0 ISA bridge: Intel Corporation Device 9d84 (rev 30)
00:1f.3 Multimedia audio controller: Intel Corporation Device 9dc8 (rev 30)
00:1f.4 SMBus: Intel Corporation Device 9da3 (rev 30)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Device 9da4 (rev 30)
01:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS525A PCI Express Card Reader (rev 01)
02:00.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
03:00.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
03:01.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
03:02.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
03:04.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
04:00.0 System peripheral: Intel Corporation JHL6540 Thunderbolt 3 NHI (C step) [Alpine Ridge 4C 2016] (rev 02)
38:00.0 USB controller: Intel Corporation JHL6540 Thunderbolt 3 USB Controller (C step) [Alpine Ridge 4C 2016] (rev 02)
6d:00.0 Non-Volatile memory controller: Sandisk Corp Device 5002
```

closed 2019-02-27 by AdnanHodzic

kmosakowski, 2020-06-21:
Hi
I have a HP Spectre x360 with integreated graphic card - Intel Iris 540. I own two external monitors and I would like to connect them through DisplayLink by USB 3.0

My system:

```
PRETTY_NAME="Kali GNU/Linux Rolling"
NAME="Kali GNU/Linux"
ID=kali
VERSION="2020.2"
VERSION_ID="2020.2"
VERSION_CODENAME="kali-rolling"
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="https://www.kali.org/"
SUPPORT_URL="https://forums.kali.org/"
BUG_REPORT_URL="https://bugs.kali.org/"
```

I know, that Kali 2020.2 is not supported in displaylink-debian but Latest Official Driver for DisplayLink was released.

Here is my debug:
```
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Kali
Release: kali-rolling
Kernel: 5.6.0-kali2-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
intel_ish_ipc
Subsystem: Graphics
Point-LP
VGA: Intel Corporation Iris Graphics 540 (rev 0a)
VGA (3D): 
X11 version: 1.20.8-2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0x161 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x13f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x11d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0xe2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

In /etc/X11/xorg.conf.d/20-displaylink.conf I used many combination from Post Installation Guide, I try with this file and without, but nothing was changed.
UEFI and Secure Boot is disabled.