AdnanHodzic/displaylink-debian #229: Dell d6000

ychen0113, 2019-02-28:
Hi, I currently have a Dell D6000 hooked up to my laptop and cannot detect the screen. 

No protocol specified
Can't open display :0

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.18.0-15-generic

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: HD
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.20.1-3ubuntu2.1~18.04.1
X11 configs:

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

AdnanHodzic, 2019-02-28:
Can you try doing same thing I suggested here: https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348

ychen0113, 2019-02-28:
i had previously deleted the ~/20-displaylink.conf file.
i created the file and wrote:
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

nothing worked so i removed the file again, displays still not being detected.

output of xrandr --listproviders
 Providers: number : 0

AdnanHodzic, 2019-02-28:
Weird, as that should've worked, what happens if make following:

```
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
```

to be part of your `/etc/X11/xorg.conf.d/20-displaylink.conf` file

ychen0113, 2019-02-28:
nothing, i edited both 20-displaylink.conf one under /usr and one under /etc

ilhanu, 2019-03-19:
I have the same problem with the D6000

christian-roggia, 2019-03-27:
I am currently working with an XPS 13 9370, the external monitor is correctly detected when I execute the `xrandr --listproviders` but as soon as I try to connect the monitor via `xrandr --setprovideroutputsource 1 0` the whole system will freeze and I have to forcefully shutdown, disconnecting the docking station doesn't help. I am currently working on Debian 9.

```
--------------- Linux system info ----------------

Distro: Debian
Release: stretch
Kernel: 4.9.0-8-amd64

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 5917 (rev 07)
VGA (3D): 
X11 version: 1.19.2-1+deb9u5
X11 configs: /etc/X11/xrdp/xorg.conf
```

Additional notes:
The file `/etc/X11/xorg.conf.d/20-displaylink.conf` originally contained
```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
```

When I tried to switch to
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

after rebooting the system would crash as soon as I do the login bringing me back to the login screen in a loop, I had to boot in recovery mode and restore the original content of the file.

I also tried to use
```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```
and I observed no changes after I applied this configuration.

christian-roggia, 2019-04-07:
This issue is fully solved by upgrading to Debian 10.

closed 2020-07-19 by Barabazs