AdnanHodzic/displaylink-debian #230: System freeze after update

gmd7, 2019-03-02:


AdnanHodzic, 2019-03-05:
Can you try doing same thing I suggested here: #228 (comment)

gmd7, 2019-03-05:
1. lspci

```
# lspci
00:00.0 Host bridge: Intel Corporation Haswell-ULT DRAM Controller (rev 09)
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 09)
00:03.0 Audio device: Intel Corporation Haswell-ULT HD Audio Controller (rev 09)
00:14.0 USB controller: Intel Corporation 8 Series USB xHCI HC (rev 04)
00:16.0 Communication controller: Intel Corporation 8 Series HECI #0 (rev 04)
00:1b.0 Audio device: Intel Corporation 8 Series HD Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 1 (rev e4)
00:1c.2 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 3 (rev e4)
00:1d.0 USB controller: Intel Corporation 8 Series USB EHCI #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation 8 Series LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 8 Series SATA Controller 1 [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 8 Series SMBus Controller (rev 04)
06:00.0 Network controller: Intel Corporation Wireless 7260 (rev 6b)
```

2. OutputClass

  a. X doesn't start

```
--------------- Linux system info ----------------

Distro: Debian
Release: stretch
Kernel: 4.9.0-8-amd64

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.19.2-1+deb9u5
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

```
$ startx -- :1

X.Org X Server 1.19.2
Release Date: 2017-03-02
[... 89 lines cut ...]
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.19.2-1+deb9u5
X11 configs:

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:modesetting
Provider 1: id: 0x14c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x11a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```

M4GNV5, 2019-03-09:
Hey,

I have the same problem.

I'm running Debian 9 with GNOME on a Zenbook 3 (UX390UA) with an i-tec dock (DL6950).
Deleting `/etc/X11/xorg.conf.d/20-displaylink.conf` made the second xrandr provider show up. Running `xrandr --setprovideroutputsource 1 0` makes it show up in the gnome display settings, but enabling it makes the whole PC freeze. sysreq keys don't work when freezed either.

There seemed to pop up a few of these issues (#228, #229). Maybe we should try contacting DisplayLink about this? Or do you have an idea on how to fix this?

You can find my outputs here if needed: https://gist.github.com/M4GNV5/14c0b111cad0c450ead757609ae9f2cb

AdnanHodzic, 2019-03-09:
That's a good idea, I'd encourage you to submit this problem with upstream driver (https://github.com/DisplayLink/evdi/issues) and make a reference to this issue.

Out of shower curiosity what would happen if you did what I suggested here: https://github.com/AdnanHodzic/displaylink-debian/issues/236#issuecomment-471213411

M4GNV5, 2019-03-13:
Nope, i tried different versions of `/etc/X11/xorg.conf.d/20-displaylink.conf`, both versions in that comment make it find the displayport connectors but freeze when trying to use them.
I also tried some versions I found on other issues, but they either result in X crashing on start, the outputs not being found or the same freeze.

Any ideas?
It's kind of frustrating because the HDMI output of the dock flickers when outputting 4K and doesn't support more than 30Hz on 4K anyways, which means I'm currently stuck with FullHD.

ehrenbrav, 2019-03-13:
Exact same issue...

Codemann14, 2019-03-14:
I am having the same issue as well. I am using the latest debian version with 4.9.0-8-amd64 kernel.

AdnanHodzic, 2019-03-17:
@M4GNV5 @ehrenbrav @Codemann14 I can't help you as I'm not experiencing any similar issues, could you please submit an issue with [upstream driver](https://github.com/DisplayLink/evdi/issues) and make a reference to this issue? Maybe someone there will have an idea who to resolve this issue.

In meantime, please let me know if there are any code changes I can do to resolve this issue.

zwierzi, 2019-03-19:
I'm having the same issue.

Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection

It detects monitor, but after confirm to use it, system freeze.

gmd7, 2019-04-08:
Upgrading to Debian Buster solved the problem.

damnhalfling, 2019-05-04:
> Upgrading to Debian Buster solved the problem.
I have upgrade tu buster and not work! :/

my config:

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-4-amd64

---------------- DisplayLink info ----------------

Driver version: 1.6.0+dfsg
5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.20.3-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
Identifier "DisplayLink"
Driver "modesetting"
Option "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x46 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Intel
Provider 1: id: 0x1c4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

gmd7, 2019-05-06:
can you try

```
Section "OutputClass"
        Identifier "DisplayLink"
        MatchDriver "evdi"
        Driver "modesetting"
        Option  "AccelMethod" "none"
EndSection
```

damnhalfling, 2019-05-07:
It not work for my it say that Option  "AccelMethod" "none" on this line Option is not valied and without this line my screen is freezy too

godspeed-you, 2019-10-10:
A little update for this problem: It seems, that this issue is related to evdi and not to the script. I faced the same issue and moved back to version 4.4, which is working perfectly. 
Some log snippets while using evdi 5.2:
```
Oct 10 10:53:34 HOSTNAME kernel: [   76.856518] evdi: [D] evdi_painter_dpms_notify:544 (dev=2) Notifying dpms mode: 0
Oct 10 10:53:34 HOSTNAME kernel: [   76.856523] evdi: [D] evdi_painter_mode_changed_notify:608 (dev=2) Notifying mode changed: 2560x1440@60; bpp 32;
Oct 10 10:53:34 HOSTNAME kernel: [   76.856524] evdi: [D] evdi_log_pixel_format:574 pixel format XR24 little-endian (0x34325258)
Oct 10 10:53:34 HOSTNAME kernel: [   76.856525] evdi: [D] evdi_painter_dpms_notify:544 (dev=2) Notifying dpms mode: 0
Oct 10 10:53:34 HOSTNAME kernel: [   76.869497] ------------[ cut here ]------------
Oct 10 10:53:34 HOSTNAME kernel: [   76.869501] WARNING: CPU: 1 PID: 1530 at /build/linux-sdMcHj/linux-4.9.189/lib/list_debug.c:62 __free_vmap_area+0x58/0xd0
Oct 10 10:53:34 HOSTNAME kernel: [   76.869501] list_del corruption. next->prev should be ffff90e6dccd4db0, but was ffff90e72830a000
...
Oct 10 10:53:34 HOSTNAME kernel: [   76.869559] CPU: 1 PID: 1530 Comm: SourceEngine_00 Tainted: G           O    4.9.0-11-amd64 #1 Debian 4.9.189-3+deb9u1                                                                                                                                
Oct 10 10:53:34 HOSTNAME kernel: [   76.869559] Hardware name: Dell Inc. Latitude 7480/00F6D3, BIOS 1.15.1 07/04/2019
Oct 10 10:53:34 HOSTNAME kernel: [   76.869561]  0000000000000000 ffffffff817353d4 ffff9e2d03f0bbd8 0000000000000000
Oct 10 10:53:34 HOSTNAME kernel: [   76.869562]  ffffffff8147a83b ffff90e6dccd4d80 ffff9e2d03f0bc30 ffff90e73e596cd8
Oct 10 10:53:34 HOSTNAME kernel: [   76.869563]  ffff9e2d03f0bcd8 ffff9e2d03f0bce0 ffff9e2d03f0bc20 ffffffff8147a8bf
Oct 10 10:53:34 HOSTNAME kernel: [   76.869564] Call Trace:
Oct 10 10:53:34 HOSTNAME kernel: [   76.869567]  [<ffffffff817353d4>] ? dump_stack+0x5c/0x78
Oct 10 10:53:34 HOSTNAME kernel: [   76.869568]  [<ffffffff8147a83b>] ? __warn+0xcb/0xf0
Oct 10 10:53:34 HOSTNAME kernel: [   76.869569]  [<ffffffff8147a8bf>] ? warn_slowpath_fmt+0x5f/0x80
Oct 10 10:53:34 HOSTNAME kernel: [   76.869571]  [<ffffffff8146986a>] ? do_flush_tlb_all+0xa/0x30
Oct 10 10:53:34 HOSTNAME kernel: [   76.869572]  [<ffffffff815ca758>] ? __free_vmap_area+0x58/0xd0
Oct 10 10:53:34 HOSTNAME kernel: [   76.869573]  [<ffffffff815cae34>] ? __purge_vmap_area_lazy+0x2d4/0x300
Oct 10 10:53:34 HOSTNAME kernel: [   76.869574]  [<ffffffff815cb2bf>] ? alloc_vmap_area+0x31f/0x3d0
Oct 10 10:53:34 HOSTNAME kernel: [   76.869575]  [<ffffffff815cb650>] ? __get_vm_area_node+0xb0/0x140
Oct 10 10:53:34 HOSTNAME kernel: [   76.869576]  [<ffffffff815ccea0>] ? __vmalloc_node_range+0x70/0x280
Oct 10 10:53:34 HOSTNAME kernel: [   76.869578]  [<ffffffff81479a53>] ? _do_fork+0xe3/0x3f0
Oct 10 10:53:34 HOSTNAME kernel: [   76.869579]  [<ffffffff815945c4>] ? __pagevec_lru_add_fn+0x154/0x260
Oct 10 10:53:34 HOSTNAME kernel: [   76.869581]  [<ffffffff8147836a>] ? copy_process.part.34+0x64a/0x1b50
Oct 10 10:53:34 HOSTNAME kernel: [   76.869582]  [<ffffffff81479a53>] ? _do_fork+0xe3/0x3f0
Oct 10 10:53:34 HOSTNAME kernel: [   76.869583]  [<ffffffff81595318>] ? __lru_cache_add+0x68/0x80
Oct 10 10:53:34 HOSTNAME kernel: [   76.869585]  [<ffffffff81479a53>] ? _do_fork+0xe3/0x3f0
Oct 10 10:53:34 HOSTNAME kernel: [   76.869586]  [<ffffffff814623b8>] ? __do_page_fault+0x278/0x4f0
Oct 10 10:53:34 HOSTNAME kernel: [   76.869587]  [<ffffffff81403b7d>] ? do_syscall_64+0x8d/0x100
Oct 10 10:53:34 HOSTNAME kernel: [   76.869589]  [<ffffffff81a1c3ce>] ? entry_SYSCALL_64_after_swapgs+0x58/0xc6
Oct 10 10:53:34 HOSTNAME kernel: [   76.869590] ---[ end trace 69d2860758b36d07 ]---
```

So there is at least a workaround: Use version 4.4 (the last version bevor 5.x)

Note: I did not test 5.1 and instead moved directly away from any version that did not use dlm.

closed 2020-07-19 by Barabazs