AdnanHodzic/displaylink-debian #231: Distribution Support Request: Latest Kali (2019-02-14) with 4.19.20 kernel

mickhawkins, 2019-03-03:
I originally attempted to install on the older 4.15 kernel, but with that now out of date, headers was only available for 4.19, so I could only install the displaylink-debian required dependencies by upgrading to the newer kernel.

I will be supremely thankful if you are able to get this running on the latest Kali, so I don't have to rebuild from scratch on a different distro!

Here's the debug info:

```
Unit dlm.service could not be found.
--------------- Linux system info ----------------

Distro: Kali
Release: n/a
Kernel: 4.19.0-kali3-amd64

---------------- DisplayLink info ----------------

Driver version: 5.1
EVDI service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
intel_ish_ipc
Subsystem: 620
Integrated
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.19.6-1
X11 configs:

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 
-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting
```

Thanks,

Mick

AdnanHodzic, 2019-03-05:
`dlm.service` doesn't exist with 5.1 driver anymore, make sure you're running latest code and run following.

Run these line by line, if one of them fails just proceed to next one:

```
sudo dkms remove evdi/1.6.0 --all -q 
sudo systemctl stop dlm.service 
sudo systemctl disable dlm.service 
sudo rm -f /lib/systemd/system/dlm.service
sudo rmmod evdi 
sudo ./displaylink-debian.sh --uninstall 
```

Make sure to reboot after uninstall is finished.

After reboot run debug (`sudo displaylink-debian.sh --debug`) and make sure nothing under "DisplayLink info" is installing/up and running. After this install again and you should be set:
`sudo displaylink-debian.sh --install`

mickhawkins, 2019-04-04:
Thanks for that, uninstalling, then reinstalling with the latest version (downloaded today) seems to have done it!
I managed to get both additional monitors working with some fiddling thanks to the post-install guide.

[image]

closed 2019-04-04 by mickhawkins