AdnanHodzic/displaylink-debian #232: can't read /lib/systemd/system/dlm.service

matthewhegarty, 2019-03-04:
Have seen the following issue when re-installing.  Also saw the error when uninstalling and re-installing from scratch.

```
Performing post install steps

sed: can't read /lib/systemd/system/dlm.service: No such file or directory
Failed to enable unit: Unit file dlm.service does not exist.
Setup DisplayLink xorg.conf depending on graphics card
```

[Full log](https://github.com/AdnanHodzic/displaylink-debian/files/2927818/err1.txt)

[Debug log](https://github.com/AdnanHodzic/displaylink-debian/files/2927821/debug1.txt)

AdnanHodzic, 2019-03-05:
`dlm.service` doesn't exist with 5.1 driver anymore, make sure you're running latest code and run following.

Run these line by line, if one of them fails just proceed to next one:

```
sudo dkms remove evdi/1.6.0 --all -q 
sudo systemctl stop dlm.service 
sudo systemctl disable dlm.service 
sudo rm -f /lib/systemd/system/dlm.service
sudo rmmod evdi 
sudo ./displaylink-debian.sh --uninstall 
```

Make sure to reboot after uninstall is finished.

After reboot run debug (`sudo displaylink-debian.sh --debug`) and make sure nothing under "DisplayLink info" is installing/up and running. After this install again and you should be set:
`sudo displaylink-debian.sh --install`

Darcidride, 2019-03-14:
@matthewhegarty any news on your issue ? is it fixed ?

matthewhegarty, 2019-03-15:
@Darcidride Yes - the issue is solved now.  thanks

closed 2019-03-17 by AdnanHodzic