AdnanHodzic/displaylink-debian #233: Can't startx on branch master

aowrew-fp, 2019-03-06:
Hi,

I've pulled the latest master and since then the Xserver can't start on boot (neither can it start manually).
After deleting the dl xorg.conf file I can start X, however there are no output providers. Since I just have a vague understanding of xorg configuration, could you please help me track down the issue?

I'm attaching the --debug output with the original xorg.conf file:

```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: sid
Kernel: 4.19.0-3-amd64

---------------- DisplayLink info ----------------

Driver version: 4.2.29
DisplayLink service status: 
EVDI service version: 1.5.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: HD
VGA: Intel Corporation HD Graphics 630 (rev 04)
VGA (3D): NVIDIA Corporation GP107M [GeForce GTX 1050 Mobile] (rev a1)
X11 version: 1.20.4-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xrdp/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

AdnanHodzic, 2019-03-09:
What happens if you completely removed `/etc/X11/xorg.conf.d/20-displaylink.conf` file and rebooted?

Darcidride, 2019-03-14:
For some reasons i don't understand, the removal of the `/etc/X11/xorg.conf.d/20-displaylink.conf` helps in some cases to make the displaylink driver works properly
See https://github.com/AdnanHodzic/displaylink-debian/issues/239 i will update this issue with some explanations

AdnanHodzic, 2019-03-17:
Thanks, because some of these are very common I just updated prerequisites section of [post install guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#prerequisites) with this information.

closed 2019-03-17 by AdnanHodzic

Darcidride, 2019-03-17:
@AdnanHodzic It's not only on ATI/Nvidia case, it was my case on Intel integrated graphics, i think It's a most general issue

AdnanHodzic, 2019-03-19:
I know, when DisplayLink 5.1 driver was just released, because now it also does `modesetting` I disabled generation of `/etc/X11/xorg.conf.d/20-displaylink.conf` based on your graphic card.

But it turns out that after all this step still needs to be done for some of all graphic cards (Intel/Nvidia/ATI). So I enabled it back.

Hence the reason, why I added it on top of"post install" guide and as removing it solves most of people problems.

If I had list of exact card this needs to be done for I could update `displaylink-debian` to do this automatically.