AdnanHodzic/displaylink-debian #235: Linux Mint LMDE2 Betsy support

triplemaya, 2019-03-07:
As requested I have run:
./displaylink-debian.sh --debug
output:
 # ./displaylink-debian.sh --debug
Failed to get D-Bus connection: Unknown error -1

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: No such file or directory
--------------- Linux system info ----------------

Distro: LinuxMint
Release: betsy
Kernel: 3.16.0-7-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
nouveau
Subsystem: Ltd
Ltd.
VGA: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller (rev 06)
NVIDIA Corporation G98 [GeForce 8400 GS Rev. 2] (rev a1)
VGA (3D): 
X11 version: 1.16.4-1+deb8u2
X11 configs:

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x99 cap: 0x7, Source Output, Sink Output, Source Offload crtcs: 2 outputs: 3 associated providers: 0 name:nouveau
Provider 1: id: 0x44 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

AdnanHodzic, 2019-03-09:
Your Nvidia card wasn't properly detected, what happens if you create `/etc/X11/xorg.conf.d/20-displaylink.conf` file with following contents:

```
Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection
Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection
Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection
Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection
Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection
```

Followed by reboot?

triplemaya, 2019-03-09:
Hi.
Thanks. That just make the graphics fail to start.
BTW that is not the machine I want to run it on. Just did the debug to be sure to have some info to send in.
Running on the laptop I want to use, as with the desktop, error message of course just gives:
Unsupported platform: LinuxMint betsy
When I run the debug the outpup is all exactly the same except that I have no nvidia on that machine. And the intel graphics are the haswell rather than ivy bridge.
All other lines identical. Except for:
Driver version: 4.2.29
whereas before on the desktop the version was blank
and the monitor details are different.

AdnanHodzic, 2019-03-09:
@triplemaya ok, now I understand what the problem was, in last commit I pushed the changes which add support for Linux Mint Betsy.

Pull the latest changes and re-run install, if the problem persists please re-open the issue.

commit 0fafbb63 (2019-03-09): added support for Linux Mint Betsy (close: #235)
closed 2019-03-09 by AdnanHodzic