AdnanHodzic/displaylink-debian #236: Graphical artifact on Text editor and terminal

sjohnson2019, 2019-03-09:
i am getting graphical artifacts, text disappears and reappears constantly yet seemingly randomly,
the terminal ages behind the system considerably (that may not be a driver issue) 

Debug info: 
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.15.0-29-generic

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [HD
VGA: Intel Corporation HD Graphics 520 (rev 07)
VGA (3D): 
X11 version: 1.19.6-1ubuntu4
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x9d cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 2 name:Intel
Provider 1: id: 0x78 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
Many thanks in advance (this driver has worked great in past systems)

AdnanHodzic, 2019-03-09:
What happens if you changed contents of `/etc/X11/xorg.conf.d/20-displaylink.conf` file to:

```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

followed by reboot?

If that doesn't work out, what would happen if you completely removed `/etc/X11/xorg.conf.d/20-displaylink.conf` file followed by reboot?

sjohnson2019, 2019-03-12:
Hello 

I  have rebooted ( I have been away from my machine since this ticket was posted and it seems okay with no intervention

I think it may have just required that extra reboot for some reason 

Many Thanks

sjohnson2019, 2019-03-12:
okay it started again. 

the displaylink.conf file already contained the first fix, so i tried the second and now the screens dont work at all (my main laptop ones do but not the ones connected up to the displaylink dock. 

is this an expected result?

sjohnson2019, 2019-03-12:
i removed it which seemed to have broke it. 

i then did a full uninstall and a reinstall and it now seems fine 

Many hanks :)

Darcidride, 2019-03-14:
Great to see it's working @sjohnson2019 
@AdnanHodzic can close this issue i guess :)

closed 2019-03-17 by AdnanHodzic

olberger, 2019-05-31:
Doesn't removing the reference to evdi from the Device section mean that it'll no longer be needed ? AFAIU, then, the intel driver does the job, and all of the DisplayLink may not be needed at all ?

r4dh4l, 2019-09-22:
> What happens if you changed contents of `/etc/X11/xorg.conf.d/20-displaylink.conf` file to:
> 
> ```
> Section "Device"
>     Identifier  "DisplayLink"
>     Driver      "modesetting"
>     Option      "PageFlip" "false"
> EndSection
> ```
> 
> followed by reboot?

Solved for me the problem after upgrading from Debian9 to Debian10.1 which caused that my ext. display was detected but showing no conent.

c24o, 2021-07-12:
I deleted 20-displaylink.conf but it didn't work, then I realized that there was another .conf file with the section "Device", I deleted that one too and then it worked :)