AdnanHodzic/displaylink-debian #237: Distribution Support Request: Latest Debin Stretch (2019-02-16) with 4.9.0-8-amd64 kernel

varelajp, 2019-03-11:
Hi Adnan!
Thank you for your hard work in this great driver.
It worked like a charm with Debian Stretch 9.3.
Recently I did a clean install of Debian Stretch 9.8 and the driver appears not to work enabling dlm.service:
**systemctl start dlm.service**
Failed to start dlm.service: Unit dlm.service not found.

In /usr/lib/x86_64-linux-gnu/ I have libstdc++.so.6 and libstdc++.so.6.0.22.
I made the symbolic link: **ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6 /opt/displaylink/libstdc++.so.6**
I overwrote libstdc++.so.6 in /opt/displaylink/
¿Maybe my mistake? ¿Maybe a recent Linux kernel? ¿Maybe a recent libstdc++.so.6?

Now the commands you asked for:

**cat /etc/X11/xorg.conf.d/20-displaylink.conf**
Section "Device"
Identifier "Intel"
Driver "intel"
EndSection

**lspci | grep -i vga**
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)

**lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'**
i915

**lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5**
Graphics

**./displaylink-debian.sh --debug**

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: stretch
Kernel: 4.9.0-8-amd64

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
VGA (3D):
X11 version: 1.19.2-1+deb9u5
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
Section "Device"
Identifier "Intel"
Driver "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:Intel

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

Regards...
Juan Varela

lhy0403, 2019-03-11:
Actually, the latest DisplayLink driver renames its service name, it's now `displaylink-driver.service`
You may want to update to the latest installation script.

AdnanHodzic, 2019-03-11:
@varelajp thanks! Please feel free to donate to the project: https://github.com/AdnanHodzic/displaylink-debian/blob/master/README.md#donate-paypal

As it was pointed out by @ihy0403 maybe try what I suggested here Can you try doing same thing I suggested here: https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348

Or if that doesn't work: https://github.com/AdnanHodzic/displaylink-debian/issues/236#issuecomment-471213411

lhy0403, 2019-03-13:
> As it was pointed out by @ihy0403 maybe try what I suggested here Can you try doing same thing I suggested here:

Haha, my ID is `Lhy0403`, it's quite confusing.

AdnanHodzic, 2019-03-17:
@lhy0403 sorry for the confusion, did my suggestion work in your case?

closed 2020-07-18 by Barabazs