AdnanHodzic/displaylink-debian #238: 3 Montiors not working

ConnerAiken, 2019-03-13:
Hi,

I've looked through the troubleshooting guide/post install guide and had no luck. I'm running linux mint 19 on a XPS 9575-735.

At first, the setup was working after a few reboots and I was able to use all three montiors. 
- My laptop screen
- Two identical dell 24" plugged into plugable usb-c docking station ( [ud-ultcdl](https://plugable.com/products/ud-ultcdl/) )

I've tried flipping displays on and off using the display application in mint but no dice. One monitor works fine but the one that is not plugged into the "4K HDMI" port will show a distorted display. It shows horziontal bars of the desktop background. If I screenshot it, the display will look fine.

Here is my debug output..

```
--------------- Linux system info ----------------

Distro: LinuxMint
Release: tessa
Kernel: 4.18.0-041800-generic

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
Integrated
USB
VGA: Intel Corporation Device 591b (rev 04)
VGA (3D): 
X11 version: 1.20.1-3ubuntu2.1~18.04.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x48 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 5 associated providers: 1 name:Intel
Provider 1: id: 0x1a8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

AdnanHodzic, 2019-03-13:
What would happen if you did what I suggested here: https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348

ConnerAiken, 2019-03-13:
Just had some luck. I think my linux kernel or headers or something were out of date.

I updated my linux-headers packages with [ukuu](https://github.com/teejee2008/ukuu) to the latest, reinstalled and now all three monitors are working fine.

Some minor cursor flickering but I can deal with that.

closed 2019-03-13 by ConnerAiken