AdnanHodzic/displaylink-debian #239: Debian Buster/unstable : Display not detected

Darcidride, 2019-03-14:
Hi, 

First of all, your work on this Displaylink-driver is amazing, keep up the great work! :)

I'm available to try to fix it with you :)

https://github.com/AdnanHodzic/displaylink-debian/issues/212#issue-406172235

I've the same configuration as the creator of this issue and the same issue as him. Also i can tell it still doesn't work after your branch merge and with the new code improvement you told before closing this issue.

Here is my debug output : 

```
systemctl status displaylink-driver.service 
● displaylink-driver.service - DisplayLink Driver Service
   Loaded: loaded (/lib/systemd/system/displaylink-driver.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-03-13 00:16:34 CET; 12min ago
  Process: 1360 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/5.1.26 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 1376 (DisplayLinkMana)
    Tasks: 33 (limit: 4915)
   Memory: 31.2M
   CGroup: /system.slice/displaylink-driver.service
           └─1376 /opt/displaylink/DisplayLinkManager
```

```
cat /etc/modprobe.d/blacklist.conf    
blacklist udlfb
blacklist udl
```

```
sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-3-amd64

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Gen
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.20.3-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x4c cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 9 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

My hardware : Macbook Pro mid-2012 (intel integrated graphics from a i5-3210M) --> Very common laptop

I'm on Xorg for the test : 
```
loginctl show-session 2 -p Type                        ✘
Type=x11
```

My OS : 
```
$ uname -a   
Linux LUNE 4.19.0-3-amd64 #1 SMP Debian 4.19.20-1 (2019-02-11) x86_64 GNU/Linux
```
```
$ cat /etc/debian_version
buster/sid
```
--> Yes it's a unstable Debian

By the way i noticed a bug in your script, for the latest kernel (4.19.0-3-amd64) to work with your script, you need "linux-kbuild-4.19" in its latest version with the command line "sudo apt install -t unstable linux-kbuild-4.19" (if you don't have the "4.19.20-1" version of "linux-kbuild-4.19" you will have dependencies issues and a crash :)):

```
apt-cache policy linux-kbuild-4.19
linux-kbuild-4.19:
  Installed: 4.19.20-1
  Candidate: 4.19.20-1
  Version table:
 *** 4.19.20-1 100
        100 https://deb.debian.org/debian unstable/main amd64 Packages
        100 /var/lib/dpkg/status
     4.19.16-1 900
        900 https://deb.debian.org/debian buster/main amd64 Packages
     4.19.16-1~bpo9+1 100
        100 https://deb.debian.org/debian stretch-backports/main amd64 Packages
```

```
cat /etc/X11/xorg.conf.d/20-displaylink.conf           ✘
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
```

xrandr can't find output sources ... :/
```
xrandr --setprovideroutputsource 1 0
Could not find provider with index 1
```
```
xrandr --setprovideroutputsource 2 0
Could not find provider with index 2
```

xranr basic info :
```
xrandr
Screen 0: minimum 8 x 8, current 1280 x 800, maximum 32767 x 32767
LVDS1 connected primary 1280x800+0+0 (normal left inverted right x axis y axis) 290mm x 180mm
   1280x800      60.22*+  59.91  
   1280x720      59.74  
   1024x768      60.00  
   1024x576      59.90    59.82  
   960x540       59.63    59.82  
   800x600       60.32    56.25  
   864x486       59.92    59.57  
   640x480       59.94  
   720x405       59.51    58.99  
   640x400       60.00  
   640x360       59.84    59.32  
DP1 disconnected (normal left inverted right x axis y axis)
DP2 disconnected (normal left inverted right x axis y axis)
DP3 disconnected (normal left inverted right x axis y axis)
HDMI1 disconnected (normal left inverted right x axis y axis)
HDMI2 disconnected (normal left inverted right x axis y axis)
HDMI3 disconnected (normal left inverted right x axis y axis)
VGA1 disconnected (normal left inverted right x axis y axis)
```

I think i gave you all files/configurations i checked when i was trying to find were was the mistake, but everything seems correct to me... I will be really graceful if you figure out what's wrong ! :)

Any help appreciated, and thank you in advance for your reading ! :)

_It's very frustrating this pain in the ass to work on this to get it working on our favorite OS :/
Will we be able to use an external display through USB3 on GNU/Linux one day... ? :S_

(PS : all of my tests was done on Debian unstable with X11, and i'm trying to plug a DVI external display to my laptop through USB3 displayport adapter which is a 'plugable USB 3.0 dual head graphics DVI/HDMI' adaptor)

Darcidride, 2019-03-14:
Just saw https://github.com/AdnanHodzic/displaylink-debian/issues/237 issue, i need to check if your adviced solutions are working in my case :)

Darcidride, 2019-03-14:
So, before all, a little bug i noticed :
When the dependencies failed with `linux-kbuild-4.19` checks (for unstable debian, because it's not hte tatally latest version), the script continue to run and do things, after updating `linux-kbuild-4.19` to the latest version for unstable, and restart the installation process with the script,  it appears that the installation isn't working as it should, some things are missing i guess. BUT, after a uninstall and then a install again, everything works.
For the users of this script : be careful to uninstall the script if you have any errors, and then use install parameter again to be sure everything is clean
For the maintainer : some things can surely be upgraded here in order to not put an half done installation

Darcidride, 2019-03-14:
My final goal is to have :
- My primary display from my laptop :heavy_check_mark: 
- + 1 external 1080p display through the plugeable (displaylink) adapter :heavy_check_mark: 
- + 1 external 1080p display through thunderbolt 2.0 to HDMI adapter (https://www.amazon.com/AmazonBasics-Mini-DisplayPort-Thunderbolt-Adapter/dp/B00NH13K8S) :x: (NOT working currently after installing displaylink driver and make it working

Darcidride, 2019-03-14:
# Thunderbolt2 Fixing

When the displaylink + thunderbolt2 adapter are plugged in my laptop, i can see this report of xrandr :
```
$ xrandr --listproviders                                                                                              
Providers: number : 3
Provider 0: id: 0x7b cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 8 associated providers: 2 name:modesetting
Provider 1: id: 0x1c2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```

With the command 
```
$ xrandr --setprovideroutputsource 1 0 && xrandr --setprovideroutputsource 2 0
$
```
No return from the command line (seems ok ?) but the thunderbolt display is still not working :/

AdnanHodzic, 2019-03-17:
Hi @Darcidride thank you for your words of encouragement, if it all ends up working for you consider [donating to the project](https://github.com/AdnanHodzic/displaylink-debian/#donate-paypal) :)

Sorry for such a delayed reply, been very busy for last few days IRL.

I didn't understand why you had to install `linux-kbuild-4.19`, was this because `linux-kbuild-$(uname -r)` was supposed to installed by `displaylink-debian` and it wasn't, or? I'd definitely want to add any changes to the tool if necessary. 

You probably did, but have you tried using any [other tools to achieve the desired monitor configuration as explained](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#screen-layout-configuration)?

Darcidride, 2019-03-17:
Hi @AdnanHodzic,
I will definitely give you a donation when everything is working, seems fair to me :)

For the `linux-kbuild-$(uname -r)`, maybe i didn't explain it well, but the fact is this package have some different version for a defined kernel, at least on Debian.

So, this package have different version for a same kernel, and for my case on Debian unstable, i needed to install the latest version of the package and dependencies to allow your script not to crash at some point because of unmet dependencies 

For your advices about the most common issues on the linked pages, I already read and tested it, i also read and tested many things in the open/closed issues on this github repository, without any improvement. (But it, at least, give me some tricks about the USB adapter to make it working)

Do you have any other ideas, or information i can give you to help me to make my goal working ? :)

Darcidride, 2019-03-19:
Ok, I found the issue, after all of the investigations, it turns out that, inside the thunderbolt2 adapter, a paste from the connector was unsoldered...

I can confirm the displaylink-debian driver is working fine, and on a Macbook Pro A1278 13' (with intel integrated HD graphics), it's working fine with 3 externals displays + the primary laptop display :)

_(PS : the 2 external displays from the USB displaylink adapter aren't 1080p, not possible because, i think, i don't have a USB3 output on my laptop but it's fine :))_

closed 2019-03-19 by Darcidride

Darcidride, 2019-04-15:
Hi, i updated the thread after some checks I did when i upgraded my kernel to a newer version