AdnanHodzic/displaylink-debian #241: Support for MX 18

gponty, 2019-03-20:
My debug conf :

Failed to retrieve unit: No such method 'GetUnit'

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: MX
Release: Continuum
Kernel: 4.19.0-1-amd64

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: 
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: Ltd
Ltd
VGA: Intel Corporation Device 3e9b
NVIDIA Corporation GP106M [GeForce GTX 1060] (rev a1)
VGA (3D): 
X11 version: 1.19.2-1+deb9u5
X11 configs: /etc/X11/xorg.conf.d/synaptics.conf /etc/X11/xorg.conf.d/10-evdev.conf /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x44 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

My script doesnt start, but see first line : 
Failed to retrieve unit: No such method 'GetUnit'

How i can resolve this ?

Linux-source is ok :
$ apt search linux-source
En train de trier... Fait
Recherche en texte intégral... Fait
linux-source/stable,stable,stable,stable,now 4.9+80+deb9u6 all  [installé]
  Linux kernel source (meta-package)

linux-source-4.13/mx,mx 4.13.13-1mx17 all
  Linux kernel source for version 4.13 with Debian patches

linux-source-4.14/mx,mx 4.14.13-1~bpo9+1mx17+2 all
  Linux kernel source for version 4.14 with Debian patches

linux-source-4.15/mx,mx 4.15.4-1~mx17+1 all
  Linux kernel source for version 4.15 with Debian patches

linux-source-4.18/mx,mx 4.18.10-1~mx17+1 all
  Linux kernel source for version 4.18 with Debian patches

linux-source-4.19/mx,mx 4.19.5-2~mx17+1 all
  Linux kernel source for version 4.19 with Debian patches

linux-source-4.9/stable-updates,stable-updates,now 4.9.144-3.1 all  [installé, automatique]
  Linux kernel source for version 4.9 with Debian patches

Thank you !

gponty, 2019-03-20:
I tried to start with my sysvinit, but script stuck.
I tried to start with debug :

$ sudo /opt/displaylink/DisplayLinkManager -logging
QQKyLaFKj5YpAlLRpeGF7ZcXS80IFDvMDwHIF4ccnaA4HNkwum7Seu5a0lrD6s4Glys8x9jreL28kEJ0/7pAYXNjUzmMuQbDyZp7Pb5YH8w=
v5kBwCjXYbdkdsHOQBY4PfaTXtyKhwhZrBsEWt0cbdN3YCH4+V+jrs7h4Nys2/NU5oyyP7QHR7kNrEp6o6DFHfI1vuE1embV3k+DRF0DyhZ+VnYwq+hYnnJUrg1gyX66
XtBP4eneJNnb07GcNqQ+hA4MCZxSlQNQpUltETELRjAn6GtYdKo04GFt8LQquZHdhTCRG0NFqaBQtQrN2hJdt1psRwoTRW9BHos9h79EFVdPyBjmx7yve2wB/TqbXwVU2uSF4EEeFdGLpAy6i20mIG2QC5/9KBp+l4UxhbZqI7u2dkmQehbyZlIV0tefPAmtcSCLdFwcjStAl5wckBrWKw==

gponty, 2019-03-20:
Another try (this command stuck)

sudo /etc/init.d/displaylink restart
[ ok ] Stopping DisplayLink Driver Service: displaylink-driver.
[....] Starting DisplayLink Driver Service: displaylink-driver

AdnanHodzic, 2019-03-24:
Did running `sudo /etc/init.d/displaylink restart` fix your problem in the end? Is driver up and running now? You can see this in top section of `sudo ./displaylink-debian.sh --debug` command.

gponty, 2019-03-25:
No, he stuck on this command :

[image]

elias-mueller, 2019-04-17:
Same here. The workaround for MX Linux is to enable systemd at the GRUB menu (advanced options).

closed 2020-07-18 by Barabazs